grzegorz914/homebridge-enphase-envoy #80: Live Data only reported for some time when opened in the app/web page

ahnungsbefreit, 2023-02-28:
Hello,

first of all: BIG THANK YOU for this plugin, the best solution to integrate Enphase in a SmartHome I found!

I have one problem: The plugin works and is reporting all kind of data on MQTT but the "live data" is only updated when I open live data in the app (or via web login) for some time and then it stops. To be precise: It is still updated every 5 sec as per my settings but with no change in values.

My setup is as follows (from homebridge.log):

`[37m[2/14/2023, 10:48:53 AM] [39mHomebridge v1.6.0 (HAP v0.11.0) (Homebridge AD15) is running on port 51986.
[37m[2/14/2023, 10:48:54 AM] [39m[36m[homebridge-enphase-envoy][39m Launched child bridge with PID 433
[37m[2/14/2023, 10:48:54 AM] [39mRegistering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[37m[2/14/2023, 10:48:54 AM] [39m[36m[homebridge-enphase-envoy][39m Loaded homebridge-enphase-envoy v6.5.5 child bridge successfully
[37m[2/14/2023, 10:48:54 AM] [39mLoaded 0 cached accessories from cachedAccessories.0EABB583479D.
[37m[2/14/2023, 10:48:54 AM] [39mHomebridge v1.6.0 (HAP v0.11.0) (homebridge-enphase-envoy) is running on port 48069.
[37m[2/14/2023, 10:49:05 AM] [39m[36m[homebridge-enphase-envoy][39m -------- Envoy-S --------
[37m[2/14/2023, 10:49:05 AM] [39m[36m[homebridge-enphase-envoy][39m Manufacturer: Enphase
[37m[2/14/2023, 10:49:05 AM] [39m[36m[homebridge-enphase-envoy][39m Model: Envoy
[37m[2/14/2023, 10:49:05 AM] [39m[36m[homebridge-enphase-envoy][39m Firmware: D7.6.122
[37m[2/14/2023, 10:49:05 AM] [39m[36m[homebridge-enphase-envoy][39m SerialNr: xxxxxxxxxxxx
[37m[2/14/2023, 10:49:05 AM] [39m[36m[homebridge-enphase-envoy][39m Time: 2/14/2023, 10:49:18 AM
[37m[2/14/2023, 10:49:05 AM] [39m[36m[homebridge-enphase-envoy][39m ------------------------------
[37m[2/14/2023, 10:49:05 AM] [39m[36m[homebridge-enphase-envoy][39m Q-Relays: 2
[37m[2/14/2023, 10:49:05 AM] [39m[36m[homebridge-enphase-envoy][39m Inverters: 22
[37m[2/14/2023, 10:49:05 AM] [39m[36m[homebridge-enphase-envoy][39m Batteries: 0
[37m[2/14/2023, 10:49:05 AM] [39m[36m[homebridge-enphase-envoy][39m --------------------------------
[37m[2/14/2023, 10:49:05 AM] [39m[36m[homebridge-enphase-envoy][39m Meters: Yes
[37m[2/14/2023, 10:49:05 AM] [39m[36m[homebridge-enphase-envoy][39m Production: Enabled
[37m[2/14/2023, 10:49:05 AM] [39m[36m[homebridge-enphase-envoy][39m Consumption: Enabled
[37m[2/14/2023, 10:49:05 AM] [39m[36m[homebridge-enphase-envoy][39m --------------------------------
[37m[2/14/2023, 10:49:05 AM] [39m[36m[homebridge-enphase-envoy][39m Encharges: Yes 1
[37m[2/14/2023, 10:49:05 AM] [39m[36m[homebridge-enphase-envoy][39m Wireless Kit: Yes 2
[37m[2/14/2023, 10:49:05 AM] [39m[36m[homebridge-enphase-envoy][39m --------------------------------
[37m[2/14/2023, 10:49:05 AM] [39mEnvoy-S 8DB7 is running on port 34475.`

My configuration:

`{
    "devices": [
        {
            "name": "Envoy-S",
            "host": "192.168.178.132",
            "envoyFirmware7xx": true,
            "envoyFirmware7xxToken": "<mytoken>",
            "disableLogInfo": false,
            "disableLogDeviceInfo": false,
            "enableDebugMode": false,
            "powerProductionMax": false,
            "powerProductionMaxAutoReset": 0,
            "powerProductionMaxDetected": 0,
            "energyProductionLifetimeOffset": 0,
            "powerConsumptionTotalMax": false,
            "powerConsumptionTotalMaxAutoReset": 0,
            "powerConsumptionTotalMaxDetected": 0,
            "energyConsumptionTotalLifetimeOffset": 0,
            "powerConsumptionNetMax": false,
            "powerConsumptionNetMaxAutoReset": 0,
            "powerConsumptionNetMaxDetected": 0,
            "energyConsumptionNetLifetimeOffset": 0,
            "supportProductionPowerMode": false,
            "supportPlcLevel": false,
            "supportEnsembleStatus": true,
            "supportLiveData": true,
            "liveDataRefreshTime": 5000,
            "metersDataRefreshTime": 10000,
            "productionDataRefreshTime": 10000,
            "enableRestFul": false,
            "enableMqtt": true,
            "mqttHost": "192.168.178.191",
            "mqttPort": 1882,
            "mqttPrefix": "/envoy",
            "mqttAuth": true,
            "mqttUser": "<myuser>",
            "mqttPasswd": "<mypasswd>",
            "mqttDebug": false
        }
    ],
    "_bridge": {
        "username": "<myusername>",
        "port": 48069
    },
    "platform": "enphaseEnvoy"
}`

Any advice how to fix this?

grzegorz914, 2023-02-28:
To check this You need enable debug log in plugin advanced section and observe when its stop report Live Data. Post the log here after record. In other case in 6.7.2 

I have added powerful RESTful server which is best to use with own automations.
To get this work in RESTFul section enable it set own port or leave default and restart plugin. 
Too get data just type in web browser or other client `http//ipaddress:port/path`  where path are: `info`, `home`, `inventory`, `meters`, `metersreading`, `ensembleinventory`, `ensemblestatus`, `gridprofile`, `livedata`, `production`, `productionct`, `microinverter`, `powermode`, `plclevel`.  
All response are as JSON data.

ahnungsbefreit, 2023-02-28:
So, I updated to 6.7.2 now and captured the log. As you can see, after 12:15:07 live data is still reported but the values relevant for me like `meters.pv.agg_p_mw` are not updated any more, always reporting the same value. No errors in the logs as far as I can see.  Am I using the wrong data points to capture live production, grid consumption etc. or is this a bug? Note: If I would open the live data in the app now, it would work again for approx. 15 minutes 
[2023-02-28 Homebridge log extract.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10850298/2023-02-28.Homebridge.log.extract.txt)

Regarding your RESTFul server: Looks promising but now I have build all my integration on MQTT...

grzegorz914, 2023-02-28:
the data is read from Envoy, if not changed then must be so reported from envoy. I have one tip for You, instead `meters.pv.agg_p_mw` use `activePower` from meters reading.

closed 2023-02-28 by ahnungsbefreit