grzegorz914/homebridge-enphase-envoy #82: No accessories showing up

bmcclure, 2023-03-14:
I have the 7.x firmware, which I have set up in the plugin successfully. It authenticates, and it spits out a bunch of info about all my devices. However, it doesn't actually add any accessories to my system. The only things it adds are the 3 power peak sensors if they are enabled in the settings, otherwise it seems to add nothing.

I use the Home Assistant HomeKit integration, and I paired with the bridge for the plugin, as well as the bridge that the plugin created for my Envoy system (so two separate bridges), but it still only shows the power peak sensors and nothing else. Shouldn't I actually be able to get information about my inverters, batteries, power usage, etc?

Thank you for the assistance!

bmcclure, 2023-03-15:
Hello,

I did read that before I submitted the issue. I have filled out all the configuration values offered in the UI, it seems like I'm following the instructions according to the documentation, and the connection is working fine. I guess I don't understand what part of the About the Plugin section you're referring to that I might be misunderstanding.

bmcclure, 2023-03-15:
For some additional detail, I have one Envoy system: an ensemble connected to panels, microinverters, and batteries,

The documentation you linked to says "All devices are detected automatically" from which I inferred that once I set up the Envoy device in the configuration and restarted Homebridge, it would detect the devices connected to the Envoy installation. It does show them in the Homebridge logs, but I just don't understand how to get it to actually create an accessory for the Envoy to access the stats.

grzegorz914, 2023-03-15:
The plugin use custom characteristics which not exists and are not supported by Home.app, to see the all data from envoy as accessories please use Controller app or Eve app

grzegorz914, 2023-03-15:
You can also get all data using MQTT or RESTful

bmcclure, 2023-03-15:
Oh ok, I don't use iOS, and it sounds like it can't integrate with Home Assistant's HomeKit integration. MQTT would likely work though, I'll keep that in mind. Thank you for the clarification!

closed 2023-03-15 by bmcclure