grzegorz914/homebridge-enphase-envoy #85: Live data not refreshing

ydestord, 2023-05-05:
Dear,

When using the plugin the live data is not refreshing unless it's triggered through a connection to the live data with the Enlighten app. 

This is documented in https://support.enphase.com/s/question/0D53m00008sfiRcCAI/live-status-via-envoy-or-the-app-offline-feature-freezes?t=1683273624635 

the sc_stream flag will become enabled and the MQTT data will be updating

{
  "connection": {
    "mqtt_state": "connected",
    "prov_state": "configured",
    "auth_state": "ok",
    "sc_stream": "disabled",
    "sc_debug": "disabled"
  },

The workaround suggested by Enphase is to query   ivp/livedata/status.

API ivp/livedata/stream needs to be used in POST, json to pass: {“enable”: 1}

Field ‘mains_admin_state’ is not a required field
Response should be, 

{

"sc_stream": "enabled"

}

KR

Yves

ahnungsbefreit, 2023-05-05:
This is exactly the issue I also described here: https://github.com/grzegorz914/homebridge-enphase-envoy/issues/80. Would be great to get this fixed...

ahnungsbefreit, 2023-05-08:
Just tried with the CURL command listed on the enphase support page listed above and this works for me. Implementing this would really make the plugin even more useful, @grzegorz914 !

grzegorz914, 2023-05-08:
will do it asap

grzegorz914, 2023-05-10:
try 6.7.33-beta.0 and let me know, debug  log please

ahnungsbefreit, 2023-05-10:
Just installed it and it works. I had enabled LiveData with CURL already before so I cannot tell (yet) if it solves the problem permanently. Last time I did the enabling with CURL it was reporting for about 1,5 days before stopping again. I will report back here when any issue appears. Many thanks!

ahnungsbefreit, 2023-05-11:
[2023-05-11 Homebridge Enphase log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/11450329/2023-05-11.Homebridge.Enphase.log.txt)
@grzegorz914 Hope this contains what you have been looking for.

grzegorz914, 2023-05-12:
Is that working correct, any issue?

ydestord, 2023-05-12:
[homebridge.log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/11460658/homebridge.log.txt)

Stopped updating this night, debug log attached. Had to restart around 7:03AM

ahnungsbefreit, 2023-05-12:
Now it also stopped on my side with `Device: 192.168.178.132 Envoy-S, Update live data error: AxiosError: Request failed with status code 400. Reconnect in 15s`. After executing the CURL again, it worked. So it seems you need to add a check and if the live data is not updated for some time, the API needs to be enabled again.

grzegorz914, 2023-05-12:
Debug Please , go to plugin settings advanced section nad enable debug, then get the whole log.

ydestord, 2023-05-12:
I had attached my debug log this morning

Yves

grzegorz914, 2023-05-12:
No, you have attached normal log, you need enable debug in my plugin settings advanced section.

grzegorz914, 2023-05-12:
Im pushed 6.7.34-beta.0 please install and let me know with debug log again

ydestord, 2023-05-14:
HI,

Stopped again today after running the latest 6.7.34-beta.0 since Friday. Restarting homebridge didn't help. Still error 400 requesting live data.  I had to force the scstream to enabled and then it started working again

Attached debug log

[homebridge.log (2).txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/11472564/homebridge.log.2.txt)

grzegorz914, 2023-05-14:
try beta.1

grzegorz914, 2023-05-23:
Any info?

ydestord, 2023-05-25:
Hello,

Last weekend it stopped working so I had put my workaround In motion again ( curl to 'enable' the sc_stream). I stopped this one again and am waiting for the issue to occur again. Once I provoke it I will share the logs

KR

Yves

grzegorz914, 2023-07-20:
Still the problem exist with latest 7.x.x versions?

ydestord, 2023-07-20:
yes, from time to time I have to push the  'enable:1' to /ivp/livedata/stream. But much less frequent then before

ahnungsbefreit, 2023-07-30:
> yes, from time to time I have to push the 'enable:1' to /ivp/livedata/stream. But much less frequent then before

Same here.

grzegorz914, 2023-07-30:
any debug log?  I don't have envoy with fw7.x.x and can not test this locally

ahnungsbefreit, 2023-07-30:
I will try to create one but this will be tricky as it only occurs every couple of days...

grzegorz914, 2023-07-30:
I have pushed 7.3.6-beta.0 with enabled debug only for live data. This reduce the log.

grzegorz914, 2023-07-30:
I have pushed beta.1 to test, can you disable live data with:
`curl -f -k -H "Authorization:bearer <token> -H "Content-Type:application/json" -d "{\"enable\":0}" https://envoy.local/ivp/livedata/stream`

and check again with beta.1

ahnungsbefreit, 2023-07-30:
Mmh, not really helping I guess:
`[37m[7/30/2023, 12:21:46 PM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.178.132 Envoy-S, Update live data error: Enable live data stream error: AxiosError: Request failed with status code 400.. Trying again in: 5 sec.[39m
[37m[7/30/2023, 12:21:51 PM] [39m[36m[homebridge-enphase-envoy][39m Device: 192.168.178.132 Envoy-S, debug: Live data: {
  "connection": {
    "mqtt_state": "connected",
    "prov_state": "configured",
    "auth_state": "ok",
    "sc_stream": "enabled",
    "sc_debug": "disabled"
  },
  "meters": {
    "last_update": 1690719729,
    "soc": 100,
    "main_relay_state": 1,
    "gen_relay_state": 5,
    "backup_bat_mode": 1,
    "backup_soc": 0,
    "is_split_phase": 0,
    "phase_count": 3,
    "enc_agg_soc": 100,
    "enc_agg_energy": 3500,
    "acb_agg_soc": 0,
    "acb_agg_energy": 0,
    "pv": {
      "agg_p_mw": 667121,
      "agg_s_mva": 851692,
      "agg_p_ph_a_mw": 210791,
      "agg_p_ph_b_mw": 182769,
      "agg_p_ph_c_mw": 273560,
      "agg_s_ph_a_mva": 244074,
      "agg_s_ph_b_mva": 225773,
      "agg_s_ph_c_mva": 381844
    },
    "storage": {
      "agg_p_mw": 0,
      "agg_s_mva": 0,
      "agg_p_ph_a_mw": 0,
      "agg_p_ph_b_mw": 0,
      "agg_p_ph_c_mw": 0,
      "agg_s_ph_a_mva": 0,
      "agg_s_ph_b_mva": 0,
      "agg_s_ph_c_mva": 0
    },
    "grid": {
      "agg_p_mw": -338577,
      "agg_s_mva": -247076,
      "agg_p_ph_a_mw": 38329,
      "agg_p_ph_b_mw": -182587,
      "agg_p_ph_c_mw": -194320,
      "agg_s_ph_a_mva": 388775,
      "agg_s_ph_b_mva": -241660,
      "agg_s_ph_c_mva": -394192
    },
    "load": {
      "agg_p_mw": 328544,
      "agg_s_mva": 604616,
      "agg_p_ph_a_mw": 249120,
      "agg_p_ph_b_mw": 182,
      "agg_p_ph_c_mw": 79240,
      "agg_s_ph_a_mva": 632849,
      "agg_s_ph_b_mva": -15887,
      "agg_s_ph_c_mva": -12348
    },
    "generator": {
      "agg_p_mw": 0,
      "agg_s_mva": 0,
      "agg_p_ph_a_mw": 0,
      "agg_p_ph_b_mw": 0,
      "agg_p_ph_c_mw": 0,
      "agg_s_ph_a_mva": 0,
      "agg_s_ph_b_mva": 0,
      "agg_s_ph_c_mva": 0
    }
  },
  "tasks": {
    "task_id": -2018101439,
    "timestamp": 1690475680
  },
  "counters": {
    "main_CfgLoad": 1,
    "main_CfgChanged": 1,
    "main_taskUpdate": 31,
    "MqttClient_publish": 908,
    "MqttClient_respond": 66,
    "MqttClient_msgarrvd": 33,
    "MqttClient_create": 28,
    "MqttClient_setCallbacks": 28,
    "MqttClient_connect": 28,
    "MqttClient_connect_err": 22,
    "MqttClient_connect_Err": 22,
    "MqttClient_subscribe": 6,
    "SSL_Keys_Create": 28,
    "sc_hdlDataPub": 609958,
    "sc_SendStreamCtrl": 7,
    "sc_SendDemandRspCtrl": 1,
    "rest_Status": 128483
  },
  "dry_contacts": {
    "": {
      "dry_contact_id": "",
      "dry_contact_type": "",
      "dry_contact_load_name": "",
      "dry_contact_status": 3050996
    }
  }
}`

grzegorz914, 2023-07-30:
`sc_stream` is enabled, the data need to be refreshed correct

ahnungsbefreit, 2023-07-30:
> `sc_stream` is enabled, the data need to be refreshed correct

Not really, I tried it again but I only see the data in MQTT when I enable the LIVE API again:
`[37m[7/30/2023, 1:09:45 PM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.178.132 Envoy-S, Update live data error: Enable live data stream error: AxiosError: Request failed with status code 400.. Trying again in: 5 sec.[39m
[37m[7/30/2023, 1:09:50 PM] [39m[36m[homebridge-enphase-envoy][39m Device: 192.168.178.132 Envoy-S, debug: Live data: {
  "connection": {
    "mqtt_state": "connected",
    "prov_state": "configured",
    "auth_state": "ok",
    "sc_stream": "disabled",
    "sc_debug": "disabled"
  },
  "meters": {
    "last_update": 1690722598,
    "soc": 100,
    "main_relay_state": 1,
    "gen_relay_state": 5,
    "backup_bat_mode": 1,
    "backup_soc": 0,
    "is_split_phase": 0,
    "phase_count": 3,
    "enc_agg_soc": 100,
    "enc_agg_energy": 3500,
    "acb_agg_soc": 0,
    "acb_agg_energy": 0,
    "pv": {
      "agg_p_mw": 5400356,
      "agg_s_mva": 5414954,
      "agg_p_ph_a_mw": 1762651,
      "agg_p_ph_b_mw": 1538336,
      "agg_p_ph_c_mw": 2099368,
      "agg_s_ph_a_mva": 1769935,
      "agg_s_ph_b_mva": 1538336,
      "agg_s_ph_c_mva": 2106683
    },
    "storage": {
      "agg_p_mw": 0,
      "agg_s_mva": 0,
      "agg_p_ph_a_mw": 0,
      "agg_p_ph_b_mw": 0,
      "agg_p_ph_c_mw": 0,
      "agg_s_ph_a_mva": 0,
      "agg_s_ph_b_mva": 0,
      "agg_s_ph_c_mva": 0
    },
    "grid": {
      "agg_p_mw": -5069412,
      "agg_s_mva": -5135115,
      "agg_p_ph_a_mw": -1500953,
      "agg_p_ph_b_mw": -1535226,
      "agg_p_ph_c_mw": -2033233,
      "agg_s_ph_a_mva": -1544777,
      "agg_s_ph_b_mva": -1540709,
      "agg_s_ph_c_mva": -2049630
    },
    "load": {
      "agg_p_mw": 330944,
      "agg_s_mva": 279839,
      "agg_p_ph_a_mw": 261698,
      "agg_p_ph_b_mw": 3110,
      "agg_p_ph_c_mw": 66135,
      "agg_s_ph_a_mva": 225158,
      "agg_s_ph_b_mva": -2373,
      "agg_s_ph_c_mva": 57053
    },
    "generator": {
      "agg_p_mw": 0,
      "agg_s_mva": 0,
      "agg_p_ph_a_mw": 0,
      "agg_p_ph_b_mw": 0,
[... 120 lines cut ...]
    "MqttClient_respond": 66,
    "MqttClient_msgarrvd": 33,
    "MqttClient_create": 28,
    "MqttClient_setCallbacks": 28,
    "MqttClient_connect": 28,
    "MqttClient_connect_err": 22,
    "MqttClient_connect_Err": 22,
    "MqttClient_subscribe": 6,
    "SSL_Keys_Create": 28,
    "sc_hdlDataPub": 612744,
    "sc_SendStreamCtrl": 10,
    "sc_SendDemandRspCtrl": 1,
    "rest_Status": 129086
  },
  "dry_contacts": {
    "": {
      "dry_contact_id": "",
      "dry_contact_type": "",
      "dry_contact_load_name": "",
      "dry_contact_status": 3050996
    }
  }
}
[37m[7/30/2023, 1:09:55 PM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.178.132 Envoy-S, Update live data error: Enable live data stream error: AxiosError: Request failed with status code 400.. Trying again in: 5 sec.[39m
`

danielpesch, 2023-09-07:
Hello Guys,

i'm owner of an enphase system since august and tryed the same thing like @ahnungsbefreit. I'm running the homebridge plugin on version v7.4.4 and got the same issue with the live data. Do you have any updates @grzegorz914 or has any body else an workaround for me?

Greetz Danny

ahnungsbefreit, 2023-09-07:
> an workaround for me?

Hi Danny, I solved it for me with a small script running in ioBroker which is checking is LiveDate is updated and if the last 5 MQTT strings did not contain an update, I re-active LiveData via an API call. This works fine so far and is triggered only every couple of days...

danielpesch, 2023-09-08:
Good morning, and thank you very much for your prompt response! I actually implemented a similar solution yesterday. I wrote a small script in IoBroker that makes periodic API calls and checks the status of 'sc_stream.' As long as it's set to 'enabled,' nothing happens. If it switches to 'disabled,' it notifies me via Telegram, makes an API call to set it back to 'enabled,' and then sends me a success message via Telegram. If you're interested in the script, please let me know.

grzegorz914, 2023-09-10:
Hi all,

Im pushed 7.4.5-beta.6 to test, here try to fix live data refresh.
I don't have encharges installed in my environment, so is hard to test it on my side.
Please test and let me know with debug log.

Thanks

grzegorz914, 2023-11-26:
fixed with 7.6.0

closed 2023-11-26 by grzegorz914