grzegorz914/homebridge-enphase-envoy #86: Why is there a switch in my Envoy that is always off?

chrisjshull, 2023-06-04:
[image]

What does this do?

grzegorz914, 2023-06-04:
This show power production mode enabled/disabled.

chrisjshull, 2023-06-04:
Why is always “off” though?

chrisjshull, 2023-06-04:
… because my solar panels show as producing power in the enlighten app.

grzegorz914, 2023-06-04:
Can You enable debug in plugin config advanced section, restart plugin and get the log Please.

chrisjshull, 2023-06-05:
[homebridge.log-18.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/11649252/homebridge.log-18.txt)

grzegorz914, 2023-06-05:
try 6.7.34-beta.11 and please enable `Power Mode` in plugin config `Advanced Settings`

grzegorz914, 2023-07-16:
Fixed in 7.1.0

closed 2023-07-16 by grzegorz914