grzegorz914/homebridge-enphase-envoy #88: Error on Meters Response: Moved Permanently

dockramer, 2023-06-16:
Hi, couple of questions
1. I'm getting this error with the xbar extension "Error on meters response: Moved Permanently"
2. How do we calculate the wattage from the inverters?  I've got 28 iq8A inverters, so is that max 500W x 28 = 14000?

I don't have installer access on my system, just the basic monitoring.  Does this require the installer access to get full info?

thanks

grzegorz914, 2023-06-17:
IQ8A max 366VA, so 28x366VA = 10248VA

dockramer, 2023-06-17:
Yes, production CT are installed, and I can monitor production via the enlighten app. 

I don’t have installer access though, so I’m wondering if that is the issue

dockramer, 2023-06-17:
Looks like the xbar authentication method no longer works, and the token method is needed.

dockramer, 2023-07-05:
> * Have Yoy Current Meters installed in your PV?
> * IQ8A max 366VA, so 28x366VA = 10248VA

Hi grzegorz,
Saw that you updated the home bridge plugin. thanks.  
wondering if you've gotten xbar working with tokens?

Thanks!
DK

grzegorz914, 2024-03-22:
Sorry, not planed yet.

closed 2024-03-22 by grzegorz914