grzegorz914/homebridge-enphase-envoy #89: Beta Test 7.0.0

grzegorz914, 2023-07-08:
I have pushed v7.0.0 beta to test:

## Changes for fw.7.x.x
- added support to get JWT Token automatically from enlighten server using user credentials data
- config schema updated

If you update from previous version then need to be added in plugin config Authorization section user name and password to the enlighten account, the token will be resolved automatically and stored in `/homebridge/enphaseEnvoy/envoyToken_xxxxx`
The properties for manual put token is removed from config.

Any response will be helpful.

Thanks

BVGorp, 2023-07-09:
installed the beta and got this in logging

[7/9/2023, 1:06:53 PM] [homebridge-enphase-envoy] Device: 172.22.23.118 Envoy, debug: Start.
[7/9/2023, 1:06:53 PM] [homebridge-enphase-envoy] Device: 172.22.23.118 Envoy, debug: Requesting JWT token.
[7/9/2023, 1:06:53 PM] [homebridge-enphase-envoy] Device: 172.22.23.118 Envoy, TypeError [ERR_INVALID_ARG_TYPE]: The "data" argument must be of type string or an instance of Buffer, TypedArray, or DataView. Received an instance of Object Reconnect in 15s.

grzegorz914, 2023-07-09:
try next beta please and let me know

BVGorp, 2023-07-09:
It fetched the token now. Will test further. Thanks for the quick fix

grzegorz914, 2023-07-14:
Released

closed 2023-07-14 by grzegorz914