grzegorz914/homebridge-enphase-envoy #90: Gateway not loading

nmeliasp, 2023-07-11:
Hello looking to see how i can get past this error i am receiving. I left the IP address blank to use the defualt

Device:  xxxx Update home data error: Error: self-signed certificate. Reconnect in 15s.

grzegorz914, 2023-07-12:
What happened if you add correct IP adress? Please also enable debug mode in plugin config advanced section and get the log.

nmeliasp, 2023-07-12:
i still get the following if i put the IP address of the gateway. These are the logs i see after enabling degug mode: 
7/12/2023, 1:55:54 PM] [enphaseEnvoy] Device:  Solar Gateway, Update info data error: Error: getaddrinfo ENOTFOUND envoy.local. Reconnect in 15s.
[7/12/2023, 1:56:09 PM] [enphaseEnvoy] Device:  Solar Gateway, Update home data error: Error: self-signed certificate. Reconnect in 15s.

nmeliasp, 2023-07-12:
Here is my config {
    "devices": [
        {
            "name": "Solar Gateway",
            "envoyFirmware7xx": false,
            "envoyPasswd": "xxxx",
            "disableLogInfo": false,
            "disableLogDeviceInfo": false,
            "enableDebugMode": true,
            "powerProductionMax": false,
            "powerProductionMaxAutoReset": 0,
            "powerProductionMaxDetected": 5400,
            "energyProductionLifetimeOffset": 0,
            "powerConsumptionTotalMax": false,
            "powerConsumptionTotalMaxAutoReset": 0,
            "powerConsumptionTotalMaxDetected": 10000,
            "energyConsumptionTotalLifetimeOffset": 0,
            "powerConsumptionNetMax": false,
            "powerConsumptionNetMaxAutoReset": 0,
            "powerConsumptionNetMaxDetected": 10000,
            "energyConsumptionNetLifetimeOffset": 0,
            "supportProductionPowerMode": false,
            "supportPlcLevel": false,
            "supportEnsembleStatus": false,
            "supportLiveData": false,
            "liveDataRefreshTime": 1000,
            "metersDataRefreshTime": 3000,
            "productionDataRefreshTime": 5000,
            "enableRestFul": false,
            "restFulPort": 3000,
            "enableMqtt": false,
            "mqttDebug": false,
            "mqttHost": "xxx",
            "mqttPort": 1883,
            "mqttPrefix": "home/envoy",
            "mqttAuth": false,
            "mqttUser": "xxxx"
        }
    ],
    "platform": "enphaseEnvoy"
}

grzegorz914, 2023-07-13:
Please put correct IP address in config, restart plugin and get the whole debug log. Looks like your envoy is not available in network.

nmeliasp, 2023-07-13:
i was able to get it configured, i was using firmware 7.xxx and had to get a token. However after adding the bridge into Homekit i am not finding any of the sensors? I see the sensors in Homebridge but not Homekit

grzegorz914, 2023-07-14:
Are you read this section:
https://github.com/grzegorz914/homebridge-enphase-envoy#about-the-plugin

nmeliasp, 2023-07-14:
Yes I used the eve app and can see the bridge but the accessories didn't come over with the bridge..

grzegorz914, 2023-07-14:
Are you ad the accessory itself to the HomeKit.ap?

nmeliasp, 2023-07-14:
they didnt show up in Home Kit either

grzegorz914, 2023-07-14:
Are you read this: 
https://github.com/grzegorz914/homebridge-enphase-envoy/wiki#adding-to-homekit 
and this:
https://github.com/grzegorz914/homebridge-enphase-envoy/wiki#troubleshooting

Can you also post whole debug log after restart plugin?

nmeliasp, 2023-07-14:
ok i got the accessories to show up in Eve now. Currently working on the top bar and i'm getting an Error on Meter Response: Moved Permanently. 

Any idea what this error means?

grzegorz914, 2023-07-15:
I can no help you one debug log, I need to know what happened in your system. Go to plugin settings Advanced Section and enable Debug Mode. Restart plugin and get the whole log.

grzegorz914, 2023-07-15:
TopBar is not working with firmware 7.x.x

closed 2023-07-15 by grzegorz914