grzegorz914/homebridge-enphase-envoy #92: No devices in Apple Home

KPfundstein, 2023-07-16:
Hi, it seems that I have some problems installing the Plugin in Homebroch or I do not understand how the plugin should work. I Installed homebridge image and the plugin on a RaspBerry 4. Afterwards I updated the configuration of the plugin to connect to my envoy. everything seems to work fine. I see a lot of communication between the plugin nd the envoy in the status screen of homebridge.  

[16/07/2023, 15:00:55] [homebridge-enphase-envoy] -------- Solar --------
[16/07/2023, 15:00:55] [homebridge-enphase-envoy] Manufacturer: Enphase
[16/07/2023, 15:00:55] [homebridge-enphase-envoy] Model: Envoy
[16/07/2023, 15:00:55] [homebridge-enphase-envoy] Firmware: D7.6.172
[16/07/2023, 15:00:55] [homebridge-enphase-envoy] SerialNr: xxxxxxxxx 
[16/07/2023, 15:00:55] [homebridge-enphase-envoy] Time: 16/07/2023, 15:01:43
[16/07/2023, 15:00:55] [homebridge-enphase-envoy] ------------------------------
[16/07/2023, 15:00:55] [homebridge-enphase-envoy] Q-Relays: 3
[16/07/2023, 15:00:55] [homebridge-enphase-envoy] Inverters: 60
[16/07/2023, 15:00:55] [homebridge-enphase-envoy] Batteries: 0
[16/07/2023, 15:00:55] [homebridge-enphase-envoy] --------------------------------
[16/07/2023, 15:00:55] [homebridge-enphase-envoy] Meters: Yes
[16/07/2023, 15:00:55] [homebridge-enphase-envoy] Production: Enabled
[16/07/2023, 15:00:55] [homebridge-enphase-envoy] Consumption: Enabled
[16/07/2023, 15:00:55] [homebridge-enphase-envoy] --------------------------------
[16/07/2023, 15:00:55] [homebridge-enphase-envoy] Encharges: Yes 3
[16/07/2023, 15:00:55] [homebridge-enphase-envoy] Wireless Kit: Yes 2
[16/07/2023, 15:00:55] [homebridge-enphase-envoy] --------------------------------

i do not See any error in the logs....

I added the bridge to my home app on my I-phone to a new room. 

when I open the room in the home app it is simply empty. I Expected to see some new devices like power meter there.

Is there any other step required ?

Thanks and best regards 

Klaus

grzegorz914, 2023-07-16:
You need add the accessory itself not the bridge only:
https://github.com/grzegorz914/homebridge-enphase-envoy/wiki#adding-to-homekit
and also read this:
https://github.com/grzegorz914/homebridge-enphase-envoy#about-the-plugin

KPfundstein, 2023-07-16:
Thank You for quick Support !!!

I enabled Power Peak Sensors it works fine and I can add the sensor to the app. I assume to monitor the current Power Production I have to enable the supportProductionPowerMode in the config... :

                    "disableLogInfo": false,
                    "disableLogDeviceInfo": false,
                    "enableDebugMode": false,
                    "powerProductionMax": true,
                    "powerProductionMaxAutoReset": 0,
                    "powerProductionMaxDetected": 0,
                    "energyProductionLifetimeOffset": 0,
                    "powerConsumptionTotalMax": false,
                    "powerConsumptionTotalMaxAutoReset": 0,
                    "powerConsumptionTotalMaxDetected": 0,
                    "energyConsumptionTotalLifetimeOffset": 0,
                    "powerConsumptionNetMax": false,
                    "powerConsumptionNetMaxAutoReset": 0,
                    "powerConsumptionNetMaxDetected": 0,
                    "energyConsumptionNetLifetimeOffset": 0,
                    "supportProductionPowerMode": true,
                    "supportPlcLevel": false,
                    "supportEnsembleStatus": false,
                    "supportLiveData": false,
                    "liveDataRefreshTime": 1000,
                    "metersDataRefreshTime": 2500,
                    "productionDataRefreshTime": 5000,
                    "enableRestFul": false,
                    "restFulDebug": false,
                    "enableMqtt": false,
                    "mqttDebug": false,
                    "mqttAuth": false
                    
After I enabled the supportProductionPowerMode I get the following ERROR in the LOG:
Device: 192.168.178.123 Solar, Get backbone app data error: ReferenceError: data is not defined. Reconnect in 15s.

grzegorz914, 2023-07-16:
ups, fixed in 7.0.3, ready o update now

KPfundstein, 2023-07-16:
Installed 7.0.3 the situation changed ....

Homebridge v1.6.1 (HAP v0.11.1) (Homebridge BF6C) is running on port 51161.
[16/07/2023, 16:04:23] [homebridge-enphase-envoy] Launched child bridge with PID 2871
[16/07/2023, 16:04:24] Registering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[16/07/2023, 16:04:24] [homebridge-enphase-envoy] Loaded homebridge-enphase-envoy v7.0.3 child bridge successfully
[16/07/2023, 16:04:24] Loaded 0 cached accessories from cachedAccessories.0E84D931E56D.
[16/07/2023, 16:04:24] Homebridge v1.6.1 (HAP v0.11.1) (homebridge-enphase-envoy) is running on port 32254.
[16/07/2023, 16:04:37] [homebridge-enphase-envoy] Device: 192.168.178.123 Solar, Update power mode error: AxiosError: Request failed with status code 401. Reconnect in 15s.

grzegorz914, 2023-07-16:
Please disable it, I don't have envoy with fw.7x.x and looks like  this data is not available with this firmware or is not accessible with Token. I will investigate into this issue in future.

grzegorz914, 2023-07-16:
It will be helpful if you get me whole debug log after restart plugin with enabled support production power mode, debug need to be enabled in plugin settings advanced section.

Thanks

grzegorz914, 2023-07-16:
Im pushed also 7.0.4-beta.0 , if you want to test then please update to beta, enable power production mode and let me know.

KPfundstein, 2023-07-16:
I send you an email with the log ...

KPfundstein, 2023-07-16:
i tested 7.0.4 beta ... it shows the same error message ... is there any other way, to get some kind of power meter to show the current production, consumption, battery charging ?

grzegorz914, 2023-07-16:
You get this all data in EVE app or Controller app

grzegorz914, 2023-07-16:
im pushed 7.0.4-beta.3, after update and before you restart the plugin please first remove envoyId file from /homebridge/enphaseEnvoy/ folder and please also the whole log.

KPfundstein, 2023-07-16:
log was send by mail ....

KPfundstein, 2023-07-16:
> You get this all data in EVE app or Controller app

isn't it necessary that there are some devices "power Meters"  are shown in HomeBridge ... So far I see only the Solar Power Peek Production "Trigger" Device in Homebridge ....

grzegorz914, 2023-07-16:
HomeBridge and HomeKit app dosent support custom characteristics, to see all this data need to use EVE or Controller app, also you can use MQTT or RESTFul to read all this data for your hour automation.

What contain your envoyId file, is 9 character number?

KPfundstein, 2023-07-16:
603980032

KPfundstein, 2023-07-16:
ok, have also eve! But how to get the device in eve, if it's not shown in homebridge ?

grzegorz914, 2023-07-16:
if you added it to the HomeKit app then will be displayed automatically

grzegorz914, 2023-07-16:
I close this as resolved main issue.

closed 2023-07-16 by grzegorz914