grzegorz914/homebridge-enphase-envoy #94: This plugin generated a warning from the characteristic 'Rest power': characteristic value expected valid finite number and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.

ahnungsbefreit, 2023-07-23:
With the update to the latest V 7.3.0, the login with Enphase credentials works (this was not the case before due to special characters in the password). However, I see now in the logs every minute the following message:

`[7/23/2023, 10:47:41 AM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Rest power': characteristic value expected valid finite number and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.`

grzegorz914, 2023-07-23:
can you get the debug?

ahnungsbefreit, 2023-07-23:
Thanks for this ultra-quick reply, much appreciated! Please find debug log attached.
[homebridge.log](https://github.com/grzegorz914/homebridge-enphase-envoy/files/12138563/homebridge.log)

grzegorz914, 2023-07-23:
Fixed in 7.3.2, looks like this properties doesn't exist in your envoy firmware

closed 2023-07-23 by grzegorz914

grzegorz914, 2023-07-23:
This was encharges summary - `rest power`, with latest fw. v7.6 172 no more exist.

commit 29ad4e71 (2023-07-23): fix #94