grzegorz914/homebridge-enphase-envoy #97: This plugin generated a warning from the characteristic 'Brightness': characteristic value expected valid finite number and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.

barthoefs, 2023-08-05:
Hello running version 7.3.5 I receive this:
```
[8/5/2023, 9:34:38 PM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Brightness': characteristic value expected valid finite number and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.
[8/5/2023, 9:34:38 PM] [homebridge-enphase-envoy] Error: 
    at Brightness.Characteristic.characteristicWarning (/var/lib/homebridge/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:3011:105)
    at Brightness.Characteristic.validateUserInput (/var/lib/homebridge/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:2903:14)
    at Brightness.Characteristic.updateValue (/var/lib/homebridge/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:2328:20)
    at Lightbulb.Service.updateCharacteristic (/var/lib/homebridge/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Service.ts:795:35)
    at /var/lib/homebridge/node_modules/homebridge-enphase-envoy/src/envoydevice.js:2262:26
```

barthoefs, 2023-08-05:
[homebridge.log](https://github.com/grzegorz914/homebridge-enphase-envoy/files/12268272/homebridge.log)

barthoefs, 2023-08-22:
Still same issue:
```
[8/22/2023, 11:32:37 PM] [homebridge-enphase-envoy] Error: 
    at Brightness.Characteristic.characteristicWarning (/var/lib/homebridge/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:3011:105)
    at Brightness.Characteristic.validateUserInput (/var/lib/homebridge/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:2903:14)
    at Brightness.Characteristic.updateValue (/var/lib/homebridge/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:2328:20)
    at Lightbulb.Service.updateCharacteristic (/var/lib/homebridge/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Service.ts:795:35)
    at /var/lib/homebridge/node_modules/homebridge-enphase-envoy/src/envoydevice.js:2262:26
```

grzegorz914, 2023-08-23:
Please update your plugin, you use 7.3.5, where the fix was with 7.3.7

commit b01b1ba7 (2023-08-06): fix #97
closed 2023-08-23 by grzegorz914