grzegorz914/homebridge-enphase-envoy #98: "On" characteristic never on

chrisjshull, 2023-08-08:
My panels are producing power, so not sure why the "On" characteristic is never on

```
[07/08/2023, 17:42:18] [homebridge-enphase-envoy] Device: 192.168.1.36 Envoy, debug: Production ct: {
  "production": [
    {
      "type": "inverters",
      "activeCount": 13,
      "readingTime": 1691455366,
      "wNow": 1114,
      "whLifetime": 390808
    },
    {
      "type": "eim",
      "activeCount": 0,
      "measurementType": "production",
      "readingTime": 1691455378,
      "wNow": 836.101,
      "whLifetime": 0,
      "varhLeadLifetime": 0,
      "varhLagLifetime": 0,
      "vahLifetime": 0,
      "rmsCurrent": 6.952,
      "rmsVoltage": 242.875,
      "reactPwr": 53.678,
      "apprntPwr": 844.353,
      "pwrFactor": 0.99,
      "whToday": 0,
      "whLastSevenDays": 0,
      "vahToday": 0,
      "varhLeadToday": 0,
      "varhLagToday": 0,
      "lines": [
        {
          "wNow": 417.831,
          "whLifetime": 0,
          "varhLeadLifetime": 0,
          "varhLagLifetime": 0,
          "vahLifetime": 0,
          "rmsCurrent": 3.474,
          "rmsVoltage": 121.44,
          "reactPwr": 26.733,
          "apprntPwr": 422.018,
          "pwrFactor": 0.99,
          "whToday": 0,
          "whLastSevenDays": 0,
          "vahToday": 0,
          "varhLeadToday": 0,
          "varhLagToday": 0
        },
        {
          "wNow": 418.271,
          "whLifetime": 0,
          "varhLeadLifetime": 0,
          "varhLagLifetime": 0,
          "vahLifetime": 0,
          "rmsCurrent": 3.478,
          "rmsVoltage": 121.435,
          "reactPwr": 26.945,
          "apprntPwr": 422.335,
          "pwrFactor": 0.99,
          "whToday": 0,
          "whLastSevenDays": 0,
          "vahToday": 0,
          "varhLeadToday": 0,
          "varhLagToday": 0
        }
      ]
    }
  ],
```

grzegorz914, 2023-08-08:
Full debug log please, enable it in plugin Advanced section and get the whole log please.

chrisjshull, 2023-08-11:
The full logs contain sensitive data, including sensitive data about the other plugins I run. Carefully reviewing a redacting every line of the log is a time consuming process (and must be done very carefully). Can you tell me specific log entries? I could get those to you a lot faster/safer.

grzegorz914, 2023-08-12:
You can enable only debug for Envoy as I described in previous post, no need log from other plugins. The sensitive data can be redacted. The log must contain data direct after plugin restart to accessory publish.

grzegorz914, 2023-08-12:
You can send me the log per email as a file.

grzegorz914, 2023-08-18:
check please with 7.3.6 and let me know

grzegorz914, 2023-08-28:
Any info?

grzegorz914, 2023-08-29:
I think this is solved now, if no free to open again

chrisjshull, 2023-08-29:
Sorry, I've been on vacation, and away from being able to update. I'm still seeing this bug in v7.4.2.
(I don't seem to have permissions to reopen this bug myself)

grzegorz914, 2023-08-29:
Ok, then I need the debug log with this error.

chrisjshull, 2023-08-29:
> You can enable only debug for Envoy as I described in previous post, no need log from other plugins. The sensitive data can be redacted. The log must contain data direct after plugin restart to accessory publish.

I don't explicitly have any other plugins with logging enabled. I also have homebridge *not* in debug mode. However, I still have tons and tons of logs for other plugins interleaved between the envoy plugin logs. And since the envoy logs are multiline, I can't easily use grep to filter to just the relevant logs. Do you have another trick?

ahnungsbefreit, 2023-08-30:
I see the same (lots of debug info in the log althoug debug switched off everywhere) like this:
`[8/30/2023, 8:23:24 AM] [homebridge-enphase-envoy] Device: 192.168.178.132 Envoy-S, debug: Live data: {
  "connection": {
    "mqtt_state": "connected",
    "prov_state": "configured",
    "auth_state": "ok",
    "sc_stream": "enabled",
    "sc_debug": "disabled"
  },
  "meters": {
    "last_update": 1693383818,
    "soc": 41,
    "main_relay_state": 1,
    "gen_relay_state": 5,
    "backup_bat_mode": 1,
    "backup_soc": 0,
    "is_split_phase": 0,
    "phase_count": 3,
    "enc_agg_soc": 41,
    "enc_agg_energy": 1435,
    "acb_agg_soc": 0,
    "acb_agg_energy": 0,
    "pv": {
      "agg_p_mw": 2347486,
      "agg_s_mva": 2391666,
      "agg_p_ph_a_mw": 430912,
      "agg_p_ph_b_mw": 884986,
      "agg_p_ph_c_mw": 1031587,
      "agg_s_ph_a_mva": 464059,
      "agg_s_ph_b_mva": 884986,
      "agg_s_ph_c_mva": 1042620
    },
    "storage": {
      "agg_p_mw": -1266001,
      "agg_s_mva": -1266001,
      "agg_p_ph_a_mw": -1266000,
      "agg_p_ph_b_mw": 0,
      "agg_p_ph_c_mw": 0,
      "agg_s_ph_a_mva": -1266000,
      "agg_s_ph_b_mva": 0,
      "agg_s_ph_c_mva": 0
    },
    "grid": {
      "agg_p_mw": -571474,
      "agg_s_mva": -582626,
      "agg_p_ph_a_mw": 1276242,
      "agg_p_ph_b_mw": -878247,
      "agg_p_ph_c_mw": -969469,
      "agg_s_ph_a_mva": 1303629,
      "agg_s_ph_b_mva": -883736,
      "agg_s_ph_c_mva": -1002519
    },
    "load": {
      "agg_p_mw": 510011,
      "agg_s_mva": 543039,
      "agg_p_ph_a_mw": 441154,
      "agg_p_ph_b_mw": 6739,
      "agg_p_ph_c_mw": 62118,
      "agg_s_ph_a_mva": 501688,
      "agg_s_ph_b_mva": 1250,
      "agg_s_ph_c_mva": 40101
    },
    "generator": {
      "agg_p_mw": 0,
      "agg_s_mva": 0,
      "agg_p_ph_a_mw": 0,
      "agg_p_ph_b_mw": 0,
      "agg_p_ph_c_mw": 0,
      "agg_s_ph_a_mva": 0,
      "agg_s_ph_b_mva": 0,
      "agg_s_ph_c_mva": 0
    }
  },
  "tasks": {
    "task_id": -1834664090,
    "timestamp": 1693322282
  },
  "counters": {
    "main_CfgLoad": 1,
    "main_CfgChanged": 1,
    "main_taskUpdate": 4,
    "MqttClient_publish": 2580,
    "MqttClient_respond": 20,
    "MqttClient_msgarrvd": 10,
    "MqttClient_create": 7,
    "MqttClient_setCallbacks": 7,
    "MqttClient_connect": 7,
    "MqttClient_connect_err": 3,
    "MqttClient_connect_Err": 3,
    "MqttClient_subscribe": 4,
    "SSL_Keys_Create": 7,
    "sc_hdlDataPub": 529446,
    "sc_SendStreamCtrl": 9,
    "sc_SendDemandRspCtrl": 1,
    "rest_Status": 111265
  },
  "dry_contacts": {
    "": {
      "dry_contact_id": "",
      "dry_contact_type": "",
      "dry_contact_load_name": "",
      "dry_contact_status": 3050996
    }
  }
}
`

grzegorz914, 2023-08-30:
I need the whole debug, without this I can not help you. The log can be downloaded using marked button and private data can be redacted.
[image]

ahnungsbefreit, 2023-08-30:
I did a restart for a clean log (attached). Not much added value, it is always just the "live data" debug in the log as shown above.
[homebridge1.log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/12473715/homebridge1.log.txt)

Plugin Version 7.4.1, but with 7.4.2 the behaviour is the same.

ahnungsbefreit, 2023-08-30:
I think I found it in your source (line 2000 in envoydevice.js):
`const debug = !this.enableDebugMode ? this.emit('debug', `Live data: ${JSON.stringify(liveData.data, null, 2)}`) : false;`

I guess the "!" should be removed, no?

grzegorz914, 2023-08-30:
I don't need Info Log, I need debug log, please enable this and get the log:

[image]

ahnungsbefreit, 2023-08-30:
Sorry for the confusion I created, my comment was only related to the fact that I still see a lot of debug info in the log, even with all debug switched off and it is always coming from the live data. So please consider my above comment https://github.com/grzegorz914/homebridge-enphase-envoy/issues/98#issuecomment-1698781008

grzegorz914, 2023-08-30:
> I think I found it in your source (line 2000 in envoydevice.js): `const debug = !this.enableDebugMode ? this.emit('debug', `Live data: ${JSON.stringify(liveData.data, null, 2)}`) : false;`
> 
> I guess the "!" should be removed, no?

Yes, this will be fixed in next update

grzegorz914, 2023-08-30:
I see in your previous info log the power state characteristic is Active(On), which Characteristic On is never On?

ahnungsbefreit, 2023-08-30:
> > I think I found it in your source (line 2000 in envoydevice.js): `const debug = !this.enableDebugMode ? this.emit('debug', `Live data: ${JSON.stringify(liveData.data, null, 2)}`) : false;`
> > I guess the "!" should be removed, no?
> 
> Yes, this will be fixed in next update

With 7.4.3, the "Live data" entries in the log are gone. Thank you!

closed 2023-10-02 by grzegorz914