grzegorz914/homebridge-enphase-envoy #99: Is there a way to control the profile from an automation?

eric-chamberlain-ck, 2023-08-14:
Hello, is there a way to control the charge profile from an automation?

I want full charge mode in the morning, then self-sufficient during peak, as an example.

grzegorz914, 2023-08-15:
Not now but can be integrated in feature.

grzegorz914, 2023-08-29:
Can you post some screenshots and profiles names hot this looks in your app because I don't have battery in my system.

echamberlain, 2023-08-29:
What do you want screenshots of? Enphase app or homebridge logs?

grzegorz914, 2023-08-30:
Enphase app with profiles

echamberlain, 2024-01-15:
Sorry for the delay.  Here's what the app screenshots look like:

[image]
[image]
[image]
[image]

tylerbenson, 2024-01-20:
A while ago I created a separate plugin to do this via the cloud api endpoints, but would rather do it via local endpoints using this plugin.

https://github.com/tylerbenson/homebridge-enphase-battery#readme

I'm not great at Javascript, but let me know if I can help.

grzegorz914, 2024-02-11:
I will start to add this possibility but I will need some one for test because I do not have encharges.

grzegorz914, 2024-02-11:
Can you post 3 debugs logs, one with Full Bakup enabled , one with Savings and Reserve 30% and one wit Self Consumption  and Reserve 75%.

tylerbenson, 2024-02-27:
I think the best way to have this work with homebridge is to have a toggle that changes the profile to full backup when enabled and back to one of the other options when disabled.  In my plugin I implemented it as changing the reserve, but I've run into problems where it doesn't seem to work reliably.

tylerbenson, 2024-03-05:
I ended up refactoring my plugin to switch profiles instead of adjusting the reserve amount.  I think it's working better now.

grzegorz914, 2024-08-08:
Right now I don't see possibility in local api to control encharge profile, if I found solution then will be added in future release. Technically is possible to read and control this data from Web API but I don want to mix this in current app. I will look into this in future.

tylerbenson, 2024-08-08:
I ended up switching over to home assistant.  The enphase integration there does allow for profile switching and I believe it is all local. Might be worth looking at their code for comparison.

closed 2024-08-09 by grzegorz914

tylerbenson, 2024-08-10:
Glad you were able to figure it out!

echamberlain, 2024-08-11:
Thanks. I'll test it out when I get back from traveling.