grzegorz914/homebridge-enphase-envoy #100: Total Power Produced Contact sensor??

ipathi1, 2023-08-24:
Hey! Loved your work with this plug-in. Is it possible to code a Total Power Produced contact sensor? If I set a value and it triggers, I can automate another device (whether there was enough sunlight to heat my Solar Water heater and accordingly turn on my geyser if there wasn’t enough sunlight). A daily value reset option would be great with this. Thank you.

grzegorz914, 2023-08-24:
Total Power Produced or Max Power Produced? I don’t quite understand what is total power produced?
Max Power Produced sensor is already there.

ipathi1, 2023-08-25:
Correct me if I'm wrong, but Power Peak Detected is the one circled in Blue in my screenshot right? 

If so, I want value for power Produced for that day.(Circled in white)

[image]

grzegorz914, 2023-08-25:
Yes, Peak power is the value in blue circle, in white is energy production total at this time

ipathi1, 2023-08-25:
Apologies, my bad.
Total Energy Produced for the day is what I am looking for.

grzegorz914, 2023-08-25:
I will add this possibility today

ipathi1, 2023-08-25:
Oh wow. Thank you for your time.

grzegorz914, 2023-08-25:
Done in 7.4.0, please test and let me know.

closed 2023-08-27 by grzegorz914