grzegorz914/homebridge-enphase-envoy #101: Device: undefined Envoy-S

Kermarc44, 2023-08-25:
Hello,

It's seems that the plugin is able to connect but that the envoy system is unknown : 
ERRORDevice: undefined Envoy-S, Update home data error: Error: self signed certificate. Reconnect in 15s.

Config : 
{
    "accessories": [],
    "platforms": [
        {
            "platform": "enphaseEnvoy",
            "devices": [
                {
                    "name": "Envoy-S",
                    "enableDebugMode": true,
                    "refreshInterval": 120,
                    "powerProductionMaxDetected": 1,
                    "energyProductionLifetimeOffset": 0,
                    "powerConsumptionTotalMaxDetected": 1,
                    "energyConsumptionTotalLifetimeOffset": 0,
                    "powerConsumptionNetMaxDetected": 1,
                    "energyConsumptionNetLifetimeOffset": 0,
                    "disableLogInfo": false
                }
            ]
        }
    ]
}

Debug log : 

25/08/2023, 16:21:35Enphase Envoy BridgeenphaseEnvoyDevice: undefined Envoy-S, debug: Start.
25/08/2023, 16:21:35Enphase Envoy BridgeenphaseEnvoyDevice: undefined Envoy-S, debug: Requesting info.
25/08/2023, 16:21:35Enphase Envoy BridgeenphaseEnvoyDevice: undefined Envoy-S, debug: Info: "<?xml version='1.0' encoding='UTF-8'?>\n<envoy_info>\n  <time>1692973265</time>\n  <device>\n    <sn>122143077347</sn>\n    <pn>800-00654-r06</pn>\n    <software>D7.6.175</software>\n    <euaid>4c8675</euaid>\n    <seqnum>0</seqnum>\n    <apiver>1</apiver>\n    <imeter>true</imeter>\n  </device>\n  <web-tokens>true</web-tokens>\n  <package name='rootfs'>\n    <pn>500-00001-r01</pn>\n    <version>02.00.00</version>\n    <build>1210</build>\n  </package>\n  <package name='kernel'>\n    <pn>500-00011-r02</pn>\n    <version>04.04.225</version>\n    <build>3eb4d3</build>\n  </package>\n  <package name='boot'>\n    <pn>590-00019-r01</pn>\n    <version>02.00.01</version>\n    <build>1f421b</build>\n  </package>\n  <package name='app'>\n    <pn>500-00002-r01</pn>\n    <version>07.06.175</version>\n    <build>f79c8d</build>\n  </package>\n  <package name='devimg'>\n    <pn>500-00005-r01</pn>\n    <version>01.02.293</version>\n    <build>9cf065</build>\n  </package>\n  <package name='geo'>\n    <pn>500-00008-r01</pn>\n    <version>02.01.24</version>\n    <build>a74d96</build>\n  </package>\n  <package name='backbone'>\n    <pn>500-00010-r01</pn>\n    <version>07.00.20</version>\n    <build>176d57</build>\n  </package>\n  <package name='meter'>\n    <pn>500-00013-r01</pn>\n    <version>03.02.08</version>\n    <build>eaa252</build>\n  </package>\n  <package name='agf'>\n    <pn>500-00012-r01</pn>\n    <version>02.02.00</version>\n    <build>40061a</build>\n  </package>\n  <package name='security'>\n    <pn>500-00016-r01</pn>\n    <version>02.00.00</version>\n    <build>54a6dc</build>\n  </package>\n  <package name='full'>\n    <pn>500-00001-r01</pn>\n    <version>02.00.00</version>\n    <build>1210</build>\n  </package>\n  <build_info>\n    <build_id>ec2-user-envoy_uber-pkg_master:pkg-Jun-22-23-18:55:22</build_id>\n    <build_time_gmt>1687460237</build_time_gmt>\n    <release_ver>02.00.4238</release_ver>\n    <release_stage>700-GA</release_stage>\n  </build_info>\n</envoy_info>\n"
25/08/2023, 16:21:35Enphase Envoy BridgeenphaseEnvoyDevice: undefined Envoy-S, debug: Parse info: {
  "envoy_info": {
    "time": 1692973265,
    "device": {
      "sn": 122143077347,
      "pn": "800-00654-r06",
      "software": "D7.6.175",
      "euaid": "4c8675",
      "seqnum": 0,
      "apiver": 1,
      "imeter": true
    },
    "web-tokens": true,
    "package": [
      {
        "pn": "500-00001-r01",
        "version": "02.00.00",
        "build": 1210,
        "@_name": "rootfs"
      },
      {
        "pn": "500-00011-r02",
        "version": "04.04.225",
        "build": "3eb4d3",
        "@_name": "kernel"
      },
      {
        "pn": "590-00019-r01",
        "version": "02.00.01",
        "build": "1f421b",
        "@_name": "boot"
      },
      {
        "pn": "500-00002-r01",
        "version": "07.06.175",
        "build": "f79c8d",
        "@_name": "app"
      },
      {
        "pn": "500-00005-r01",
        "version": "01.02.293",
        "build": "9cf065",
        "@_name": "devimg"
      },
      {
        "pn": "500-00008-r01",
        "version": "02.01.24",
        "build": "a74d96",
        "@_name": "geo"
      },
      {
        "pn": "500-00010-r01",
        "version": "07.00.20",
        "build": "176d57",
        "@_name": "backbone"
      },
      {
        "pn": "500-00013-r01",
        "version": "03.02.08",
        "build": "eaa252",
        "@_name": "meter"
      },
      {
        "pn": "500-00012-r01",
        "version": "02.02.00",
        "build": "40061a",
        "@_name": "agf"
      },
      {
        "pn": "500-00016-r01",
        "version": "02.00.00",
        "build": "54a6dc",
        "@_name": "security"
      },
      {
        "pn": "500-00001-r01",
        "version": "02.00.00",
        "build": 1210,
        "@_name": "full"
      }
    ],
    "build_info": {
      "build_id": "ec2-user-envoy_uber-pkg_master:pkg-Jun-22-23-18:55:22",
      "build_time_gmt": 1687460237,
      "release_ver": "02.00.4238",
      "release_stage": "700-GA"
    }
  }
}
25/08/2023, 16:21:35Enphase Envoy BridgeenphaseEnvoyDevice: undefined Envoy-S, debug: Envoy password: xxxxx
25/08/2023, 16:21:35Enphase Envoy BridgeenphaseEnvoyDevice: undefined Envoy-S, debug: Installer password: xxxx
25/08/2023, 16:21:35Enphase Envoy BridgeenphaseEnvoyDevice: undefined Envoy-S, debug: Requesting home.
25/08/2023, 16:21:35Enphase Envoy BridgeenphaseEnvoyERRORDevice: undefined Envoy-S, Update home data error: Error: self signed certificate. Reconnect in 15s.

grzegorz914, 2023-08-25:
What firmware use Your Envoy? if 7.x.x enable support for it in plugin settings advanced settings section and let me know

Kermarc44, 2023-08-25:
D7.6.175

grzegorz914, 2023-08-25:
Please enable support for fw 7.x.x, go to plugin settings Advanced Settings >> Authorization and let me know

Kermarc44, 2023-08-27:
Thanks it's working now.

closed 2023-08-27 by grzegorz914