grzegorz914/homebridge-enphase-envoy #103: [9/13/2023, 6:44:02 AM] [enphaseEnvoy] Device: 192.168.1.41 Envoy-S, Update home data error: Error: Parse Error: Content-Length can't be present with Transfer-Encoding. Reconnect in 15s.

barthoefs, 2023-09-13:
Hello,
I always get this message the last days (weeks):
[9/13/2023, 6:44:02 AM] [enphaseEnvoy] Device: 192.168.1.41 Envoy-S, Update home data error: Error: Parse Error: Content-Length can't be present with Transfer-Encoding. Reconnect in 15s.

I don't get any info in the EVE app for example.

barthoefs, 2023-09-13:
```
[9/13/2023, 6:51:03 AM] [enphaseEnvoy] Device: 192.168.1.41 Envoy-S, debug: Start.
[9/13/2023, 6:51:03 AM] [enphaseEnvoy] Device: 192.168.1.41 Envoy-S, debug: Requesting info.
[9/13/2023, 6:51:03 AM] [enphaseEnvoy] Device: 192.168.1.41 Envoy-S, debug: Info: "<?xml version='1.0' encoding='UTF-8'?>\n<envoy_info>\n  <time>1694587814</time>\n  <device>\n    <sn>121921106049</sn>\n    <pn>800-00553-r02</pn>\n    <software>D5.0.62</software>\n    <euaid>4c8675</euaid>\n    <seqnum>0</seqnum>\n    <apiver>1</apiver>\n    <imeter>false</imeter>\n  </device>\n  <package name='rootfs'>\n    <pn>500-00001-r01</pn>\n    <version>02.00.00</version>\n    <build>953</build>\n  </package>\n  <package name='kernel'>\n    <pn>500-00011-r01</pn>\n    <version>04.01.15</version>\n    <build>8f3564</build>\n  </package>\n  <package name='boot'>\n    <pn>590-00018-r01</pn>\n    <version>02.00.01</version>\n    <build>426697</build>\n  </package>\n  <package name='app'>\n    <pn>500-00002-r01</pn>\n    <version>05.00.62</version>\n    <build>5e57a9</build>\n  </package>\n  <package name='devimg'>\n    <pn>500-00005-r01</pn>\n    <version>01.02.346</version>\n    <build>b1b724</build>\n  </package>\n  <package name='geo'>\n    <pn>500-00008-r01</pn>\n    <version>02.01.22</version>\n    <build>2faa48</build>\n  </package>\n  <package name='backbone'>\n    <pn>500-00010-r01</pn>\n    <version>05.00.02</version>\n    <build>4fe435</build>\n  </package>\n  <package name='meter'>\n    <pn>500-00013-r01</pn>\n    <version>03.02.07</version>\n    <build>276642</build>\n  </package>\n  <package name='agf'>\n    <pn>500-00012-r01</pn>\n    <version>02.02.00</version>\n    <build>a81a58</build>\n  </package>\n  <package name='security'>\n    <pn>500-00016-r01</pn>\n    <version>02.00.00</version>\n    <build>54a6dc</build>\n  </package>\n  <package name='full'>\n    <pn>500-00001-r01</pn>\n    <version>02.00.00</version>\n    <build>953</build>\n  </package>\n  <build_info>\n    <build_id>jenkins-envoy_uber-pkg_5.0-build:pkg-Jul-14-21-16:11:44</build_id>\n    <build_time_gmt>1626297189</build_time_gmt>\n  </build_info>\n</envoy_info>\n"
[9/13/2023, 6:51:03 AM] [enphaseEnvoy] Device: 192.168.1.41 Envoy-S, debug: Parse info: {
  "envoy_info": {
    "time": 1694587814,
    "device": {
      "sn": 121921106049,
      "pn": "800-00553-r02",
      "software": "D5.0.62",
      "euaid": "4c8675",
      "seqnum": 0,
      "apiver": 1,
      "imeter": false
    },
    "package": [
      {
        "pn": "500-00001-r01",
        "version": "02.00.00",
        "build": 953,
        "@_name": "rootfs"
      },
      {
        "pn": "500-00011-r01",
        "version": "04.01.15",
        "build": "8f3564",
        "@_name": "kernel"
      },
      {
        "pn": "590-00018-r01",
        "version": "02.00.01",
        "build": 426697,
        "@_name": "boot"
      },
      {
        "pn": "500-00002-r01",
        "version": "05.00.62",
        "build": "5e57a9",
        "@_name": "app"
      },
      {
        "pn": "500-00005-r01",
        "version": "01.02.346",
        "build": "b1b724",
        "@_name": "devimg"
      },
      {
        "pn": "500-00008-r01",
        "version": "02.01.22",
        "build": "2faa48",
        "@_name": "geo"
      },
      {
        "pn": "500-00010-r01",
        "version": "05.00.02",
        "build": "4fe435",
        "@_name": "backbone"
      },
      {
        "pn": "500-00013-r01",
        "version": "03.02.07",
        "build": 276642,
        "@_name": "meter"
      },
      {
        "pn": "500-00012-r01",
        "version": "02.02.00",
        "build": "a81a58",
        "@_name": "agf"
      },
      {
        "pn": "500-00016-r01",
        "version": "02.00.00",
        "build": "54a6dc",
        "@_name": "security"
      },
      {
        "pn": "500-00001-r01",
        "version": "02.00.00",
        "build": 953,
        "@_name": "full"
      }
    ],
    "build_info": {
      "build_id": "jenkins-envoy_uber-pkg_5.0-build:pkg-Jul-14-21-16:11:44",
      "build_time_gmt": 1626297189
    }
  }
}
[9/13/2023, 6:51:03 AM] [enphaseEnvoy] Device: 192.168.1.41 Envoy-S, debug: Envoy password: xxxx
[9/13/2023, 6:51:03 AM] [enphaseEnvoy] Device: 192.168.1.41 Envoy-S, debug: Installer password: xxxx
[9/13/2023, 6:51:03 AM] [enphaseEnvoy] Device: 192.168.1.41 Envoy-S, debug: Requesting home.
[9/13/2023, 6:51:06 AM] [enphaseEnvoy] Device: 192.168.1.41 Envoy-S, Update home data error: Error: Parse Error: Content-Length can't be present with Transfer-Encoding. Reconnect in 15s.
```

grzegorz914, 2023-09-13:
Try reboot your envov

barthoefs, 2023-09-13:
> Try reboot your envov

Rebooting did the trick. Thx