grzegorz914/homebridge-enphase-envoy #105: Detect grid status

eleventhaus, 2023-10-01:
Is there a way to detect when the system goes off grid or when the power from the grid stops?

It would be handy to be able to run an automation if the power from the grid goes out.

grzegorz914, 2023-10-01:
We can detect enpower states: no grid, open, and closed. If this is this what you need then it can be added as a sensor.

grzegorz914, 2023-10-01:
@eleventhaus 
in this case I need for you 2 debug logs, one if your device is ON Grid andond one if is OFF Grid, to enable debug please go to plugin settings>>advanced settings>>system>>debug enable it. Restart and get the whole log.

grzegorz914, 2023-10-01:
I have pushed 7.5.0-beta
- added *Enpower Grid State* contact sensor for monitoring,
- added *Encharge Grid State* contact sensor for monitoring,
- added *Solar Grid State* contact sensor for monitoring

all configured in plugin settings Advanced Settings>>Ensemble section

please test and let me know

eleventhaus, 2023-10-01:
Looking good! Thank you

grzegorz914, 2023-10-02:
Please provide the debug log too.

grzegorz914, 2023-10-02:
continue please in #106

closed 2023-10-02 by grzegorz914

eleventhaus, 2024-02-10:
> We can detect enpower states: no grid, open, and closed

Will the enpower state show "off grid" (meaning power grid from the city is down)? or will it only show open/closed?

grzegorz914, 2024-02-10:
@eleventhaus i don’t know because I don’t have Ensemble/Enpower, if You send me the debug log during this mode then I can check it.

grzegorz914, 2024-02-10:
@eleventhaus read this: https://support.enphase.com/s/article/when-will-my-enphase-energy-system-go-off-grid then put your system in Grid-Off and send me the debug log.

grzegorz914, 2024-02-10:
@eleventhaus  added in 7.11.0