grzegorz914/homebridge-enphase-envoy #108: Self-Signed Certificated

samueltalbot, 2023-10-24:
Hello there!

I've recently had a system installed recently and have the following issue:

[enphaseEnvoy] Device: envoy.local Enphase Solar, Update home data error: Error: self-signed certificate. Reconnect in 15s.

When I log into envoy.local, I get the following error: "Sorry, you are not authorized to view that page".

After dismissing the error, I can log in using my Enphase Enlighten username and password.

Similarly, when I navigate to the envoy IP address (found on my modem's settings), I get the same "not authorized" message.

I suspect these issues are caused by not having an access token? (I'm new to this world... please forgive my ignorance!)

Any ideas how I can still add my system to Homekit?

grzegorz914, 2023-10-24:
Just enable support fot envoy firmware 7.x, fill the envoy serial number and the account data save and restart the plugin. After this add it to the HomeKit.

grzegorz914, 2023-11-19:
Still the issue exist on your side?

closed 2023-11-19 by grzegorz914