grzegorz914/homebridge-enphase-envoy #112: Error 401 after power cut

zoic21, 2023-11-29:
Hello,

To begin thank for your plugin and your time on it. I find maybe an issue, when there is a power cut, after Envoy restart token is invalid (401 error on hombridge plugin log) and I need to restart plugin to make it work again.

Maybe it's possible when there is a 401 unauthorized to test a renew of token and retry ?

Thank in advance.

zoic21, 2023-11-29:
Hello, 

Yes of course : 

```
2023/11/29 08:30:01	stdout	[37m[11/29/2023, 8:30:01 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update production data error: AxiosError: Request failed with status code 401. Trying again in: in: 30 sec.[39m
2023/11/29 08:29:37	stdout	[37m[11/29/2023, 8:29:37 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update meters data error: AxiosError: Request failed with status code 401. Trying again in: 30 sec.[39m
2023/11/29 08:29:31	stdout	[37m[11/29/2023, 8:29:31 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update production data error: AxiosError: Request failed with status code 401. Trying again in: in: 30 sec.[39m
2023/11/29 08:29:31	stdout	[37m[11/29/2023, 8:29:31 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update inventory data error: AxiosError: Request failed with status code 401. Trying again in: 60 sec.[39m
2023/11/29 08:29:22	stdout	[37m[11/29/2023, 8:29:22 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update microinverters data error: AxiosError: Request failed with status code 401. Trying again in: 70 sec.[39m
2023/11/29 08:29:07	stdout	[37m[11/29/2023, 8:29:07 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update meters data error: AxiosError: Request failed with status code 401. Trying again in: 30 sec.[39m
2023/11/29 08:29:01	stdout	[37m[11/29/2023, 8:29:01 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update production data error: AxiosError: Request failed with status code 401. Trying again in: in: 30 sec.[39m
2023/11/29 08:28:37	stdout	[37m[11/29/2023, 8:28:37 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update meters data error: AxiosError: Request failed with status code 401. Trying again in: 30 sec.[39m
2023/11/29 08:28:31	stdout	[37m[11/29/2023, 8:28:31 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update production data error: AxiosError: Request failed with status code 401. Trying again in: in: 30 sec.[39m
2023/11/29 08:28:29	stdout	[37m[11/29/2023, 8:28:29 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update inventory data error: AxiosError: Request failed with status code 401. Trying again in: 60 sec.[39m
2023/11/29 08:28:12	stdout	[37m[11/29/2023, 8:28:12 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update microinverters data error: AxiosError: Request failed with status code 401. Trying again in: 70 sec.[39m
2023/11/29 08:28:07	stdout	[37m[11/29/2023, 8:28:07 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update meters data error: AxiosError: Request failed with status code 401. Trying again in: 30 sec.[39m
2023/11/29 08:28:01	stdout	[37m[11/29/2023, 8:28:01 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update production data error: AxiosError: Request failed with status code 401. Trying again in: in: 30 sec.[39m
2023/11/29 08:27:37	stdout	[37m[11/29/2023, 8:27:37 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update meters data error: AxiosError: Request failed with status code 401. Trying again in: 30 sec.[39m
2023/11/29 08:27:31	stdout	[37m[11/29/2023, 8:27:31 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update production data error: AxiosError: Request failed with status code 401. Trying again in: in: 30 sec.[39m
2023/11/29 08:27:28	stdout	[37m[11/29/2023, 8:27:28 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update inventory data error: AxiosError: Request failed with status code 401. Trying again in: 60 sec.[39m
2023/11/29 08:27:07	stdout	[37m[11/29/2023, 8:27:07 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update meters data error: AxiosError: Request failed with status code 401. Trying again in: 30 sec.[39m
2023/11/29 08:27:02	stdout	[37m[11/29/2023, 8:27:02 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update microinverters data error: AxiosError: Request failed with status code 401. Trying again in: 70 sec.[39m
2023/11/29 08:27:01	stdout	[37m[11/29/2023, 8:27:01 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update production data error: AxiosError: Request failed with status code 401. Trying again in: in: 30 sec.[39m
2023/11/29 08:26:37	stdout	[37m[11/29/2023, 8:26:37 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update meters data error: AxiosError: Request failed with status code 401. Trying again in: 30 sec.[39m
2023/11/29 08:26:31	stdout	[37m[11/29/2023, 8:26:31 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update production data error: AxiosError: Request failed with status code 401. Trying again in: in: 30 sec.[39m
2023/11/29 08:26:27	stdout	[37m[11/29/2023, 8:26:27 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update inventory data error: AxiosError: Request failed with status code 401. Trying again in: 60 sec.[39m
[... cut ...] 0 sec.[39m
2023/11/29 08:24:31	stdout	[37m[11/29/2023, 8:24:31 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update production data error: AxiosError: Request failed with status code 401. Trying again in: in: 30 sec.[39m
2023/11/29 08:24:24	stdout	[37m[11/29/2023, 8:24:24 AM] [39m[36m[homebridge-enphase-envoy][39m [31mDevice: 192.168.1.46 Envoy, Update inventory data error: AxiosError: Request failed with status code 401. Trying again in: 60 sec.[39m
```

grzegorz914, 2023-11-29:
try 7.7.3-beta.0 and let me know

zoic21, 2023-11-29:
Hello,

No problem, just where can found this version ? I check on this git but I don't any new commit or release...

grzegorz914, 2023-11-29:
In config ui select install another version.

grzegorz914, 2023-11-29:
or just in terminal

`npm install -g homebridge-enphase-envoy@v7.7.3-beta.1`

zoic21, 2023-11-29:
Ok thanks it's seem ok but I am not sure in the way you want : 

```
[11/29/2023, 6:20:03 PM] [homebridge-enphase-envoy] Device: 192.168.1.46 Envoy, Update meters data error: AxiosError: Request failed with status code 401. Trying again in: 30 sec.

/homebridge/node_modules/homebridge-enphase-envoy/index.js:53
            const validateToken = error.response.status == 401 ? envoyDevice.start() : false;
                                                 ^
TypeError: Cannot read properties of undefined (reading 'status')
    at EnvoyDevice.<anonymous> (/homebridge/node_modules/homebridge-enphase-envoy/index.js:53:50)
    at EnvoyDevice.emit (node:events:517:28)
    at EnvoyDevice.updateMeters (/homebridge/node_modules/homebridge-enphase-envoy/src/envoydevice.js:440:18)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
[11/29/2023, 6:20:03 PM] [homebridge-enphase-envoy] Child bridge process ended
[11/29/2023, 6:20:03 PM] [homebridge-enphase-envoy] Process Ended. Code: 1, Signal: null
[11/29/2023, 6:20:10 PM] [homebridge-enphase-envoy] Restarting Process...
[11/29/2023, 6:20:11 PM] [homebridge-enphase-envoy] Launched child bridge with PID 1499
[11/29/2023, 6:20:11 PM] Registering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[11/29/2023, 6:20:11 PM] [homebridge-enphase-envoy] Loaded homebridge-enphase-envoy v7.7.3-beta.1 child bridge successfully
[11/29/2023, 6:20:11 PM] Loaded 0 cached accessories from cachedAccessories.0EA720C1C3ED.
[11/29/2023, 6:20:11 PM] Homebridge v1.7.0 (HAP v0.11.1) (homebridge-enphase-envoy) is running on port 53335.
[11/29/2023, 6:20:11 PM] [homebridge-enphase-envoy] Device: 192.168.1.46 Envoy, MQTT Connected.
```

grzegorz914, 2023-11-29:
Fixed in beta 2

closed 2023-11-29 by grzegorz914

zoic21, 2023-11-29:
Perfect it's work now. Thank a lot for your support and fast correction.

grzegorz914, 2023-11-29:
OK, try 7.7.4-beta.0

zoic21, 2023-11-29:
Ok I start the test but I think it's not you but a bug in envoy firmware, after 10min all it's ok....

zoic21, 2023-11-29:
It's good now on log, still errors on the values ​​during the few minutes following the restart but I think it is due to the send itself.

grzegorz914, 2023-11-30:
im pushed 7.7.4

commit 3c21bd9b (2023-11-29): fix #112