grzegorz914/homebridge-enphase-envoy #114: Enphase System, Get backbone app data error: Error: self-signed certificate.  Reconnect in 15s

CaptainPic, 2023-12-03:
Looked like I was doing everything right to set up this plug in but I get this recurring error when I restart Homebridge.  I'm also a bit confused by not seeing the firmware V7.X.X.  It's a newly commissioned system in the US using the latest Controller 3 and 5P batteries.   The IQ Gateway shows Firmware: D8.2.62  and the IQ system controller 3 is firmware 3.0.5509_rel/3.11 from the Enlighten App.   Is there any reason my system should not work with your plug-in?

grzegorz914, 2023-12-04:
Are you enabled support for Firmware 7.x.x in plugin config?

grzegorz914, 2023-12-04:
Any info?

CaptainPic, 2023-12-06:
Yes- selecting the 7.x.x. option (Didn’t realize it was your software version) enabled it to load into homebridge without problems.  I see a few Enphase accessories in yhr  homebridge  accessory view now and have added the bridge to Apple Home.  Give me a day or two to play with this some more as I’m not seeing the accessories as switches to use as triggers in Apple Home or Eve yet.

Bottom line what I’m trying to do is set up an automation that will kick down the HVAC and other big energy consumers when the grid is down and the batteries and solar are below a threshold in order  to preserve the battery life for more essential loads.  Kind of a software defined load shedding solution.

Thanks for the help I may have another question or two but I need to get a bit more familiar with it. First.

Jack

grzegorz914, 2023-12-06:
You need to add the accessory itself to the HomeKit app not the bridge only.  In readme is all described what is displayed in native HomeKit app.

CaptainPic, 2023-12-10:
Hello,

Still not quite getting there although for a brief time I did see an Enphase accessory in HomeKit that looked like a contact sensor but didn’t appear to have any functionality.

I did an uninstall in Homebridge, removed the Enphase bridge from Home Kit Home Settings  and reinstall of 7.7.9 in Homebridge.  Setup configuration just included checking the 7.7.x box and adding system Serial number and enlighten credentials.  – nothing else for now.

I get these two errors in the Homebridge logs after the restart

***@***.***

***@***.***
Repeating….

I realized I didn’t set it as a child bridge so I did that and restarted Homebrigde – same error
I’m away from the house and don’t know if I can add Enhphase to the home over the vpn right now, but when I did this earlier the bridge added but not accories are visible.   I don’t understand your comment about adding the accessories separately from the Hub nor see where this is explained in the readme.  For all other plug-ins the accessories seem to get added automatically as part of adding the plug-in to Homekit.
Please let know if something doesn’t look right here.

Thank you
Jack

grzegorz914, 2023-12-11:
First of all if you get error please go to plugin settings advanced settings>>system>>debug and enable this option. After this restart plugin and post the whole log here. Please also post your config without credential data.

CaptainPic, 2024-01-09:
Taking another crack at this.   I believe I understand the manual add accessory process once I have a Homekit code from within Hombridge to add, however, I’m not to the point of getting a clean install in Homebridge.

See attached image for the recurring error message that show up in HB status log.     II did a completely clean install unpairing/removing several prior bridges I saw under “unpair/remove bridges settings area of HB.  I also only set up the Envoy Serial number / Enlighten user and Enhlighten password in the configuration of the plug in and left everything else default.  I’m on the latest release of everything  Homebridge 3.1.1; ephase-envoy plug-in 7.9.2-latest.

I do get the code to add the bridge to Homekit which adds but still get the warnings and do not see any accessories to add.   Please let me know if you something wrong here as I can’t get to the point it installs and gives me an accessory code.

Thank you

Jack Denenberg

***@***.***

The bridge adds to homkit but I see no accessories or codes for them.

[A screenshot of a qr code  Description automatically generated]

About adding to HomeKit https://github.com/grzegorz914/homebridge-enphase-envoy/wiki#adding-to-homekit

grzegorz914, 2024-01-09:
Where you post the debug log? I need this to check what happens.

CaptainPic, 2024-01-09:
homebridge.log (1).txt.crdownload

Is this the log file that helps?

grzegorz914, 2024-01-10:
which file

CaptainPic, 2024-01-10:
[homebridge.log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/13891506/homebridge.log.txt)

I think I attahed it properly this time

grzegorz914, 2024-01-10:
To debug my plugin You need go to -login settings Advanced Settings>>>System>>>Debug enable it restart plugin then send me the log.

CaptainPic, 2024-01-11:
Thank you for your patience here.  I don’t specifically see Advanced settings in the latest Homebridge update, however, I turned on the Debug -D mode and downloaded the latest log file – Looks the same as the last one I sent so ple ase advise if I’m not getting the debug log properly.

***@***.***

racteristic'). Trying again in 15s.
[1/11/2024, 5:08:52 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:09:12 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:09:32 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:09:35 AM] [HomeConnect] [Refrigerator] Door open
[1/11/2024, 5:09:51 AM] [HomeConnect] [Refrigerator] Door closed
[1/11/2024, 5:09:53 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:10:13 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:10:33 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:10:54 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:11:17 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:11:38 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:11:57 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:12:18 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:12:40 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:13:00 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:13:18 AM] [HomeConnect] Found 8 appliances (0 added, 0 removed)
[1/11/2024, 5:13:19 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:13:39 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:14:00 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:14:23 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:14:46 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:15:08 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:15:27 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:15:47 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:16:09 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:16:29 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 5:16:49 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[... cut ...]
[1/11/2024, 7:01:44 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 7:02:04 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[1/11/2024, 7:02:24 AM] [enphaseEnvoy] Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in

grzegorz914, 2024-01-12:
You need go to plugin settings not to Homebridge settings.

CaptainPic, 2024-01-12:
Got it – should have looked for the debug option there too. Sorry

grzegorz914, 2024-01-13:
And, any debug log?

grzegorz914, 2024-01-15:
Fixed in 7.9.3

CaptainPic, 2024-01-23:
Thank You!    I'm getting the plug-in to work  and becoming familiar with the accessories.   I do still get the error if I enable the Ensemble Status so I turned that off.  I'm thinking that's where the Grid Status switch appears which is really the high order bit in terms of making a decision to start to shed loads.  If that's the case I can get the logs there but for now I'm still working through what can be done using Controller for HomeKit.   Thanks again - This is probably the most interesting and practical information for automating load shedding when you have a solar/battery system when the utility grid may have issues.

grzegorz914, 2024-01-23:
Im happy to hear that is working now. If you can post the debug log with enabled ensemble status then will be very helpful for me to fix potentially existed bug.

CaptainPic, 2024-02-07:
Been busy moving in to the new place but got some time to get the debug log when I enable the Ensemble Status and enable Empower Grid State Sensor.  Here is the Homebidge and Enphse Plug in log files.  

 I haven't looked into the Apple Home App of Controller in detail.  It looks like I need to re-add the Enphase 80DC accessory in HomeKit as I'm getting No Response - Is that correct that it needs to be removed and re-added when I change the plug in configuration?

I wanted to pass on a couple thoughts and ideas.  I see in Controller for HomeKit how to do the automations.  My focus is on creating a visual load shedding panel based on some logic on Solar Power and Battery Level in combination with the Grid Status which I'm assuming will be surfaced as a contact sensor from the Envoy detail in Controller.  

The other item that would be next high value feature in my mind would be to get information on the nw Enphase IQ EV charger which I have in my system.  The first use case for the automation in my mind is just to have a status reminder if the car is not connected in the evening.  Perhaps also how much energy the car pulled but the Enphase App has most of the settings to show this information.  However it doesn't give me a notification if I forgot to plug the car in!

Thanks for the great work in this area! - If you need anything else let me know.   

Jack

[homebridge-enphase-envoy.log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/14202707/homebridge-enphase-envoy.log.txt)
[homebridge.log (3).txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/14202708/homebridge.log.3.txt)

grzegorz914, 2024-02-08:
please install `7.9.8-beta.0 `and let me know, im fixed ensemble status and more other cases

CaptainPic, 2024-02-08:
Still getting the error in the logs.   Here is the log using 7.9.8-beta10

[Uploading homebridge-enphase-envoy.log (4).txt…]()

grzegorz914, 2024-02-08:
Log not available

grzegorz914, 2024-02-09:
@CaptainPic please install beta 11 and get me the log again.

CaptainPic, 2024-02-09:
Looks like the plug-in log has no errors using beta 11 based on the fact I'm not seeing anything in red anymore.  Here is the log - When I get home later I'll add it to HomeKit.  
[homebridge-enphase-envoy.log (1).txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/14226957/homebridge-enphase-envoy.log.1.txt)

grzegorz914, 2024-02-09:
Many thanks for your help, 7.10.0 released with all fixes. In your log now is all OK, the accessory can be added to the Home app.

closed 2024-02-09 by grzegorz914

CaptainPic, 2024-02-10:
Error has come back with 7.11.0.

I had it almost there at least the accessories for Envoy Grid Contact sensors as well as the solar and battery levels were showing up in Homebridge as accessories and then I saw you pushed out 7.11.0 , so I installed that before moving on to the add to Homekit step.  Unfortunately I’m seeing the errors again with 7.11.0 and the accessories are no longer in Homebridge anymore.

Logs attached

[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318020636, producing: No
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318020636, communicating: No
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318020636, provisioned: No
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318020636, operating: No
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318020636, status: Failure to report
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318020636, firmware: 521-00005-r06-v02.61.01
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318020636, last report: 2/9/2024, 3:26:42 PM
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318011164, last power: 0 W
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318011164, peak power: 329 W
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318011164, producing: No
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318011164, communicating: No
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318011164, provisioned: No
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318011164, operating: No
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318011164, status: Failure to report
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318011164, firmware: 521-00005-r06-v02.61.01
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318011164, last report: 2/9/2024, 3:21:06 PM
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318010148, last power: 0 W
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318010148, peak power: 329 W
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318010148, producing: No
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318010148, communicating: No
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318010148, provisioned: No
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318010148, operating: No
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318010148, status: Failure to report
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318010148, firmware: 521-00005-r06-v02.61.01
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318010148, last report: 2/9/2024, 3:20:11 PM
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318020081, last power: 0 W
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318020081, peak power: 330 W
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318020081, producing: No
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318020081, communicating: No
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318020081, provisioned: No
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318020081, operating: No
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318020081, status: Failure to report
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318020081, firmware: 521-00005-r06-v02.61.01
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318020081, last report: 2/9/2024, 3:26:49 PM
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318021302, last power: 0 W
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318021302, peak power: 329 W
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318021302, producing: No
[2/9/2024, 9:02:16 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318021302, communicating: No
[... cut ...]
[2/10/2024, 12:07:13 AM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318010389, producing: No
[2/10/2024, 12:07:13 AM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318010389, communicating: No
[2/10/2024, 12:07:13 AM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318010389, provisioned: No
[2/10/2024, 12:07:13 AM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Microinverter: 202318010389, operating: No
[2/10/2024, 12:07:13 AM] [home

grzegorz914, 2024-02-10:
@CaptainPic  i don't see any error in log

grzegorz914, 2024-02-10:
@CaptainPic  7.11.2 released with encharge sensor fix

CaptainPic, 2024-02-10:
Going to give it a go now

CaptainPic, 2024-02-10:
Going to attach the file in Github rather than email - 7.11.5 still giving an error about circular reference 
[homebridge-enphase-envoy.log (5).txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/14230516/homebridge-enphase-envoy.log.5.txt)

CaptainPic, 2024-02-10:
Got error in 7.11.2 and 7.11.5

Device: 192.168.170.14 Enphase, Requesting inventory error: TypeError: Converting circular structure to JSON
    --> starting at object with constructor 'TLSSocket'
    |     property '_httpMessage' -> object with constructor 'ClientRequest'
    --- property 'socket' closes the circle. Trying again in 15s.

Attached logs in Gifthub comment – hopefully they come through

Thanks for your help

Jack

grzegorz914, 2024-02-10:
Fixed in 7.11.6

CaptainPic, 2024-02-11:
Still seeing errors in 7.11.6.     Here's the Enhphase Plug in log
[homebridge-enphase-envoy.log (3).txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/14232789/homebridge-enphase-envoy.log.3.txt)

grzegorz914, 2024-02-11:
this fixed in 7.11.7

CaptainPic, 2024-02-11:
Thanks so much.

7.11.9 The Enhpase Accessories are now showing up under Homebridge accessories.

I did see one error flash by in the debug log  - 4th line down in the cut and paste below but this didn’t prevent the items showing up.   Let me know if you want the whole log.

Can you clarify the meaning of the Empower Grid State Sensor?  Picture below.  It shows Empower Grid Sensor as Closed either when I set grid mode to Grid On or Grid Off – If the external power grid is up shouldn’t this grid contact sensor show closed when the Grid is On? –  also there are a couple spelling errors in the docutmentation  grod instead of grid

[A screenshot of a phone  Description automatically generated]

I know I have to add the accessories to Homekit/Controller separately and when done correctly there are two bridges that show up.  I got the first one added to Homekit  (homebridge-enphase-envoy) but don’t remember how I added the second one which showed up as  “Enphase 80DC”. It needs a code to add manually and I’m just not seeing where to find it.

Thanks again

Jack

2/11/2024, 8:57:54 AM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, debug: Ensemble status: {
[2/11/2024, 8:57:54 AM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Requesting ensemble status error: ReferenceError: enchargesRatedPowerSum is not defined. Trying again.
[2/11/2024, 8:57:55 AM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, debug: Production ct: {
[2/11/2024, 8:57:55 AM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, debug: Production power state: true
[2/11/2024, 8:57:55 AM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, debug: Production power level: 100.0 %
[2/11/2024, 8:57:55 AM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, debug: Production power peak detected: false

grzegorz914, 2024-02-11:
Ok, this error fixed in 7.11.10 now need to work all correct

grzegorz914, 2024-02-11:
The grid mode sensors will not working in 7.11.9 because was error, should working correct in 7.11.10. The grid mode sensors need to be first configured in plugin settings. 
If you don’t see the accessory in home app to add it try this: https://github.com/grzegorz914/homebridge-enphase-envoy/wiki#troubleshooting

grzegorz914, 2024-02-12:
@CaptainPic if all working without error please post the debug log, I want to check what new properties are in 8.x.x firmware. 

Have You Enphase EV Charger in the system installed?

CaptainPic, 2024-02-12:
Getting the debug log now.  Will post to the gift hub comments.

All appears to be working, however, I’m still uncertain on the status of the Grid Mode Sensor.   I configured it as shown below but it shows as closed in Homebride/Homekit.   I can disconnect from the grid in the Enphase app to test to see if this sensor changes but don’t want to do that today.
[A screenshot of a computer  Description automatically generated]

[A screenshot of a phone  Description automatically generated]

Yes I have the IQ EV60 charger installed and working.  Vehicle is unplugged at the moment as reflected in the app.

[A screenshot of a phone  Description automatically generated]

Trying to figure out the power level sensors.  It appears they trigger when you set a level for example if solar production is over 1KW set a trigger to open.  Not sure I see a use case in my application for that.

CaptainPic, 2024-02-12:
Latest debug log  7.11.10  no errors and bridge / accessories loaded properly

[homebridge-enphase-envoy.log (6).txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/14241224/homebridge-enphase-envoy.log.6.txt)

grzegorz914, 2024-02-12:
Try to configure sensors as below:
```
 "enchargeGridModeSensors": [
 {
      "name": "Grid OFF",
      "gridMode": "off-grid",
      "mode": true
 },
{
      "name": "Multimode On Grid",
      "gridMode": "multimode-ongrid",
      "mode": true
    }
  ],
  "solarGridModeSensors": [
    {
      "name": "Off Grid",
      "gridMode": "off-grid",
      "mode": true
    }
  ]
```

switch device to grid off and after 5 min get the debug log.

CaptainPic, 2024-02-12:
I’ll give this a go later this afternoon.  Does it need to be done editing the json file or can I follow these setting in the UX?

grzegorz914, 2024-02-12:
just use UX, is simply and save

CaptainPic, 2024-02-12:
Got it. So when you say switch device off you mean disconnect the emphase system from the public utility grid?

grzegorz914, 2024-02-12:
set to grid off

CaptainPic, 2024-02-12:
Ok.  Never took it off grid in the app but its off now.  Lots of solar at the moment so the production automatically scaled back since there is nowhere to export it.  Got a 5 minute timer running and will send logs in gift hub momentarily.   I. need to run out so won’t be on line for a couple hours.

CaptainPic, 2024-02-12:
Debug log after going off grid for about 5-6 minutes

[homebridge-enphase-envoy.log (7).txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/14246665/homebridge-enphase-envoy.log.7.txt)

grzegorz914, 2024-02-12:
nice, as you see your device report `multimode-offgrid`  and `multimode-ongrid`

```
"Enchg_grid_mode": "multimode-offgrid",
"Solar_grid_mode": "multimode-offgrid"
```

I suggest You to create two sensors for encharge and solar like bottom, in this case the sensors will fired if go to grid off or on

 ```
"enchargeGridModeSensors": [
 {
      "name": "Multimode Grid OFF",
      "gridMode": "multimode-offgrid",
      "mode": true
 },
{
      "name": "Multimode On Grid",
      "gridMode": "multimode-ongrid",
      "mode": true
    }
  ],
  "solarGridModeSensors": [
    {
      "name": "Multimode Grid OFF",
      "gridMode": "multimode-offgrid",
      "mode": true
 },
{
      "name": "Multimode On Grid",
      "gridMode": "multimode-ongrid",
      "mode": true
    }
  ]
```

CaptainPic, 2024-02-12:
Thank you for all your help.  I hope the data I’ve been providing is equally helpful.

I set up the two encharge and solar multimode sensors.  With a little solar left in the day they show up like this below:
[A screenshot of a phone  Description automatically generated]
The left colum is Multimode ON, the right Multimode OFF.

I’m struggling a bit with how to interpret these sensors and what “multimode” means.    I’m guessing this has to do with the multi-directional way the power can flow e.g. solar panels to either local consumption, charging the battery or exporting out to the public utility.

I’ll watch these and try to deduce what the proper definition is but I’m really not clear right now what they mean.   Perhaps my Solar installer can help as they are very interested in this solution.

What I’m looking to determine is the binary option that I toggled earlier to get the logs that shows up in the upper left corner of the Enlighten App below as either On Grid (public utility) or Off  which would be a power failure in the public utility assuming you didn’t disconnect it deliberately, and  then building automations around that state (with a short timer just to avoid doing anything for a momentary glitch).

[A screenshot of a cell phone  Description automatically generated]

Thanks again

Jack

grzegorz914, 2024-02-12:
I have published 7.11.11 with some fixes and improvements.
I don’t see your screenshots. 
The sensors fired if your installation go to selected mode, tor example:
If You create sensor `Multimode Grid Off` then will be fired if your Installation go to `Off Grid`.  
If You create sensor `Multimode Grid On` then will be fired if your Installation go to `On Grid`.

CaptainPic, 2024-02-13:
7.11.11 looks good - don't see anything red indicating an error in the logs

Let me see if I can add the screen shots here.  

I added two Empower Grid Multimode Grid Sensors for ON and OFF and took the system off grid twice today once when running minimal loads and later with about 8KW of power consumed.  Looks like the Power Level Sensor which I set at 4KW is also working fine.

All Three of the sets of  Mulitmode Grid Sensors (Empower, Encharge and Solar) show triggered for Grid ON when the system is connected and the Grid OFFs show as not triggered when the grid I on.  They all flip appropriately when I take the Enphase system off grid.   Seems like about a 30 second to minute delay at most.

Since all 6 of these sensors (3 pairs) have similar state changes, it appears that I can use any of the Multimode sensors to determine if the system is off grid and therefore indicative of a power outage (assuming I'm not manually manipulating it).  Is there a reason why I should prefer to use one or the other sensors I set up under the Empower, Encharge or Solar tabs in the UX configuration?  I'm thinking the Empower Sensor makes the most sense since the public grid status isn't specific to the batteries (Encharge) or Solar aspects of the Enhpase system.   Enphase terminology is a bit confusing here and I don't think they use Encharge anymore to refer to batteries.

I tried to attach  the picture of the three sets of grid sensors when grid was on and off

BTW, Did you know you can ask Siri to get the status of the Enphase System with these accessories?  Using a Personal Shortcut have her tell you the the correct name instead of calling everything lightbulbs and contact sensors.  Controller may help but since I see formatting this into an apple intercom response probably don't need controller to do this.

[image]
[image]
[image]

grzegorz914, 2024-02-13:
Im happy that all working fine now.
I have add all possible sensors for Empower, Encharge, Solar, because we can have different configurations of our PV systems and not all have Empower installed.
You can use only one of available sensors to identify the grid state. All sensor will always report same mode.

I see now the screenshots. 
If You find any problems please report it as new issue.

grzegorz914, 2024-02-13:
in 7.12.0 beta I have reduced the refresh time for grid mode sensor, it will by more quickly react  on changes.