grzegorz914/homebridge-enphase-envoy #115: ENC rated power

eleventhaus, 2023-12-04:
I keep getting this error from homebridge: "This plugin generated a warning from the characteristic 'ENC rated power': characteristic value expected valid finite number and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info."

Not sure what ENC Rated Power is or where the value is coming from. The only value that I did not change was the energy lifetime offset which I left at 0. Not sure where to get this value from or if I should change it from 0.

Here is my plugin config:
```
{
    "devices": [
        {
            "name": "Envoy",
            "host": "192.168.107.60",
            "envoyFirmware7xx": true,
            "envoyPasswd": "XXX",
            "envoySerialNumber": "XXX",
            "enlightenUser": "XXX",
            "enlightenPasswd": "XXX",
            "powerProductionSummary": 6120,
            "powerProductionStateSensor": true,
            "powerProductionPeakSensor": false,
            "powerProductionPeakSensorDetected": 0,
            "powerProductionPeakAutoReset": 0,
            "energyProductionStateSensor": true,
            "energyProductionLevelSensor": false,
            "energyProductionLevelSensorDetected": 0,
            "energyProductionLifetimeOffset": 0,
            "powerConsumptionTotalStateSensor": false,
            "powerConsumptionTotalPeakSensor": false,
            "powerConsumptionTotalPeakSensorDetected": 0,
            "powerConsumptionTotalPeakAutoReset": 0,
            "energyConsumptionTotalStateSensor": false,
            "energyConsumptionTotalLevelSensor": false,
            "energyConsumptionTotalLevelSensorDetected": 0,
            "energyConsumptionTotalLifetimeOffset": 0,
            "powerConsumptionNetStateSensor": false,
            "powerConsumptionNetPeakSensor": false,
            "powerConsumptionNetPeakSensorDetected": 0,
            "powerConsumptionNetPeakAutoReset": 0,
            "energyConsumptionNetStateSensor": false,
            "energyConsumptionNetLevelSensor": false,
            "energyConsumptionNetLevelSensorDetected": 0,
            "energyConsumptionNetLifetimeOffset": 0,
            "supportProductionPowerMode": false,
            "supportPlcLevel": false,
            "supportEnsembleStatus": true,
            "enpowerGridStateSensor": true,
            "enchargeGridStateSensor": true,
            "solarGridStateSensor": true,
            "supportLiveData": false,
            "liveDataRefreshTime": 1000,
            "metersDataRefreshTime": 2500,
            "productionDataRefreshTime": 5000,
            "enableDebugMode": false,
            "disableLogInfo": true,
            "disableLogDeviceInfo": false,
            "enableRestFul": false,
            "restFulDebug": false,
            "enableMqtt": false,
            "mqttDebug": false,
            "mqttAuth": true
        }
    ],
    "_bridge": {
        "username": "XXX",
        "port": XXX
    },
    "platform": "enphaseEnvoy"
},
```

eleventhaus, 2023-12-04:
same error here is the [debug log](https://gist.github.com/eleventhaus/746ccf6f9213b13a37ab9840991661d6)

Also, inventory shows 0 batteries but it does show 3 encharges:

```
[04/12/2023, 11:03:44] [homebridge-enphase-envoy] -------- Envoy --------
[04/12/2023, 11:03:44] [homebridge-enphase-envoy] Manufacturer: Enphase
[04/12/2023, 11:03:44] [homebridge-enphase-envoy] Model: X-IQ-AM1-240-3-ES
[04/12/2023, 11:03:44] [homebridge-enphase-envoy] Firmware: D7.3.517
[04/12/2023, 11:03:44] [homebridge-enphase-envoy] SerialNr: 202206048058
[04/12/2023, 11:03:44] [homebridge-enphase-envoy] Time: 04/12/2023, 11:03:30
[04/12/2023, 11:03:44] [homebridge-enphase-envoy] ------------------------------
[04/12/2023, 11:03:44] [homebridge-enphase-envoy] Q-Relays: 0
[04/12/2023, 11:03:44] [homebridge-enphase-envoy] Inverters: 17
[04/12/2023, 11:03:44] [homebridge-enphase-envoy] Batteries: 0
[04/12/2023, 11:03:44] [homebridge-enphase-envoy] --------------------------------
[04/12/2023, 11:03:44] [homebridge-enphase-envoy] Meters: Yes
[04/12/2023, 11:03:44] [homebridge-enphase-envoy] Production: Enabled
[04/12/2023, 11:03:44] [homebridge-enphase-envoy] Consumption: Enabled
[04/12/2023, 11:03:44] [homebridge-enphase-envoy] --------------------------------
[04/12/2023, 11:03:44] [homebridge-enphase-envoy] Enpowers: Yes 1
[04/12/2023, 11:03:44] [homebridge-enphase-envoy] Encharges: Yes 3
[04/12/2023, 11:03:44] [homebridge-enphase-envoy] Wireless Kit: Yes 2
[04/12/2023, 11:03:44] [homebridge-enphase-envoy] --------------------------------
```

grzegorz914, 2023-12-04:
try beta.22

grzegorz914, 2023-12-04:
Batteries are other devices as encharges, the inventory info is displayed correct.

eleventhaus, 2023-12-04:
> try beta.22

Same error

grzegorz914, 2023-12-04:
try again beta.23

eleventhaus, 2023-12-04:
No error on beta.25 or beta.26 👍🏼

closed 2023-12-04 by eleventhaus