grzegorz914/homebridge-enphase-envoy #116: Cannot get light to function

richie510, 2023-12-07:
I can see the details of all of my microinverters, but I cannot get the basic "light" to turn on or show the % of max wattage. 

Is there something wrong with my setup, or is this a bug? I just installed this plugin a couple days ago. I have never gotten it to work.

My envoy firmware is 
Envoy
    Part Number
    800-00663-r05
    Software Version
    D7.3.466 (9debea) 

here is my setup:

            "devices": [
                {
                    "name": "Solar",
                    "host": "10.x.x.x",
                    "envoyFirmware7xx": true,
                    "envoySerialNumber": "xxxxxxxxxxxxxxxx",
                    "enlightenUser": "xxxxxxxxxxxxxxxx",
                    "enlightenPasswd": "xxxxxxxxxxxxxxxx",
                    "powerProductionSummary": 6200,
                    "powerProductionStateSensor": true,
                    "powerProductionPeakSensor": true,
                    "powerProductionPeakSensorDetected": 4000,
                    "powerProductionPeakAutoReset": 0,
                    "energyProductionStateSensor": true,
                    "energyProductionLevelSensor": true,
                    "energyProductionLevelSensorDetected": 1000,
                    "energyProductionLifetimeOffset": 0,
                    "powerConsumptionTotalStateSensor": false,
                    "powerConsumptionTotalPeakSensor": false,
                    "powerConsumptionTotalPeakSensorDetected": 0,
                    "powerConsumptionTotalPeakAutoReset": 0,
                    "energyConsumptionTotalStateSensor": false,
                    "energyConsumptionTotalLevelSensor": false,
                    "energyConsumptionTotalLevelSensorDetected": 0,
                    "energyConsumptionTotalLifetimeOffset": 0,
                    "powerConsumptionNetStateSensor": false,
                    "powerConsumptionNetPeakSensor": false,
                    "powerConsumptionNetPeakSensorDetected": 0,
                    "powerConsumptionNetPeakAutoReset": 0,
                    "energyConsumptionNetStateSensor": false,
                    "energyConsumptionNetLevelSensor": false,
                    "energyConsumptionNetLevelSensorDetected": 0,
                    "energyConsumptionNetLifetimeOffset": 0,
                    "supportProductionPowerMode": false,
                    "supportPlcLevel": false,
                    "supportEnsembleStatus": false,
                    "enpowerGridStateSensor": false,
                    "enchargeGridStateSensor": false,
                    "solarGridStateSensor": false,
                    "supportLiveData": true,
                    "liveDataRefreshTime": 1000,
                    "metersDataRefreshTime": 2500,
                    "productionDataRefreshTime": 5000,
                    "enableDebugMode": false,
                    "disableLogInfo": false,
                    "disableLogDeviceInfo": false,
                    "enableRestFul": false,
                    "restFulDebug": false,
                    "enableMqtt": false,
                    "mqttDebug": false,
                    "mqttAuth": false
                }
            ],
            "platform": "enphaseEnvoy"

grzegorz914, 2023-12-07:
Can you enable debug in plugin config advanced settings >> system>>debug restart and get the whole log

richie510, 2023-12-07:
[homebridge.log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/13602500/homebridge.log.txt)

Thanks for the quick follow up!

grzegorz914, 2023-12-07:
please try 7.7.8-beta.2 and let me know with the log again

richie510, 2023-12-07:
[homebridge.log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/13603144/homebridge.log.txt)

The light and percent work now with the beta.

commit b63e0872 (2023-12-07): fix #116  and other small changes
closed 2023-12-07 by grzegorz914