grzegorz914/homebridge-enphase-envoy #117: Automation at 5kW flow to net

rrebel01, 2023-12-22:
Iam looking for a way to switch on my waterboiler when production - consumption = >5kW and switch it back off when < 5kW. 

Not sure if this is already possible, if so ill like to have some guidance. If not it would be great to have this implemented

grzegorz914, 2023-12-22:
please install 7.7.13-beta.8 then go to plugin settings Advanced Settings>>>Consumption>>>Net>>>Power  enable `Power Peak Sensor`  and set  `Power Peak Detected` to 5000W , save and restart plugin. Check and let me know.

grzegorz914, 2023-12-22:
i have pushed 7.8.0-beta where you can create multiple sensors for Power and Energy - Production, Consumption Total and Net (in the Net You can use negative values like -5000W, this is production-consumption = > 5kW)

grzegorz914, 2023-12-23:
added in 7.8.0

closed 2023-12-23 by grzegorz914