grzegorz914/homebridge-enphase-envoy #118: Add Battery percentage as light sensor in HomeKit

AndreasBrack, 2023-12-31:
First: Thank you very much for this very good tool.

You are already reading the current battery percentage from the envoy (seen in the logs), but as far as I see this is currently not available in the Apple-Homekit representation. 

Would it be possible to add that? If it is already available in HomeKit then this Issue can be treated as a documentation request.

Regards, Andreas

grzegorz914, 2023-12-31:
I will add this in next release.

grzegorz914, 2023-12-31:
try 7.9.0-beta.0 and let me know, the lightbulb should appear automatically where Tile ON = Batteries/Encharges `Charged level > 0`, Charged Level displayed as a slider in `%`

AndreasBrack, 2024-01-01:
Sorry, I can’t see such a tag or a branch where you added it. 

Since I am not used to work with GitHub or manually updating the homebridge plugins I am unsure if I searched correctly. Would you please provide a link to the beta?

grzegorz914, 2024-01-01:
npm install g home ridge-enphase-envoy@7.9.0-beta.0

AndreasBrack, 2024-01-01:
[image]

works perfectly!

thank you very much!

closed 2024-01-01 by AndreasBrack