grzegorz914/homebridge-enphase-envoy #121: Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic')

echamberlain, 2024-01-15:
I'm getting the following error while trying to set up a new install:

```
[15/01/2024, 08:34:22] [homebridge-enphase-envoy] Device: envoy.local Envoy, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
[15/01/2024, 08:34:45] [homebridge-enphase-envoy] Device: envoy.local Envoy, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Trying again in 15s.
```

Any ideas on how I can fix this?

grzegorz914, 2024-01-15:
Go to,plugin settings Advanced Settings>>System>>Debug enable it and get the whole log.

grzegorz914, 2024-01-15:
Fixed in 7.9.3

echamberlain, 2024-01-15:
Here are the logs:

[homebridge-enphase-envoy.log](https://github.com/grzegorz914/homebridge-enphase-envoy/files/13941468/homebridge-enphase-envoy.log)

grzegorz914, 2024-01-15:
This was other error, fixed in 7.9.4

closed 2024-01-15 by grzegorz914

echamberlain, 2024-01-15:
Thanks for the quick turnaround.  The error is gone.

commit 0d15937f (2024-01-15): fix #121 and #114