grzegorz914/homebridge-enphase-envoy #123: No inventory with 7.11.x

xorguy, 2024-02-10:
Hi.

Since updating to 7.11.x with firmware D5.0.62 it just gets Home but no inventory. Tried rolling back to 7.10 and it works. With node.js v20.11.0

`[2/10/2024, 8:33:51 PM] [homebridge-enphase-envoy] Device: 10.0.0.113 Envoy-S, debug: Requesting inventory.
[2/10/2024, 8:33:52 PM] [homebridge-enphase-envoy] Device: 10.0.0.113 Envoy-S, Requesting inventory error: TypeError: Converting circular structure to JSON
    --> starting at object with constructor 'ClientRequest'
    |     property 'socket' -> object with constructor 'Socket'
    --- property '_httpMessage' closes the circle. Trying again in 15s.`

Regards.

BenouGui, 2024-02-10:
Same problem

Device: 192.168.1.6 Envoy-Ss, Requesting inventory error: TypeError: Converting circular structure to JSON
    --> starting at object with constructor 'TLSSocket'
    |     property '_httpMessage' -> object with constructor 'ClientRequest'
    --- property 'socket' closes the circle. Trying again in 15s.

grzegorz914, 2024-02-11:
This was fixed with 7.11.6

closed 2024-02-11 by grzegorz914

xorguy, 2024-02-11:
Updated to latest 7.11.7 and it seems to gather few more inventory information but just after loading inventory an error is thrown:
`[2/11/2024, 5:07:15 PM] [homebridge-enphase-envoy] Device: 10.0.0.113 Envoy-S, Requesting inventory error: TypeError: Cannot read properties of undefined (reading 'devices'). Trying again in 15s.`

And it does not get any more information.

grzegorz914, 2024-02-11:
Fixed in 7.11.9