grzegorz914/homebridge-enphase-envoy #130: Requesting grid profile error: AxiosError: Request failed with status code 401. Trying again in 15s.

rrebel01, 2024-03-09:
After version update getting error: Requesting grid profile error: AxiosError: Request failed with status code 401. Trying again in 15s.

Tried reverting but no luck.

rrebel01, 2024-03-09:
upgrading to v7.11.11 or above wil give this issues.. below still works

grzegorz914, 2024-03-09:
debug please

grzegorz914, 2024-03-09:
401 is authorization error

rrebel01, 2024-03-09:
It started after upgrade to latest version.  ive tried every version error is fixed after downgrading to v7.11.10

grzegorz914, 2024-03-09:
I don’t know what happened in your system, please post the debug log.

cyberb0b, 2024-03-19:
Same issue here now. I received a firmware update last night (after I called with Enphase and told me my grid profile was gone) . First local certificate error. I have entered Username passs, serial and v7 firmware (Received fw D7.3.88). Now "Device: 10.10.1.248 Solar, Requesting grid profile error: AxiosError: Request failed with status code 401. Trying again in 15s."
[debug.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/14651236/debug.txt)

grzegorz914, 2024-03-19:
Should be fixed, let me know.

grzegorz914, 2024-03-19:
try again update

cyberb0b, 2024-03-19:
So far it looks okay now! Much Appriciated. Especially with those new sensors! Time for me to put a relais between my washmahine and dishwasher door sensor :)

commit 4a5f38b1 (2024-03-19): fix #130
commit 45c07578 (2024-03-19): fix #130
closed 2024-03-19 by grzegorz914