grzegorz914/homebridge-enphase-envoy #133: Please adhere to semver

chrisjshull, 2024-03-19:
Release notes say:
```
# After update to v7.11.0 and above from plevious version the grid mode sensors need to be configured again!!!
# After update to v7.8.0 and above from plevious version the sensors need to be configured again!!!
# After update to v7.5.0 and above from plevious version the sensors need to be configured again❕❕!
```

however, semver would ask that backwards-incompatible changes be marked by new major versions.

grzegorz914, 2024-03-28:
All this versions are compatible with previous, just added some new features and need to be configured again in some sections.

closed 2024-03-28 by grzegorz914

chrisjshull, 2024-05-09:
The whole point of semver is that I shouldn't have to reconfigure anything if I don't pull in a major version - I should only be getting bug fixes and new features. But otherwise my setup should remain compatible. https://semver.org

> MAJOR version when you make incompatible API changes

The API you provide us (which we super appreciate, btw) to use homebridge-enphase-envoy *is* the config.

I just spent the last day trying to figure out why homebridge envoy wasn't working after updating homebridge and several plugins. Turns out the change from milliseconds to seconds in the config was the issue. That change should have been a major semver update.

grzegorz914, 2024-05-09:
Why you not read changelog before update, this will save your time?
[image]

chrisjshull, 2024-05-09:
> Why you not read changelog before update, this will save your time?

We shouldn't have to though? Not reading the changelog for a server-minor change is zero time (and that's how I get to treat my other homebridge plugins). You can't beat zero ;)

But look - I acknowledge that at the end of the day, it's your project and you can run it how you want. I very much appreciate that it exists and will keep using it. This is just some feedback for something I think would be useful, and I'll leave it at that.