grzegorz914/homebridge-enphase-envoy #138: Missing sensor in HomeKit app, but als not visible on accessoires overview page in Homebridge

eronnes, 2024-06-15:
I've got te script working, in the Homebridgelog I see data which comes from my solar installation. But I don't see any sensor-information in my Homekit app. What do I wrong?

Could you please advise, thank you.

grzegorz914, 2024-06-15:
Aere you ad the accessory to the home app?

eronnes, 2024-06-15:
Yes, and a lot of assessories were visible, except for the Enphase plugin assessories.

grzegorz914, 2024-06-16:
Please read this: https://github.com/grzegorz914/homebridge-enphase-envoy?tab=readme-ov-file#about-the-plugin

eronnes, 2024-06-16:
I did, but looks like I did everything as stated. Should I see the Accessoires (sensors) in Accessoires overview in Homebridge itself?

grzegorz914, 2024-06-16:
yes, the sensors are visible in homebridge and in home app, if you have some problems please post the debug log then I can look what is going wrong. To enable debug mode go to plugin settings advanced section>>>system>>debug

eronnes, 2024-06-20:
I found this issue in the logging:
[20-6-2024, 09:11:39] [homebridge-enphase-envoy] Device: xxx.xxx.xxx.xxx Envoy-S, Start error: Requesting power production mode error: AxiosError: Request failed with status code 401., trying again in 15s.

Replaced ip with xxx.xxx.xxx.xxx

eronnes, 2024-06-20:
I've turned of supportProductionPowerMode 

If enabled, control Production Power Mode will be possible in Envoy section (EVE or Controler app), Fw. 7.x.x and newer required installer credentials data.

Now I see sensors coming up!

grzegorz914, 2024-06-20:
Yes, to read/write Production Power Mode and PLC level, Installer credentials data are required in other things wil return error 401 - wrong credentials data.

I think is solved now.

closed 2024-06-20 by grzegorz914