grzegorz914/homebridge-enphase-envoy #140: Token Instead of username and password

unserbaublog, 2024-06-23:
While it might be beneficial for most users to enter Enlighten username and password, I‘d appreciate the ability to enter an existing auth token instead. It appears this config variable was removed at some point in time. 
Is there a possibility to bring it back?

grzegorz914, 2024-06-23:
Right now is not possible, in current version the app get the token automatically and check its expired date, if the token expired then token will refresh automatically. In your case after token expired need to paste new token manually, in this case some users loses the connectivity to Envoy. Technically is this possible but current version is very easy and simply.

unserbaublog, 2024-06-23:
Thanks gregorz914. Sadly I‘m relying on multiple tools and the only option to get them all going is to use the manually created token. AFAIK there can be only one token. Using username and password would break the other tools, so co-existence won‘t be possible.

grzegorz914, 2024-06-23:
Ok, I agree, I will prepare beta release with possibility to put own token instead user and password.

unserbaublog, 2024-06-23:
Amazing! That‘s very much appreciated.

grzegorz914, 2024-06-23:
please try 7.16.0-beta.4 and let me know

grzegorz914, 2024-06-23:
7.16.0 released

unserbaublog, 2024-06-24:
Thank you for the efforts. It works. 🙏🏼

closed 2024-06-24 by grzegorz914