grzegorz914/homebridge-enphase-envoy #141: Trouble using with HOOBS

logal727, 2024-07-22:
Hi there,

Is there a trick to get this to work with HOOBS, I see it's HOOBS certified, and Ive done all sorts of changes to the config file, but it seems like under HOOBS it does not recognize the changes I put there. I keep getting the "self signed certificate" login error, which I've generated the token and tried to use that as well and also just using Enlighten UN and PW config options. 

The `envoyToken` and enlighten username and password config options don't seem to do anything even though I fill them out. Anyone had success lately with HOOBS and this? Thanks!

grzegorz914, 2024-07-23:
Please post the debug log (go to plugin settings>>advanced settings>>system>>debug), plugin version, envoy version firmware and the config.

logal727, 2024-07-23:
Actually, I got it working! The documentation with the GitHub and the HOOBS listing is different, HOOBS listing may need updating at some point

closed 2024-07-23 by grzegorz914