grzegorz914/homebridge-enphase-envoy #142: Is there a way to detect if the local city grid is down?

eleventhaus, 2024-07-25:
On the enphase web portal and on the enphase app, there is a indicator for the local municipal grid.

[image]

Is there a way to expose this indicator in the plugin as a sensor which reflects the status of the municipal grid?

A sensor would make it easy to create shortcuts using homekit in iOS, where a certain set of actions could be run if the local grid goes down.

grzegorz914, 2024-07-25:
Yes, the sensor can be created in Enpower or Solar section
[image]

eleventhaus, 2024-07-26:
I have the following sensors created in the plugin config: Enpower Grid On, Enpower Grid Off, Enpower Multimode On, Enpower Multimode Off

https://github.com/user-attachments/assets/9d50d8fb-ceab-4a0c-a222-22c388af82fb

This is what my homekit looks like:
[image]

This is what my enphase portal looks like:
[image]

If I have Enpower Grid On active as a sensor in the plugin config, why is it off in Homekit?

Will the Grid On sensor not work with my system because I have solar and storage? Should I only use multimode grid then?

grzegorz914, 2024-07-26:
Yes, You need use multimode-grid on or off sensor for Your purpose.

eleventhaus, 2024-07-27:
Thought everything was working well but this morning I noted that although my system is on the grid and everthing is working as expected, all of the sensors I created in homekit using the plugin are reading as closed or off

[image]

[image]

These are all enpower sensors
[image]

grzegorz914, 2024-07-28:
Can you get me debug log, go to plugin settings advanced settings>>system>>debug enable it and restart plugin then after a while download and post the log file.

eleventhaus, 2024-07-30:
[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/16433749/homebridge-enphase-envoy.log.txt)

grzegorz914, 2024-08-04:
Should be fixed in 8.0.2, try and let me know

closed 2024-08-04 by grzegorz914

grzegorz914, 2024-08-04:
You can also try beta 9.0.0, here is added support for Homebridge 2.0 and new sensor profiles to select , `Grid Tied`and `Grid Forming`

eleventhaus, 2024-08-04:
Just to confirm:

- grid tied means municipal grid is working
- grid forming mean municipal grid is down

grzegorz914, 2024-08-04:
in release v8.1.1 fixed display and publish ensemble status.
Also fixed  encharge and solar grid mode sensors, please update and get me the debug please

eleventhaus, 2024-08-04:
Updated to 8.1.1

[homebridge-enphase-envoy.log (2).txt](https://github.com/user-attachments/files/16488312/homebridge-enphase-envoy.log.2.txt)

grzegorz914, 2024-08-04:
Encharge profile are in experimental phase and not working yet, please disable it in plugin config Ensemble>>Encharge>>Profile

eleventhaus, 2024-08-05:
I'm not sure the Grid Tied/Forming or Multimode sensors can be used to indicate if the utility is online or offline.

In my case, Grid Tied shows as off/Multimode shows as on when my system starts using battery power regardless of utility grid status. When utility grid is off none of the indicators reflect that.

I need a sensor that will indicate when the ultility is down. Is that possible?

grzegorz914, 2024-08-05:
I Don’t have ensemble in my system, Please post debug log if grid is off,  albo if grid is on

grzegorz914, 2024-08-08:
HI @eleventhaus  in latest 9.0 beta im added empower grid state sensor which represent Grid ON/OFF state, please try it and let mi know.

eleventhaus, 2024-08-08:
I updated to the latest beta and I'm getting the following error:
"Device: 192.168.107.60 Envoy, Start error: Requesting ensemble inventory error: ReferenceError: enpwrGridMode is not defined., trying again in 15s."

[homebridge-enphase-envoy.log (3).txt](https://github.com/user-attachments/files/16552171/homebridge-enphase-envoy.log.3.txt)

grzegorz914, 2024-08-08:
fixed, beta 132 added also Enpower Grid State control enable it in enpower section

eleventhaus, 2024-08-08:
Fixed the last error.

New error: "Device: 192.168.107.60 Envoy, Start error: Requesting ensemble status error: TypeError: parseFloat is not a function., trying again in 15s."

[homebridge-enphase-envoy.log (4).txt](https://github.com/user-attachments/files/16552450/homebridge-enphase-envoy.log.4.txt)

grzegorz914, 2024-08-08:
try 134

eleventhaus, 2024-08-08:
I'm on 135. I added the Grid State Control as a switch and Grid State Sensor as a contact sensor but not seeing them on homekit

[image]

[image]

[image]

[homebridge-enphase-envoy.log (6).txt](https://github.com/user-attachments/files/16552616/homebridge-enphase-envoy.log.6.txt)

grzegorz914, 2024-08-08:
136 fixed

eleventhaus, 2024-08-08:
On 141

Still not seeing the Grid State Control or Grid State Sensor in homekit

Also, getting these non-critical errors on homebridge:

```
[08/08/2024, 16:07:01] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Rest power': characteristic value expected valid finite number and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.
[08/08/2024, 16:07:15] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Rest power': characteristic value expected valid finite number and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.
[08/08/2024, 16:07:30] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Rest power': characteristic value expected valid finite number and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.
[08/08/2024, 16:07:45] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Rest power': characteristic value expected valid finite number and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.
[08/08/2024, 16:08:00] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Rest power': characteristic value expected valid finite number and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.
[08/08/2024, 16:08:15] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Rest power': characteristic value expected valid finite number and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.
[08/08/2024, 16:08:30] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Rest power': characteristic value expected valid finite number and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.
[08/08/2024, 16:08:45] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Rest power': characteristic value expected valid finite number and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.
[08/08/2024, 16:09:00] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Rest power': characteristic value expected valid finite number and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.
[08/08/2024, 16:09:15] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Rest power': characteristic value expected valid finite number and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.
```

[homebridge-enphase-envoy.log (7).txt](https://github.com/user-attachments/files/16554098/homebridge-enphase-envoy.log.7.txt)

grzegorz914, 2024-08-09:
thanks for your patience, the control grid is fixed in 142, please test it and let me know with log

grzegorz914, 2024-08-09:
please post in this topic #144

eleventhaus, 2024-08-10:
Geting error:

```
[10/08/2024, 09:23:14] [homebridge-enphase-envoy] Device: 192.168.107.60 Envoy, Start error: Requesting ensemble status error: TypeError: Cannot set properties of undefined (setting 'status')., trying again in 15s.
```

[homebridge-enphase-envoy.log (8).txt](https://github.com/user-attachments/files/16570259/homebridge-enphase-envoy.log.8.txt)

grzegorz914, 2024-08-10:
ok, try 196

eleventhaus, 2024-08-11:
v202

`[10/08/2024, 18:28:45] [homebridge-enphase-envoy] Device: 192.168.107.60 Envoy, Start error: Requesting ensemble dry contacts error: AxiosError: Request failed with status code 404., trying again in 15s.`

[homebridge-enphase-envoy.log (9).txt](https://github.com/user-attachments/files/16571284/homebridge-enphase-envoy.log.9.txt)

grzegorz914, 2024-08-11:
203 fixes the wrong url to dry contacts

eleventhaus, 2024-08-11:
On 203

`[10/08/2024, 22:20:58] [homebridge-enphase-envoy] Device: 192.168.107.60 Envoy, Start error: Requesting ensemble dry contacts error: AxiosError: Request failed with status code 404., trying again in 15s.`

[homebridge-enphase-envoy.log (10).txt](https://github.com/user-attachments/files/16571651/homebridge-enphase-envoy.log.10.txt)

grzegorz914, 2024-08-11:
fixed in 204

eleventhaus, 2024-08-11:
`[10/08/2024, 22:41:16] [homebridge-enphase-envoy] Device: 192.168.107.60 Envoy, Start error: Prepare accessory error: Error: Cannot add a Characteristic with the same UUID as another Characteristic in this Service: 00000254-000B-1000-8000-0026BB765291, trying again in 15s.`

[homebridge-enphase-envoy.log (11).txt](https://github.com/user-attachments/files/16571683/homebridge-enphase-envoy.log.11.txt)

grzegorz914, 2024-08-11:
fixed in 205

grzegorz914, 2024-08-11:
plese post debug also if working correct

eleventhaus, 2024-08-11:
On 206

Homekit sensors are finally working and showing correct status.

Getting the following error:
`[10/08/2024, 23:29:34] [homebridge-enphase-envoy] Device: 192.168.107.60 Envoy, Update ensemble error: Requesting ensemble encharge settings. error: TypeError: Cannot read properties of undefined (reading 'mode')..`

[homebridge-enphase-envoy.log (13).txt](https://github.com/user-attachments/files/16571820/homebridge-enphase-envoy.log.13.txt)

grzegorz914, 2024-08-11:
208 ready

eleventhaus, 2024-08-11:
v 241

`[11/08/2024, 08:02:57] [homebridge-enphase-envoy] Device: 192.168.107.60 Envoy, Start error: Requesting ensemble status error: TypeError: Cannot read properties of undefined (reading 'forEach')., trying again in 15s.`

[homebridge-enphase-envoy.log (14).txt](https://github.com/user-attachments/files/16574099/homebridge-enphase-envoy.log.14.txt)

grzegorz914, 2024-08-11:
Fixed in 243

eleventhaus, 2024-08-11:
243

`[11/08/2024, 08:12:14] [homebridge-enphase-envoy] Device: 192.168.107.60 Envoy, Start error: Requesting ensemble status error: TypeError: Cannot read properties of undefined (reading 'gridProfile')., trying again in 15s.`

[homebridge-enphase-envoy.log (15).txt](https://github.com/user-attachments/files/16574122/homebridge-enphase-envoy.log.15.txt)

grzegorz914, 2024-08-11:
Fixed 245

eleventhaus, 2024-08-11:
`[11/08/2024, 08:24:07] [homebridge-enphase-envoy] Device: 192.168.107.60 Envoy, Start error: Requesting dry contacts error: ReferenceError: date is not defined., trying again in 15s.`

[homebridge-enphase-envoy.log (16).txt](https://github.com/user-attachments/files/16574158/homebridge-enphase-envoy.log.16.txt)

grzegorz914, 2024-08-11:
Fixed 247

eleventhaus, 2024-08-11:
`[11/08/2024, 08:47:38] [homebridge-enphase-envoy] Device: 192.168.107.60 Envoy, Start error: Prepare accessory error: TypeError: Cannot read properties of undefined (reading 'type'), trying again in 15s.`

[homebridge-enphase-envoy.log (17).txt](https://github.com/user-attachments/files/16574212/homebridge-enphase-envoy.log.17.txt)

grzegorz914, 2024-08-11:
Fixed 248

eleventhaus, 2024-08-11:
No critical errors and homekit is working ✅😌

Thank you!

[homebridge-enphase-envoy.log (18).txt](https://github.com/user-attachments/files/16574225/homebridge-enphase-envoy.log.18.txt)

grzegorz914, 2024-08-11:
i have fixed some warnings, try 251 and post the log please

eleventhaus, 2024-08-12:
270

`[11/08/2024, 17:39:10] [homebridge-enphase-envoy] Device: 192.168.107.60 Envoy, Start error: Requesting ensemble status error: TypeError: Cannot set properties of undefined (setting 'status')., trying again in 15s.`

[homebridge-enphase-envoy.log (19).txt](https://github.com/user-attachments/files/16576583/homebridge-enphase-envoy.log.19.txt)

grzegorz914, 2024-08-12:
fixed 271

eleventhaus, 2024-08-12:
289

`[12/08/2024, 08:35:10] [homebridge-enphase-envoy] Device: 192.168.107.60 Envoy, Start error: Requesting tariff. error: TypeError: Cannot read properties of undefined (reading 'code')., trying again in 15s.`

[homebridge-enphase-envoy.log (20).txt](https://github.com/user-attachments/files/16589010/homebridge-enphase-envoy.log.20.txt)

grzegorz914, 2024-08-12:
Fixed typo 290 ready

eleventhaus, 2024-08-12:
Not seeing 290 on the version list for the plugin

grzegorz914, 2024-08-12:
Try now

eleventhaus, 2024-08-12:
292

Homekit is working, no critical errors

[homebridge-enphase-envoy.log (21).txt](https://github.com/user-attachments/files/16589531/homebridge-enphase-envoy.log.21.txt)

grzegorz914, 2024-08-12:
293 fixed rest power characteristics warning

grzegorz914, 2024-08-12:
294 should fixed ConfiguredName warning,

I think all errors are fixed now, please try call functions regarding uncharges empowers, I don't have it in my environment.

eleventhaus, 2024-08-12:
294

No warning or errors. Homekit working.

[homebridge-enphase-envoy.log (22).txt](https://github.com/user-attachments/files/16589758/homebridge-enphase-envoy.log.22.txt)

eleventhaus, 2024-08-12:
>please try call functions regarding uncharges empowers

Not sure what this means. What are call functions? Encharges or Enpower? Or both?

grzegorz914, 2024-08-12:
for example for encharge You can change Profile, for empower set Grid ON/OFF, all functions can be created in plugin config menu

grzegorz914, 2024-08-12:
you can also control Dry Contacts

eleventhaus, 2024-08-12:
296

`[12/08/2024, 11:03:26] [homebridge-enphase-envoy] Device: 192.168.107.60 Envoy, Start error: Prepare accessory error: TypeError: this.pv.envoy.home.wirelessConnektions is not iterable, trying again in 15s.`

[Uploading homebridge-enphase-envoy.log (23).txt…]()

grzegorz914, 2024-08-12:
fixed

[15 more replies]

commit c4870d7f (2024-08-04): fix #142