grzegorz914/homebridge-enphase-envoy #149: Axios Error

eleventhaus, 2024-08-17:
v9.2.0

`[17/08/2024, 08:01:45] [homebridge-enphase-envoy] Device: 192.168.107.60 Envoy, Prepare accessory error: TypeError [ERR_INVALID_ARG_TYPE]: The "data" argument must be of type string or an instance of Buffer, TypedArray, or DataView. Received undefined, trying again in 15s.
[17/08/2024, 08:01:45] [homebridge-enphase-envoy] Device: 192.168.107.60 Envoy, Start error: Error: homebridge-enphase-envoy attempt to register an accessory that isn't PlatformAccessory!, trying again in 15s.
[17/08/2024, 08:01:46] [homebridge-enphase-envoy] Device: 192.168.107.60 Envoy, Requesting info error: AxiosError: Request failed with status code 504., trying again in 15s.`

[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/16645357/homebridge-enphase-envoy.log.txt)

grzegorz914, 2024-08-18:
fixed in 9.2.2

eleventhaus, 2024-08-19:
`Device: 192.168.107.60 Envoy, {}, trying again in 15s.`

[homebridge-enphase-envoy.log (1).txt](https://github.com/user-attachments/files/16662827/homebridge-enphase-envoy.log.1.txt)

grzegorz914, 2024-08-19:
Please remove your log from homebridge first before download and post new. Your log is so big and contain thousands of old lines.

grzegorz914, 2024-08-19:
In your system is something wrong, what an envoy you use and what firmware is installed.

eleventhaus, 2024-08-19:
> Please remove your log from homebridge first before download and post new

Not sure how to clear logs

grzegorz914, 2024-08-19:
[image]

eleventhaus, 2024-08-19:
9.2.5

[homebridge-enphase-envoy.log (2).txt](https://github.com/user-attachments/files/16664213/homebridge-enphase-envoy.log.2.txt)

grzegorz914, 2024-08-19:
Try 9.2.6 beta and let me know

eleventhaus, 2024-08-19:
9.2.6 beta 3

No critical errors

[homebridge-enphase-envoy.log (3).txt](https://github.com/user-attachments/files/16664350/homebridge-enphase-envoy.log.3.txt)

commit 0ec28a0f (2024-08-19): fix #149, release 9.2.6
closed 2024-08-19 by grzegorz914