grzegorz914/homebridge-enphase-envoy #150: Bug with Power feature

bbleyzat, 2024-08-19:
Hello there, 

First thanks for this amazing plugin!

I saw you publish the 9.x 1 week ago, (I am not in Homebridge 2 yet, 'cause still in beta), i tried several version 9.x of you plugin, but each time this feature :"Here set the power summary in (W) of all microinverters, based on this value the power production level will be calculated and displayed as (%) in HomeKit." is working like a charm in 8.1.1 but not in 9.x, it always display "100%".

Could you help me? 

Log example: v8.1.1: [8/19/2024, 8:18:48 AM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production power level: 9.8 %

v9.2.3: [8/19/2024, 9:21:00 AM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production power level: 0.03013333333333333 %

Thanks in advance.

grzegorz914, 2024-08-19:
fixed in 9.2.4

closed 2024-08-19 by grzegorz914

grzegorz914, 2024-08-19:
this was happen only if you do not have CT meters