grzegorz914/homebridge-enphase-envoy #152: Error: Requesting production ct error

eleventhaus, 2024-08-27:
`[27/08/2024, 08:16:30] [homebridge-enphase-envoy] Device: 192.168.107.60 Envoy, Error: Requesting production ct error: timeout of 15000ms exceeded, trying again in 20s.`

[homebridge-enphase-envoy.log (5).txt](https://github.com/user-attachments/files/16764335/homebridge-enphase-envoy.log.5.txt)

grzegorz914, 2024-08-27:
in 9.3.4 i have increased timeout to 20 sec, please reduce little refresh time for production because lead to freeze the envoy.

please post whole debug log after update, I want to check characteristic 'On': characteristic value expected boolean and received undefined.

eleventhaus, 2024-08-28:
I increased refresh time to 3 seconds for meters, production, ensemble and live data just to be sure

[homebridge-enphase-envoy.log (6).txt](https://github.com/user-attachments/files/16786104/homebridge-enphase-envoy.log.6.txt)

grzegorz914, 2024-08-29:
can you enable debug and then post the log please

eleventhaus, 2024-09-03:
Seems to be working fine now

[homebridge-enphase-envoy.log (7).txt](https://github.com/user-attachments/files/16850829/homebridge-enphase-envoy.log.7.txt)

0berkampf, 2024-09-08:
Same issue here with version 9.3.6, those 2 errors are shown:
Error:  Requesting production ct error: timeout of 25000ms exceeded, trying again.
Error: Requesting inventory error: timeout of 25000ms exceeded, trying again.

grzegorz914, 2024-09-09:
Try to power cycle your Envoy, because is not a plugin problem, I thinh. Please albo post the debug log.

grzegorz914, 2024-09-09:
Probably You have very poor WiFi connection to your Envoy, 25sec timeout is very long.

0berkampf, 2024-09-09:
I’m not sure about the WIFI connection quality because those errors occur less frequently when I increase the Production Refresh Time. I’m trying with 20s for the moment and check tomorrow if errors are shown again.

grzegorz914, 2024-11-05:
Still the issue?

grzegorz914, 2024-11-24:
This error happened if the refresh time is too short, Envoy CPU is poor.

closed 2024-11-24 by grzegorz914