grzegorz914/homebridge-enphase-envoy #154: Still showing No Response and Errors in Homebridge referencing a requirement for Homebridge Version 2.0

CaptainPic, 2024-09-02:
Still throwing several errors in the Homebridge UI monitor when 9.3.5 starts and Apple Home showing the No Response status

[9/2/2024, 1:30:29 PM] [Home-Annette] Setting switch to false

/volume1/homebridge/node_modules/homebridge-enphase-envoy/src/envoydevice.js:1024
            throw new Error(error);
                  ^
Error: Error: Requesting dry contacts error: Cannot read properties of undefined (reading 'updateCharacteristic')
    at EnvoyDevice.start (/volume1/homebridge/node_modules/homebridge-enphase-envoy/src/envoydevice.js:1024:19)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at HomebridgeAPI.<anonymous> (/volume1/homebridge/node_modules/homebridge-enphase-envoy/index.js:131:11)
[9/2/2024, 1:30:43 PM] [homebridge-enphase-envoy] Child bridge process ended
[9/2/2024, 1:30:43 PM] [homebridge-enphase-envoy] Process Ended. Code: 1, Signal: null
[9/2/2024, 1:30:50 PM] [homebridge-enphase-envoy] Restarting Process...
[9/2/2024, 1:30:51 PM] [homebridge-enphase-envoy] Launched child bridge with PID 1878
[9/2/2024, 1:30:51 PM] The plugin "homebridge-enphase-envoy" requires a Homebridge version of >=2.0.0 which does not satisfy the current Homebridge version of 1.8.4. You may need to update this plugin (or Homebridge) to a newer version. You may face unexpected issues or stability problems running this plugin.
[9/2/2024, 1:30:51 PM] Registering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[9/2/2024, 1:30:51 PM] [homebridge-enphase-envoy] Loaded homebridge-enphase-envoy v9.3.5 child bridge successfully
[9/2/2024, 1:30:51 PM] Loaded 0 cached accessories from cachedAccessories.0E3B86A8C11C.
[9/2/2024, 1:30:51 PM] Homebridge v1.8.4 (HAP v0.12.2) (homebridge-enphase-envoy) is running on port 37700.
[9/2/2024, 1:30:51 PM] 

NOTICE TO USERS AND PLUGIN DEVELOPERS
> Homebridge 2.0 is on the way and brings some breaking changes to existing plugins.
> Please visit the following link to learn more about the changes and how to prepare:
> https://github.com/homebridge/homebridge/wiki/Updating-To-Homebridge-v2.0

[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] -------- Enphase --------
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] Manufacturer: Enphase
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] Model: 800-00664-r05
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] Firmware: D8.2.127
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] SerialNr: 202323173895
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] Time: 9/2/2024, 1:31:12 PM
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] Token Valid: 12/4/2024, 4:30:42 PM
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] ------------------------------
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] Q-Relays: 0
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] Inverters: 26
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] --------------------------------
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] Meters: Yes
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] Production: Enabled
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] Consumption: Enabled
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] Storage: Enabled
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] --------------------------------
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] Ensemble: Yes
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] Enpowers: 1
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] Encharges: 4
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] Dry Contacts: 4
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] Wireless Kit: 1
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] --------------------------------
[9/2/2024, 1:31:04 PM] [homebridge-enphase-envoy] Device: 192.168.170.14 Enphase, Error: Error: Prepare accessory error:, trying again.

/volume1/homebridge/node_modules/homebridge-enphase-envoy/src/envoydevice.js:1024
            throw new Error(error);
                  ^
Error: Error: Requesting dry contacts error: Cannot read properties of undefined (reading 'updateCharacteristic')
    at EnvoyDevice.start (/volume1/homebridge/node_modules/homebridge-enphase-envoy/src/envoydevice.js:1024:19)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at HomebridgeAPI.<anonymous> (/volume1/homebridge/node_modules/homebridge-enphase-envoy/index.js:131:11)
[9/2/2024, 1:31:32 PM] [homebridge-enphase-envoy] Child bridge process ended
[9/2/2024, 1:31:32 PM] [homebridge-enphase-envoy] Process Ended. Code: 1, Signal: null
[homebridge.log (6).txt](https://github.com/user-attachments/files/16841163/homebridge.log.6.txt)
[homebridge.log (6).txt](https://github.com/user-attachments/files/16841180/homebridge.log.6.txt)

grzegorz914, 2024-09-03:
fixed in 9.3.6, update and let me know

closed 2024-09-03 by grzegorz914

CaptainPic, 2024-09-08:
9.3.6 Looks good and has eliminated the errors and “no response” in Apple Home.   I’ll share a few other details on my experience so far with this release.

When I first open the Apple Home app the Batteries or “Enchargers”  show “Off”  and then after 10-15 seconds or in some cases up to almost a minute  they refresh to show their value.

I’ve played around with this a bit, and it happens anytime I bring up the home app either by unlocking my phone with Home app the last app viewed or even if I change to another app and then back to the Home app the status will change to “Off”, then after several seconds it refresh to the brightness level.

I see the same thing in the Controller for Homekit app but here the brightness initially shows up properly and the power state slider toggles from Off to On after several seconds delay.

I’ve also noticed the status for the Solar will say “Off” when there is no solar production instead of 0% which it previously showed..

It would seem to me the Home App should cache the last battery status rather than showing it as ‘Off” and the Solar would show 0% at night the way it previously did instead of “Off”.

I’m not too concerned about the way the Home App UI shows status.  My automations trigger off Controller for Homekit where it appears the brightness level is always available as a percentage between 0-100.

I haven’t spent a lot of time researching and will get back if I see anything else.

One other item more documentation related is that I find the various contact sensors and options a bit confusing in terms of what they mean and how they can be used.  Some of it I’ve figured out but it’s a bit of trial and error.   In a prior version the Savings / Self Consumption mode seemed to be available but now all show as Dry Contact 1,2,3….   I’ll eventually figure this out but it’s not really straightforward on how configuring the plug -in translates to contact sensors in the Home app.

I’m starting to play with the temperature detail on the Batteries available in the Controller for Homekit app and use this to trigger turning on a fan in the garage when the batteries bet over 100F.  I’m not sure how reliable the battery temperature variable is but I live in Florida in the USA and the garage gets very hot so turning on a fan particularly when it’s hot and the garage door is open might be effective in bringing the temperature down a bit.  While the batteries operate up to 50 deg C / 122 F the optimal temperature Enphase recommends is 0-32 degrees C or 32 to 86 F.  I’ve put an Eve Air Quality Monitor next to the battiers to see if the battery temperature tracks with a second measurement of temperature in the area.

One final comment is knowing the status of whether a vehicle is plugged into the EV IQ charger would be nice.   I’ve forgotten to plug the car in overnight a few times now.

Thanks so much for your help.   Great plug-in.  I keep sending my solar installer information as I add automations here with praise for what you’ve done.  Unfortunately, I think it’s a bit over their head.

Jack

grzegorz914, 2024-09-09:
Test 9.3.7.bete.0, here is the status of bateries/encharges fixed, if all will correct with this a will go to the next thema.

CaptainPic, 2024-09-10:
Yes – status shows immediately without delay.  Thanks!

grzegorz914, 2024-09-11:
Thanks, in next beta added encharge profile sensors in section Ensemble>>Encharge you can use it to identify encharge profile state.

commit ff8c8fd5 (2024-09-03): fix #154