grzegorz914/homebridge-enphase-envoy #155: Assistance Needed for Configuring Plugin to Fetch Data

Tiimoon, 2024-09-12:
Hello,

I’m having trouble configuring the Homebridge Enphase Envoy plugin to correctly fetch and display my data. Despite following the documentation, I am struggling to get the plugin to work as expected.

I am able to retrieve the data using curl from my terminal, and I have the correct token. However, I’m not sure how to translate this into a working configuration for the plugin.

Could someone please provide a sample JSON configuration file that includes all the possible data points and settings that the plugin can handle? This will help me ensure that I have all the necessary information and settings correctly configured.

Thank you for your assistance!

Tiimoon, 2024-09-12:
Here is an example of the curl command being executed : 

```
curl --location --request GET 'https://MY_@IPivp/meters/reports/production' \ 
--header 'Authorization: Bearer MY_TOKEN \
--header 'Accept: application/json' \
--insecure
{
    "createdAt": 1726151045,
    "reportType": "production",
    "cumulative": {
        "currW": 1391.072,
        "actPower": 1391.072,
        "apprntPwr": 1439.788,
        "reactPwr": 320.215,
        "whDlvdCum": 29637735.142,
        "whRcvdCum": 20123.459,
        "varhLagCum": 7986955.528,
        "varhLeadCum": 0.206,
        "vahCum": 34594158.240,
        "rmsVoltage": 231.207,
        "rmsCurrent": 6.224,
        "pwrFactor": 0.97,
        "freqHz": 50.00
    },
    "lines": [
        {
            "currW": 1391.072,
            "actPower": 1391.072,
            "apprntPwr": 1439.788,
            "reactPwr": 320.215,
            "whDlvdCum": 29637735.142,
            "whRcvdCum": 20123.459,
            "varhLagCum": 7986955.528,
            "varhLeadCum": 0.206,
            "vahCum": 34594158.240,
            "rmsVoltage": 231.207,
            "rmsCurrent": 6.224,
            "pwrFactor": 0.97,
            "freqHz": 50.00
        }
    ]
}
```

grzegorz914, 2024-09-12:
Just use plugin config UI to configure IT.

Tiimoon, 2024-09-13:
Thanks, after searching I found where to enter my credentials. I see the data appearing in the HomeBridge logs. However, I am unable to view the real-time production and consumption figures on my Home app. I can only select occupancy sensors, etc. Can we use this plugin to view real-time production and consumption in the Home app? If yes, how?

grzegorz914, 2024-09-13:
https://github.com/grzegorz914/homebridge-enphase-envoy?tab=readme-ov-file#about-the-plugin

Tiimoon, 2024-09-13:
If I understand correctly, you need to have an installer account to get the information instantly?

Tiimoon, 2024-09-13:
I can't understand your documentation. I am receiving the data from my panels, but I don't understand how to display them in Apple Home. I can't figure out what needs to be activated.

grzegorz914, 2024-09-13:
You can use also MQTT or RESTFul integration to read all data.

closed 2024-11-05 by grzegorz914