grzegorz914/homebridge-enphase-envoy #156: JWT token not valid, refreshing (forever)

thunder-god-cid, 2024-09-13:
Getting this constantly in logs. Known good credentials for Envoy. Also tried with generated token to no avail (same error). What am I doing wrong?

grzegorz914, 2024-09-13:
Wrong credentials data in config?

grzegorz914, 2024-11-05:
Still the issue ?

thunder-god-cid, 2024-11-05:
Yep - confirmed still with known good credentials.

grzegorz914, 2024-11-05:
But it’s working after refresh, get me the log please.

grzegorz914, 2024-11-24:
Please post the log.

LyricPants66133, 2024-11-28:
Hello @grzegorz914, 

Thank you for creating this plugin. I am having the same issue. 

Here is my system info:
IQ Gateway SKU: X-IQ-AM1-240-3
Firmware: D7.3.621

I have homebridge set up inside of a docker container. 

I have my envoy system set up wirelessly. envoy.local works properly, and my credentials give me proper access to my system. I have a very basic setup of the plug-in, with only my email, password, and S/N provided to the plug-in - as well as support for  Envoy with firmware v7.x.x and newer enabled.

Here are my logs for v9.5.2-beta.9. Please let me know if you need anything else to troubleshoot this issue. 

P.S. The plugin-in UI asks for the system password twice - this may be a visual bug. 

[11/27/2024, 9:47:23 PM] [homebridge-enphase-envoy] Restarting child bridge...
[11/27/2024, 9:47:23 PM] Got SIGTERM, shutting down child bridge process...
[11/27/2024, 9:47:28 PM] [homebridge-enphase-envoy] Child bridge process ended
[11/27/2024, 9:47:28 PM] [homebridge-enphase-envoy] Process Ended. Code: 143, Signal: null
[11/27/2024, 9:47:35 PM] [homebridge-enphase-envoy] Restarting Process...
[11/27/2024, 9:47:36 PM] [homebridge-enphase-envoy] Launched child bridge with PID 21140
[11/27/2024, 9:47:36 PM] [homebridge-enphase-envoy] Loaded homebridge-enphase-envoy v9.5.2-beta.9 child bridge successfully
[11/27/2024, 9:47:39 PM] [homebridge-enphase-envoy] Device: envoy.local Home Enphase, JWT Token refresh success.
[11/27/2024, 9:47:39 PM] [homebridge-enphase-envoy] Device: envoy.local Home Enphase, JWT Token valid: 11/27/2025, 9:47:39 PM
[11/27/2024, 9:47:41 PM] [homebridge-enphase-envoy] Device: envoy.local Home Enphase, Cookie not valid, refreshing.
[11/27/2024, 9:47:46 PM] [homebridge-enphase-envoy] Device: envoy.local Home Enphase, Error: Start error: Error: Check JWT token error: Validate JWT token error: getaddrinfo ENOTFOUND envoy.local, trying again.
[11/27/2024, 9:48:23 PM] [homebridge-enphase-envoy] Device: envoy.local Home Enphase, JWT Token refresh success.
[11/27/2024, 9:48:23 PM] [homebridge-enphase-envoy] Device: envoy.local Home Enphase, JWT Token valid: 11/27/2025, 9:48:23 PM
[11/27/2024, 9:48:25 PM] [homebridge-enphase-envoy] Device: envoy.local Home Enphase, Cookie not valid, refreshing.
[11/27/2024, 9:48:30 PM] [homebridge-enphase-envoy] Device: envoy.local Home Enphase, Error: Start error: Error: Check JWT token error: Validate JWT token error: getaddrinfo ENOTFOUND envoy.local, trying again.
[11/27/2024, 9:49:08 PM] [homebridge-enphase-envoy] Device: envoy.local Home Enphase, JWT Token refresh success.
[11/27/2024, 9:49:08 PM] [homebridge-enphase-envoy] Device: envoy.local Home Enphase, JWT Token valid: 11/27/2025, 9:49:08 PM
[11/27/2024, 9:49:10 PM] [homebridge-enphase-envoy] Device: envoy.local Home Enphase, Cookie not valid, refreshing.
[11/27/2024, 9:49:15 PM] [homebridge-enphase-envoy] Device: envoy.local Home Enphase, Error: Start error: Error: Check JWT token error: Validate JWT token error: getaddrinfo ENOTFOUND envoy.local, trying again.

grzegorz914, 2024-11-28:
Please put up address instead host please and let me know

grzegorz914, 2024-11-28:
Where ask for password twice, please post the screenshot please.

grzegorz914, 2024-11-28:
displays duplicate credentials fixed in 9.5.2, try and let me know

LyricPants66133, 2024-11-28:
Thank you for the quick response. The visual bug is fixed. 

---
Here is my attempt with a defined IP address, and login credentials provided:

[11/28/2024, 10:43:04 AM] [homebridge-enphase-envoy] Restarting child bridge...
[11/28/2024, 10:43:04 AM] Got SIGTERM, shutting down child bridge process...
[11/28/2024, 10:43:09 AM] [homebridge-enphase-envoy] Child bridge process ended
[11/28/2024, 10:43:09 AM] [homebridge-enphase-envoy] Process Ended. Code: 143, Signal: null
[11/28/2024, 10:43:16 AM] [homebridge-enphase-envoy] Restarting Process...
[11/28/2024, 10:43:17 AM] [homebridge-enphase-envoy] Launched child bridge with PID 17449
[11/28/2024, 10:43:18 AM] [homebridge-enphase-envoy] Loaded homebridge-enphase-envoy v9.5.2 child bridge successfully
[11/28/2024, 10:43:18 AM] [homebridge-enphase-envoy] Device: 192.168.1.100 Home Enphase, JWT Token expired, refreshing.
[11/28/2024, 10:43:50 AM] [homebridge-enphase-envoy] Device: 192.168.1.100 Home Enphase, Error: Start error: Error: Check JWT token error: Get JWT token error: Refresh token error: Error: Login to Enlighten error: Request failed with status code 401, trying again.
[11/28/2024, 10:44:03 AM] [homebridge-enphase-envoy] Device: 192.168.1.100 Home Enphase, JWT Token expired, refreshing.
[11/28/2024, 10:44:33 AM] [homebridge-enphase-envoy] Device: 192.168.1.100 Home Enphase, Error: Start error: Error: Check JWT token error: Get JWT token error: Refresh token error: Error: Login to Enlighten error: Request failed with status code 401, trying again.
[11/28/2024, 10:44:48 AM] [homebridge-enphase-envoy] Device: 192.168.1.100 Home Enphase, JWT Token expired, refreshing.
[11/28/2024, 10:45:18 AM] [homebridge-enphase-envoy] Device: 192.168.1.100 Home Enphase, Error: Start error: Error: Check JWT token error: Get JWT token error: Refresh token error: Error: Login to Enlighten error: Request failed with status code 401, trying again.
[11/28/2024, 10:45:33 AM] [homebridge-enphase-envoy] Device: 192.168.1.100 Home Enphase, JWT Token expired, refreshing.
[11/28/2024, 10:46:03 AM] [homebridge-enphase-envoy] Device: 192.168.1.100 Home Enphase, Error: Start error: Error: Check JWT token error: Get JWT token error: Refresh token error: Error: Login to Enlighten error: Request failed with status code 401, trying again.

---

Here is my attempt with a defined IP address, and valid token provided:

[11/28/2024, 10:58:41 AM] [homebridge-enphase-envoy] Restarting child bridge...
[11/28/2024, 10:58:41 AM] Got SIGTERM, shutting down child bridge process...
[11/28/2024, 10:58:46 AM] [homebridge-enphase-envoy] Child bridge process ended
[11/28/2024, 10:58:46 AM] [homebridge-enphase-envoy] Process Ended. Code: 143, Signal: null
[11/28/2024, 10:58:53 AM] [homebridge-enphase-envoy] Restarting Process...
[11/28/2024, 10:58:54 AM] [homebridge-enphase-envoy] Launched child bridge with PID 18197
[11/28/2024, 10:58:55 AM] [homebridge-enphase-envoy] Loaded homebridge-enphase-envoy v9.5.2 child bridge successfully
[11/28/2024, 10:58:55 AM] [homebridge-enphase-envoy] Device: 192.168.1.100 Home Enphase, Cookie refresh success.
[11/28/2024, 10:59:11 AM] [homebridge-enphase-envoy] Device: 192.168.1.100 Home Enphase, Error: Start error: Error: Update power production state error: Request failed with status code 401, trying again.
[11/28/2024, 10:59:40 AM] [homebridge-enphase-envoy] Device: 192.168.1.100 Home Enphase, Cookie refresh success.
[11/28/2024, 10:59:56 AM] [homebridge-enphase-envoy] Device: 192.168.1.100 Home Enphase, Error: Start error: Error: Update power production state error: Request failed with status code 401, trying again.
[11/28/2024, 11:00:25 AM] [homebridge-enphase-envoy] Device: 192.168.1.100 Home Enphase, Cookie refresh success.
[11/28/2024, 11:00:38 AM] [homebridge-enphase-envoy] Device: 192.168.1.100 Home Enphase, Error: Start error: Error: Update power production state error: Request failed with status code 401, trying again.

---

Also to note, when trying to switch from credentials to token, homebridge reported: "Error failed to save config". This issue only arises when switching directly from Generated Token to Enlighten Credentials, and vice versa. When None is selected first, saved, and then switched to the new desired Token Generation Mode, this issue does not happen. 

My steps in reproducing are:

reset to the default json file configuration for the plugin (no credentials data provided)
Setup the plugin using the UI for either token or credentials
Save the config, and allow the plugin to start
Edit the plugin config, go to credentials data, and change the Token generation mode to the other mode( do not select none)
Enter the new Token generation credentials(or token). 
Save - > gives error

grzegorz914, 2024-11-28:
Check your credentials data first, error 401 is wrong user or password.

LyricPants66133, 2024-11-28:
When I try to sign in to the website https://enlighten.enphaseenergy.com/login/login, I get a json response "message: Too many active sessions..". Could it be that running the plugin overnight left too many sessions running, without closing them? I am using the same credentials as yesterday.

LyricPants66133, 2024-11-28:
I am unsure of how I could close my active sessions so that I can troubleshoot again.

grzegorz914, 2024-11-28:
The issue with config save cannot be reproduced on my side, all is saved correct. 
Looks like you have temporary account locked, wait little and try again.

grzegorz914, 2024-11-28:
Im pushed 9.5.3 wit some small changes, try and let me know

LyricPants66133, 2024-11-28:
9.5.3 Worked to an extent. I am still unable to log in to my account due to the active session issue, but the plugin was able to read some information with the provided token. Attached is my log

[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/17951994/homebridge-enphase-envoy.log.txt)

closed 2025-01-06 by github-actions[bot]