grzegorz914/homebridge-enphase-envoy #159: Profile status not showing current profile or able to set profile

echamberlain, 2024-09-17:
I'm not able to view the current profile or change the profile.  The controls all appear as off/closed, regardless of actual profile state.

What's info do you need to troubleshoot the issue?

grzegorz914, 2024-09-17:
Debug log

echamberlain, 2024-09-18:
`[17/09/2024, 17:26:20] [homebridge-enphase-envoy] Device: envoy.local Envoy, Encharges profile: self-consumption, state: Not Active
[17/09/2024, 17:26:20] [homebridge-enphase-envoy] Device: envoy.local Envoy, Production power level: 18.925333333333334 %
[17/09/2024, 17:26:20] [homebridge-enphase-envoy] Device: envoy.local Envoy, Production power state: Enabled
[17/09/2024, 17:26:20] [homebridge-enphase-envoy] Device: envoy.local Envoy, Encharges profile: savings-mode, reserve: undefined %
[17/09/2024, 17:26:20] [homebridge-enphase-envoy] Device: envoy.local Envoy, Encharges profile: savings-mode, state: OFF
[17/09/2024, 17:26:20] [homebridge-enphase-envoy] Device: envoy.local Envoy, Encharges profile: backup, state: OFF
[17/09/2024, 17:26:20] [homebridge-enphase-envoy] Device: envoy.local Envoy, Encharges profile: backup, reserve: undefined %
[17/09/2024, 17:26:20] [homebridge-enphase-envoy] Device: envoy.local Envoy, Encharges profile: self-consumption, state: OFF
[17/09/2024, 17:26:20] [homebridge-enphase-envoy] Device: envoy.local Envoy, Encharges profile: self-consumption, reserve: undefined %
[17/09/2024, 17:26:20] [homebridge-enphase-envoy] Device: envoy.local Envoy, Encharges energy state: Charged
[17/09/2024, 17:26:20] [homebridge-enphase-envoy] Device: envoy.local Envoy, Encharges energy level: 100 %
[17/09/2024, 17:26:20] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Brightness': characteristic value expected valid finite number and received "undefined" (undefined). See https://homebridge.io/w/JtMGR for more info.
[17/09/2024, 17:26:20] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Brightness': characteristic value expected valid finite number and received "undefined" (undefined). See https://homebridge.io/w/JtMGR for more info.
[17/09/2024, 17:26:20] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Brightness': characteristic value expected valid finite number and received "undefined" (undefined). See https://homebridge.io/w/JtMGR for more info.`

echamberlain, 2024-09-18:
Here is the debug log.
[homebridge.log.txt](https://github.com/user-attachments/files/17036775/homebridge.log.txt)

grzegorz914, 2024-09-18:
To get correct debug log You need go to plugin settings Advanced Settings >> System >> Debug enable it and then get the log.

grzegorz914, 2024-09-18:
try 9.4.1-beta and get me the debug log

grzegorz914, 2024-11-05:
Close as no response from user.

closed 2024-11-05 by grzegorz914