grzegorz914/homebridge-enphase-envoy #161: Enphase Plugin Stops Updating After Night Without Restart

BenouGui, 2024-10-16:
Hi,

I'm using the Enphase plugin to send data from my solar panels to MQTT. Everything works well during the day, but after a night of no solar production, the plugin stops updating, and I no longer receive any panel data. To fix this, I need to restart the plugin each morning.

Additionally, it's quite difficult to pinpoint in the logs exactly when the plugin stops updating or goes inactive. This makes troubleshooting more complicated.

Could you please look into this issue? Let me know if you need any logs or further details.

Thank you for your help!

[image]

grzegorz914, 2024-10-16:
Try latest beta and let me know

grzegorz914, 2024-10-17:
beta 14 will try reconnect if error occurs, please let me know tomorrow after test.

BenouGui, 2024-10-19:
Thank you for the suggestion!

Yesterday, I installed beta 16, but unfortunately, this morning the plugin stopped updating again, and I had to reboot it to get it working. It seems the issue still persists.

I've attached the logs from this morning:

The first part shows when the plugin stopped updating.
The second part is from after I rebooted the plugin.
Let me know if there's anything else I should try or provide.

Thanks again for your help!

[homebridge.log.before.txt](https://github.com/user-attachments/files/17444447/homebridge.log.before.txt)
[homebridge.log.after.txt](https://github.com/user-attachments/files/17444448/homebridge.log.after.txt)

grzegorz914, 2024-10-19:
Your envoy just doesn’t respond to the request, as You see the plugin trying to connect again without success. May be You set to short fats refresh time.

BenouGui, 2024-10-19:
Thank you for the response!

I understand that the plugin is trying to connect without success, but I’m wondering why everything works perfectly after I restart the plugin, and I didn’t have this issue before. If the refresh time was too short, wouldn’t the problem persist even after restarting?

I’ll try increasing the refresh time and see if that makes a difference. Let me know if there’s anything else I should check.

Thanks again for your help!

grzegorz914, 2024-10-19:
I testing this on my side and is working correct as of beta 16, I will investigate more time for this issue.

BenouGui, 2024-10-22:
Just an update:

This morning, with version v9.4.1-beta.47, I didn’t need to reboot the plugin, and everything worked as expected. I’ll continue monitoring, but it looks promising so far.

Thanks again for your help!

grzegorz914, 2024-10-26:
I think latest beta.77 solved the issue, any info from your side.?

BenouGui, 2024-10-27:
Yes, with beta 77 everything worked fine, no issues to report. I’ve just updated to beta 78 and will check tomorrow morning to see if everything is still stable.

Thanks for following up!

grzegorz914, 2024-10-30:
v9.4.1 released, I think is fixed finally.

grzegorz914, 2024-11-03:
@BenouGui any info ?

BenouGui, 2024-11-04:
Everything is working well now—no issues to report. Thanks for the support and the updates!

closed 2024-11-05 by grzegorz914