grzegorz914/homebridge-enphase-envoy #163: Error for version 9 and above

rrebel01, 2024-11-15:
Iam keep getting this error if i upgrade from v8.1.1 and above:

[11/14/2024, 9:19:33 PM] [homebridge-enphase-envoy] Device: 192.168.250.53 Envoy Rebel, Error: Start error: Error: Update ensemble status error: Request failed with status code 401, trying again.

V8.1.1 seems to work fine. All versions higher are not working.

grzegorz914, 2024-11-15:
Can you post debug log please?

rrebel01, 2024-11-17:
what is the best way to find thedebug log?

grzegorz914, 2024-11-17:
Go to plugin config, Advanced Settings>>>System>>>Debug enable it, save and restart, then download the log and post.

rrebel01, 2024-11-17:
Where is the log saved?

rrebel01, 2024-11-17:
[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/17791018/homebridge-enphase-envoy.log.txt)

rrebel01, 2024-11-17:
[Uploading homebridge-enphase-envoy.log.txt…]()

rrebel01, 2024-11-17:
[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/17791033/homebridge-enphase-envoy.log.txt)

grzegorz914, 2024-11-17:
Fixed in 9.4.3

rrebel01, 2024-11-17:
Yes it works! thanks for the fast and great service!!

closed 2024-11-17 by grzegorz914

rrebel01, 2024-11-17:
Just found out that i lost my 2 sensors in Homekit. 

config is still there but sensors not visable:

     "displayType": 0
                    },
                    "powerConsumptionNetLevelSensors": [
                        {
                            "name": ">5Kw to Net",
                            "powerLevel": -5000,
                            "displayType": 0
                        },
                        {
                            "name": ">4.4Kw to Net",
                            "powerLevel": -4400,
                            "displayType": 0
                        },
                        {
                            "powerLevel": 0,
                            "displayType": 0

rrebel01, 2024-11-17:
[image]

grzegorz914, 2024-11-17:
You need enable it, display type `0` mean `disabled`

rrebel01, 2024-11-17:
I re-enabled it and changed it back to contact. But my shortcuts in homekit have disappared. Those shortcuts where pretty complex. Is there a way to get them back?