grzegorz914/homebridge-enphase-envoy #164: Negative power level sensors are seen as positive.

rrebel01, 2024-11-18:
In older versions i had contacts to trigger my waterheater if solor production excees 4400W. 

After upgrade contact keeps trigger at consumption instead of productions. See attachments.
[image]
[image]
[image]

grzegorz914, 2024-11-18:
In Your case you neeed to create this sensors in Production section.

rrebel01, 2024-11-18:
I need production - consumption.

if total flow to net is greater then 4400w.

i always did this on your advise to use a negative number in the powerlevel sensor. What would be the best way now?

grzegorz914, 2024-11-18:
You have 3 options to create sensors:
Production - always positive, PV
Consumption Total - always positive, Home
Consumption Net - positive or negative based on current power, Grid

rrebel01, 2024-11-18:
Consumption Net is what i use, but its not handling negative numbers correctly:

[image]

grzegorz914, 2024-11-18:
Can you post debug, I will look into this issue.

grzegorz914, 2024-11-18:
Try 9.4.5-beta.0 and let me know

rrebel01, 2024-11-19:
At this monent the contacts dont seem to react to positive nor negative values

grzegorz914, 2024-11-19:
I have already debug this on my side and all working as expected, if doesn’t work on your side please post debug log

rrebel01, 2024-11-19:
her you go:

[image]

[image]

[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/17819467/homebridge-enphase-envoy.log.txt)

grzegorz914, 2024-11-19:
All correct:
Here the power:
[image]
Here the sensors;
[image]

rrebel01, 2024-11-19:
Sensors have not been active in homekit all day, ive put notification on it. they have not been triggered i will delete the bridge and rejoin it in homekit, to see if that helps

rrebel01, 2024-11-20:
ive removed the bridge from homekit and rejoined. But contacts in homekit dont change state.

grzegorz914, 2024-11-20:
Really strange, on my side all is working correct and no any other user report problem with this

[image]

rrebel01, 2024-11-20:
Ive added a positive sensor as a test and get this error now:

/homebridge/node_modules/homebridge-enphase-envoy/index.js:41
          log.warn(`Device: ${host} ${deviceName}, Name missing: ${deviceName}.`);
                              ^
ReferenceError: Cannot access 'host' before initialization
    at HomebridgeAPI.<anonymous> (/homebridge/node_modules/homebridge-enphase-envoy/index.js:41:31)
    at HomebridgeAPI.emit (node:events:514:28)
    at HomebridgeAPI.signalFinished (/homebridge/node_modules/homebridge/src/api.ts:275:10)
    at ChildBridgeFork.startBridge (/homebridge/node_modules/homebridge/src/childBridgeFork.ts:194:14)
[11/20/2024, 10:57:53 AM] [homebridge-enphase-envoy] Child bridge process ended
[11/20/2024, 10:57:53 AM] [homebridge-enphase-envoy] Process Ended. Code: 1, Signal: null

grzegorz914, 2024-11-20:
Post your config please

rrebel01, 2024-11-20:
{
    "devices": [
        {
            "name": "Envoy Rebel",
            "host": "192.168.250.53",
            "envoyFirmware7xx": true,
            "envoyFirmware7xxTokenGenerationMode": 0,
            "envoySerialNumber": "122033006047",
            "enlightenUser": "xxxxxl",
            "enlightenPasswd": "xxxx",
            "supportPlcLevel": false,
            "plcLevelControl": {
                "displayType": 0
            },
            "supportPowerProductionState": false,
            "powerProductionStateControl": {
                "displayType": 0
            },
            "powerProductionSummary": 11500,
            "powerProductionStateSensor": {
                "displayType": 0
            },
            "powerProductionLevelSensors": [
                {
                    "powerLevel": 0,
                    "displayType": 0
                }
            ],
            "energyProductionStateSensor": {
                "displayType": 0
            },
            "energyProductionLevelSensors": [
                {
                    "energyLevel": 0,
                    "displayType": 0
                }
            ],
            "energyProductionLifetimeOffset": 0,
            "powerConsumptionTotalStateSensor": {
                "displayType": 0
            },
            "powerConsumptionTotalLevelSensors": [
                {
                    "powerLevel": 0,
                    "displayType": 0
                }
            ],
            "energyConsumptionTotalLifetimeOffset": 0,
            "energyConsumptionTotalStateSensor": {
                "displayType": 0
            },
            "energyConsumptionTotalLevelSensors": [
                {
                    "energyLevel": 0,
                    "displayType": 0
                }
            ],
            "powerConsumptionNetStateSensor": {
                "displayType": 0
            },
            "powerConsumptionNetLevelSensors": [
                {
                    "name": ">5Kw to Net",
                    "compareMode": 0,
                    "powerLevel": -5000,
                    "displayType": 3
                },
                {
                    "name": ">4.4Kw to Net",
                    "compareMode": 0,
[... 214 lines cut ...]
                "displayType": 0
            },
            "dataRefreshSensor": {
                "displayType": 0
            },
            "enableDebugMode": false,
            "disableLogInfo": false,
            "disableLogDeviceInfo": false,
            "restFul": {
                "enable": false,
                "debug": false
            },
            "mqtt": {
                "enable": false,
                "debug": false,
                "auth": false
            }
        }
    ],
    "_bridge": {
        "username": "0E:3B:59:9C:9E:4E",
        "port": 55365
    },
    "platform": "enphaseEnvoy"
}

grzegorz914, 2024-11-20:
Pleas do not use special characters in `name` like `>` and `.`

grzegorz914, 2024-11-20:
Reference error is fixed in 9.5.1

grzegorz914, 2024-11-20:
Please change refresh time, the time is in seconds, You have set 1000 sec, I think this is Your problem, because the data is refresh every 1000 seconds.

rrebel01, 2024-11-20:
Let me try, those used to be ms in the past?

grzegorz914, 2024-11-20:
Was long ago changed to sec.

rrebel01, 2024-11-21:
Seems we are getting there. just get this error in the logs:
Device: 192.168.250.53 Envoy Rebel, Impulse generator: Error: Update production ct error: timeout of 20000ms exceeded

grzegorz914, 2024-11-21:
To many request to Envoy, please increase refresh time for production. Envoy have very poor CPU.

closed 2024-11-24 by grzegorz914