grzegorz914/homebridge-enphase-envoy #166: Certificate error

Aldunnn, 2024-11-29:
I have an error when starting the plugin.

[11/29/2024, 5:14:02 PM] [homebridge-enphase-envoy] Device: 192.168.1.75 Solar, Error: Start error: Error: Update home error: self-signed certificate, trying again.

grzegorz914, 2024-11-29:
What a envoy firmware, debug log please

Aldunnn, 2024-11-30:
Thank you for the fast reply ENVOY firmware D8.2.4264
[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/17965068/homebridge-enphase-envoy.log.txt)

grzegorz914, 2024-11-30:
You need enable,support for FW 7.x.x and newer in plugin config.

Aldunnn, 2024-11-30:
I have checked the Advanced Settings > Envoy box for Firmware 7.x.x.  Please note that I do not have any credentials provided by the installer of the system.
I see that there are three options, and for now, I have selected No
[homebridge-enphase-envoy.log-2.txt](https://github.com/user-attachments/files/17965458/homebridge-enphase-envoy.log-2.txt)
ne. 
So I need a token, but how to get one?

Aldunnn, 2024-11-30:
Ok, I found some documentation on getting a token.  https://enphase.com/download/accessing-iq-gateway-local-apis-or-local-ui-token-based-authentication 
So my token seems to be accepted.  But now I get the message
[11/30/2024, 2:21:45 PM] [homebridge-enphase-envoy] Device: 192.168.1.99 Solar, Requesting check JWT token.
Reading the other issues, this is connected to credentials?  Does this mean I need to request credentials from the installer?

grzegorz914, 2024-11-30:
You need to select Token Generation Mode >> Enlighten Credentials and add envoy serial number, user, and password, same as You use for login to https://enlighten.enphaseenergy.com/

closed 2024-12-17 by grzegorz914