grzegorz914/homebridge-enphase-envoy #169: Issue release after 9.4.0

bbleyzat, 2024-12-04:
Hello, 

I have a weird issue for **all releases after 9.4.0.** 
When I reboot the plug in everything worked fine but after a regular day in Homekit I am not able to see solar panels status. They never started up.

Home bridge version: 1.8.5
UI: 6.64.0
Node: 22.12.0

If I rollback to 9.4.0 there is a normal behavior.

Thanks in advance for your help.
I didn't join the log because I didn't see anything relevant here.

bbleyzat, 2024-12-04:
[12/4/2024, 2:22:43 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production power peak reset: Off
[12/4/2024, 2:25:47 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production power state: Enabled
[12/4/2024, 2:25:47 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production power level: 43.94666666666667 %
[12/4/2024, 2:25:55 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production power state: Enabled
[12/4/2024, 2:29:54 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production power level: 43.94666666666667 %
[12/4/2024, 2:29:54 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production power state: Enabled
[12/4/2024, 2:30:26 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production power state: Enabled
[12/4/2024, 2:30:26 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production power level: 43.94666666666667 %
[12/4/2024, 2:30:30 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production power state: Enabled

For instance here, with the latest plug in once initialized, the % is always the same until i reboot the plug in.

bbleyzat, 2024-12-04:
with 9.4.0:
[12/4/2024, 2:34:06 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production power level: 44.61333333333334 %
[12/4/2024, 2:34:06 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production power state: Enabled
[12/4/2024, 2:34:13 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production power state: Enabled
[12/4/2024, 2:36:11 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production power state: Enabled
[12/4/2024, 2:36:11 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production power level: 43.70666666666667 %

No issue, updating correctly

grzegorz914, 2024-12-04:
Are you trying latest version? 9.6.5

bbleyzat, 2024-12-04:
Hey, I tried with all versions. The log where it didn't work were from the latest.

grzegorz914, 2024-12-04:
I need  the debug log with this issue

bbleyzat, 2024-12-04:
Sure here the last reboot. Stuck at 1%.
[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/18011385/homebridge-enphase-envoy.log.txt)

grzegorz914, 2024-12-04:
Post your config please

bbleyzat, 2024-12-04:
"devices": [
                {
                    "name": "Énergie solaire",
                    "envoyFirmware7xx": true,
                    "envoyFirmware7xxTokenGenerationMode": 0,
                    "envoyPasswd": "XXXXX",
                    "envoyToken": "XXXXX",
                    "envoyTokenInstaller": false,
                    "envoySerialNumber": "XXXXX",
                    "enlightenUser": "XXXXX",
                    "enlightenPasswd": "XXXXXX",
                    "supportPlcLevel": false,
                    "plcLevelControl": {
                        "displayType": 3,
                        "namePrefix": true
                    },
                    "supportPowerProductionState": false,
                    "powerProductionStateControl": {
                        "displayType": 1,
                        "namePrefix": true
                    },
                    "powerProductionSummary": 3750,
                    "powerProductionStateSensor": {
                        "displayType": 0,
                        "namePrefix": false
                    },
                    "powerProductionLevelSensors": [
                        {
                            "displayType": 0
                        }
                    ],
                    "energyProductionStateSensor": {
                        "displayType": 0
                    },
                    "energyProductionLevelSensors": [
                        {
                            "energyLevel": 0,
                            "displayType": 0
                        }
                    ],
                    "energyProductionLifetimeOffset": 0,
                    "powerConsumptionTotalStateSensor": {
                        "displayType": 0
                    },
                    "powerConsumptionTotalLevelSensors": [
                        {
                            "powerLevel": 0,
                            "displayType": 0
                        }
                    ],
                    "energyConsumptionTotalLifetimeOffset": 0,
                    "energyConsumptionTotalStateSensor": {
                        "displayType": 0
                    },
                    "energyConsumptionTotalLevelSensors": [
                        {
                            "energyLevel": 0,
                            "displayType": 0
                        }
                    ],
                    "powerConsumptionNetStateSensor": {
                        "displayType": 0
                    },
                    "powerConsumptionNetLevelSensors": [
                        {
                            "powerLevel": 0,
                            "displayType": 0
                        }
                    ],
                    "energyConsumptionNetLifetimeOffset": 0,
[... 74 lines cut ...]
                    "metersDataRefreshTime": 60,
                    "productionDataRefreshTime": 60,
                    "liveDataRefreshTime": 60,
                    "ensembleDataRefreshTime": 60,
                    "enableDebugMode": false,
                    "disableLogInfo": false,
                    "disableLogDeviceInfo": false,
                    "restFul": {
                        "enable": false,
                        "debug": false
                    },
                    "mqtt": {
                        "enable": false,
                        "debug": false,
                        "auth": false
                    }
                }
            ],
            "_bridge": {
                "username": "XXXXXX",
                "port": 52259
            },
            "platform": "enphaseEnvoy"
            
            Basic config at the end :)

bbleyzat, 2024-12-04:
maybe link to the issue #170 same behavior (freeze)

grzegorz914, 2024-12-04:
In your debug log the production is only call on start, please restart homebridge and get the whole log again. On my side all is working correct. Very strange.

grzegorz914, 2024-12-04:
Also trybo reduce production data refresh time

bbleyzat, 2024-12-04:
Sure I'll keep you posted tomorrow!

bbleyzat, 2024-12-05:
@grzegorz914 what is the best option for all refresh parameters please?

grzegorz914, 2024-12-05:
Here my settings:
Meters 6sec, live data 8sec, production 12sec, ensemble 15 sec

The envoy have poor CPU and if you set to short refresh time then is possible to freeze it.