grzegorz914/homebridge-enphase-envoy #170: Freeze

Tonal1ty, 2024-12-04:
Hi since lasts days i see the sensor and the soldat status is freeze, when i restart plugin , all refrech and freeze immediatly

Tonal1ty, 2024-12-04:
[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/18011605/homebridge-enphase-envoy.log.txt)

Tonal1ty, 2024-12-04:
> Hi since lasts days i see the sensor and the solar status is freeze, when i restart plugin , all refrech and freeze immediatly

bbleyzat, 2024-12-04:
Try to downgrade to 9.4.0 and see if it solved your issue, I've got the same problem.

Tonal1ty, 2024-12-06:
Yes, its Work in 9.4.0 but dont workd on the lastest

bbleyzat, 2024-12-06:
Could you provide your refresh settings under [SYSTEM] > Data Refresh

Tonal1ty, 2024-12-06:
[image]

grzegorz914, 2024-12-06:
https://github.com/grzegorz914/homebridge-enphase-envoy/issues/169#issuecomment-2521152231

grzegorz914, 2024-12-06:
> ![IMG_4134](https://private-user-images.githubusercontent.com/150700152/393176944-987f0870-d14e-440f-b894-3c56fea97968.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM0NzkzMzMsIm5iZiI6MTczMzQ3OTAzMywicGF0aCI6Ii8xNTA3MDAxNTIvMzkzMTc2OTQ0LTk4N2YwODcwLWQxNGUtNDQwZi1iODk0LTNjNTZmZWE5Nzk2OC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwNlQwOTU3MTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04Mjk4MDZlODNhNDE3NTE1OGVkNTFkNGYwMDk4ZmM3NTU5ODE5NDY0YmVlNzM1ZDEzY2E3NWY3MzQ1OGYzMjhlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.jXZIM28Er90KJv23p34JycDHYm90DfizScWZzNXum7A)

Your settings for ensemble is wrong, 15000 second? Also reducie other settings to get the data more real time

bbleyzat, 2024-12-06:
Production put 30s, I hjad the exact same issue. Now everything worked perfectly :)