grzegorz914/homebridge-enphase-envoy #172: Update inventory error: timeout of 25000ms exceeded

eweiman, 2024-12-12:
Plugin version 9.6.7
Homebridge v1.8.5 (HAP v0.12.3)

I've noted that this is getting logged continuously as of late. 
I've only had the system for a month now so I'm not sure if this is really "new" or that I've just realized it was happening now. 

``` 

[12/12/2024, 3:21:04 PM] Registering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[12/12/2024, 3:21:04 PM] [homebridge-enphase-envoy] Loaded homebridge-enphase-envoy v9.6.7 child bridge successfully
[12/12/2024, 3:21:04 PM] Homebridge v1.8.5 (HAP v0.12.3) (homebridge-enphase-envoy) is running on port 33065.
[12/12/2024, 3:21:07 PM] [homebridge-enphase-envoy] Device: envoy.e4.home Enphase, JWT Token refresh success.
[12/12/2024, 3:21:07 PM] [homebridge-enphase-envoy] Device: envoy.e4.home Enphase, JWT Token user, valid: 12/12/2025, 3:21:07 PM
[12/12/2024, 3:21:09 PM] [homebridge-enphase-envoy] Device: envoy.e4.home Enphase, Cookie not valid, validating.
[12/12/2024, 3:21:09 PM] [homebridge-enphase-envoy] Device: envoy.e4.home Enphase, Cookie validate success.
[12/12/2024, 3:21:32 PM] [homebridge-enphase-envoy] Device: envoy.e4.home Enphase, Connect Success.
[12/12/2024, 3:21:32 PM] [homebridge-enphase-envoy] -------- Enphase --------
[12/12/2024, 3:21:32 PM] [homebridge-enphase-envoy] Manufacturer: Enphase
[12/12/2024, 3:21:32 PM] [homebridge-enphase-envoy] Model: 800-00664-r05
[12/12/2024, 3:21:32 PM] [homebridge-enphase-envoy] Firmware: D8.2.127
[12/12/2024, 3:21:32 PM] [homebridge-enphase-envoy] SerialNr: *****************
[12/12/2024, 3:21:32 PM] [homebridge-enphase-envoy] Time: 12/12/2024, 3:20:38 PM
[12/12/2024, 3:21:32 PM] [homebridge-enphase-envoy] ------------------------------
[12/12/2024, 3:21:32 PM] [homebridge-enphase-envoy] Inverters: 40
[12/12/2024, 3:21:32 PM] [homebridge-enphase-envoy] Meters: Yes
[12/12/2024, 3:21:32 PM] [homebridge-enphase-envoy] Production: Enabled
[12/12/2024, 3:21:32 PM] [homebridge-enphase-envoy] --------------------------------
[12/12/2024, 3:21:32 PM] [homebridge-enphase-envoy] Wireless Kit: 1
[12/12/2024, 3:21:32 PM] [homebridge-enphase-envoy] --------------------------------
[12/12/2024, 3:21:32 PM] [homebridge-enphase-envoy] Device: envoy.e4.home Enphase, Published as external accessory.
[12/12/2024, 3:21:32 PM] [homebridge-enphase-envoy] Device: envoy.e4.home Enphase, Impulse generator started.
...skipping a bunch of info...
[12/12/2024, 3:22:14 PM] [homebridge-enphase-envoy] Device: envoy.e4.home Enphase, Production power level: 11.973407894736843 %
[12/12/2024, 3:22:14 PM] [homebridge-enphase-envoy] Device: envoy.e4.home Enphase, Production power state: Enabled
[12/12/2024, 3:28:57 PM] [homebridge-enphase-envoy] Device: envoy.e4.home Enphase, Impulse generator: Error: Update home error: timeout of 25000ms exceeded
...skipping a bunch of the same messages...
[12/12/2024, 4:31:58 PM] [homebridge-enphase-envoy] Device: envoy.e4.home Enphase, Impulse generator: Error: Update home error: timeout of 25000ms exceeded
[12/12/2024, 4:32:58 PM] [homebridge-enphase-envoy] Device: envoy.e4.home Enphase, Impulse generator: Error: Update home error: timeout of 25000ms exceeded
[12/12/2024, 4:35:58 PM] [homebridge-enphase-envoy] Device: envoy.e4.home Enphase, Impulse generator: Error: Update home error: timeout of 25000ms exceeded
[12/12/2024, 4:36:58 PM] [homebridge-enphase-envoy] Device: envoy.e4.home Enphase, Impulse generator: Error: Update home error: timeout of 25000ms exceeded
[12/12/2024, 4:39:58 PM] [homebridge-enphase-envoy] Device: envoy.e4.home Enphase, Impulse generator: Error: Update home error: timeout of 25000ms exceeded
```

Restarting homebridge makes it OK for about 5-10 minutes i'd say, then this comes up again. 
It seems like once it starts that error it never recovers / updates with any current power values. 

Now even with the sun down homebridge thinks I'm still making 11% of production. 

Solar wise i have the IQ Gateway (wireless) + 40 IQ8X microinverters

friendofasquid, 2024-12-15:
I'm having the same issue. I not that the local web interface to the Envoy is very slow, and itself times out. I opened a support ticket and this was their reply:

> The envoy local API is not meant to be used as your main information area for your system. The app and online website is meant for that for homeowners. As such, the local API is very slow, even at the best of sites. We don't recommend using it.

I think we just need that timeout to be configurable, if possible. I can see it is hard-coded in the codebase.

grzegorz914, 2024-12-16:
Can you post your config.json

friendofasquid, 2024-12-16:
```json
{
    "devices": [
        {
            "name": "Enphase",
            "host": "envoy.local",
            "envoyFirmware7xx": true,
            "envoyFirmware7xxTokenGenerationMode": 1,
            "envoyToken": "<redacted>",
            "envoyTokenInstaller": false,
            "envoySerialNumber": "<redacted>",
            "enlightenUser": "<redacted>",
            "enlightenPasswd": "<redacted>",
            "supportPlcLevel": false,
            "plcLevelControl": {
                "displayType": 0
            },
            "supportPowerProductionState": false,
            "powerProductionStateControl": {
                "displayType": 0
            },
            "powerProductionStateSensor": {
                "displayType": 0
            },
            "powerProductionLevelSensors": [
                {
                    "displayType": 0
                }
            ],
            "energyProductionStateSensor": {
                "displayType": 0
            },
            "energyProductionLevelSensors": [
                {
                    "displayType": 0
                }
            ],
            "powerConsumptionTotalStateSensor": {
                "name": "Total Power Consumption",
                "displayType": 1
            },
            "powerConsumptionTotalLevelSensors": [
                {
                    "name": "Power > 5000 Watts",
                    "compareMode": 0,
                    "powerLevel": 5000,
                    "displayType": 1
                }
            ],
            "energyConsumptionTotalStateSensor": {
                "displayType": 0
            },
            "energyConsumptionTotalLevelSensors": [
                {
                    "displayType": 0
                }
            ],
            "powerConsumptionNetStateSensor": {
                "name": "Power State Sensor",
                "displayType": 1
            },
            "powerConsumptionNetLevelSensors": [
                {
                    "name": "Exporting to Grid",
                    "compareMode": 3,
                    "powerLevel": 0,
                    "displayType": 3
                }
            ],
            "energyConsumptionNetStateSensor": {
                "displayType": 0
            },
            "energyConsumptionNetLevelSensors": [
                {
                    "displayType": 0
                }
            ],
            "enchargeProfileControl": [
                {
                    "profile": "backup",
                    "displayType": 0
                }
            ],
            "enchargeProfileSensors": [
                {
                    "displayType": 0
                }
            ],
            "enchargeStateSensor": {
                "displayType": 0
            },
            "enchargeGridModeSensors": [
                {
                    "displayType": 0
                }
            ],
            "enchargeBackupLevelSensors": [
                {
                    "displayType": 0
                }
            ],
            "enpowerGridStateControl": {
                "displayType": 0
            },
            "enpowerGridStateSensor": {
                "displayType": 0
            },
            "enpowerGridModeSensors": [
                {
                    "displayType": 0
                }
            ],
            "enpowerDryContactsControl": false,
            "enpowerDryContactsSensors": false,
            "solarGridModeSensors": [
                {
                    "displayType": 0
                }
            ],
            "generatorStateControl": {
                "displayType": 0
            },
            "generatorStateSensor": {
                "displayType": 0
            },
            "generatorModeControl": [
                {
                    "mode": "off",
                    "displayType": 0
                }
            ],
            "generatorModeSensors": [
                {
                    "displayType": 0
                }
            ],
            "dataRefreshControl": {
                "displayType": 0
            },
            "dataRefreshSensor": {
                "displayType": 0
            },
            "enableDebugMode": false,
            "disableLogInfo": false,
            "disableLogDeviceInfo": false,
            "restFul": {
                "enable": false,
                "debug": false
            },
            "mqtt": {
                "enable": false,
                "debug": false,
                "auth": false
            }
        }
    ],
    "_bridge": {
        "username": "0E:74:7A:C9:CF:93",
        "port": 46195
    },
    "platform": "enphaseEnvoy"
}
```

grzegorz914, 2024-12-16:
Please go to plugin settings and set refresh time for:
meters 6sec, live data 8sec, production 12sec, ensemble 15sec

friendofasquid, 2024-12-16:
As I went to change those parameters, it was working again. No idea why.

grzegorz914, 2024-12-16:
Because default refresh time is set to short for poor envoy CPU, i will change it on next update.

closed 2024-12-17 by grzegorz914