grzegorz914/homebridge-enphase-envoy #175: Error: Cannot add more than 100 services to a single accessory!

ipendle, 2025-01-02:
Thanks for creating this plugin. I value what you are doing for the open source community.

I'd like to try out your plugin. I got past the credentials but now I get this error. I think it is because I have a pretty big system (126 panels on the roof a large workshop). 

I get this error:
`[1/2/2025, 8:01:10 AM] [homebridge-enphase-envoy] Device: envoy.local Solar, Error: Start error: Error: Prepare accessory error: Error: Cannot add more than 100 services to a single accessory!, trying again.`

Then this one repeats:
`[1/2/2025, 8:01:47 AM] [homebridge-enphase-envoy] Device: envoy.local Solar, Error: Start error: Error: Update inventory error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'), trying again.`

Any way to increase the limit?

grzegorz914, 2025-01-02:
Unfortunately this is HomeKit limitation, it can not be increased by me.

ipendle, 2025-01-02:
Thanks for the quick response. Is there any way for me to limit what HomeKit tries to see (ie I dont care about all the panels, just overall production).  Otherwise I guess I have to give up.

grzegorz914, 2025-01-02:
I will add a workaround for this asap.

grzegorz914, 2025-01-03:
please try 9.6.12-beta.0 and let me know

ipendle, 2025-01-03:
Thanks for looking at this. It seems to progress further now. I get no red errors in the log, but I still dont see anything in Homekit itself. Not sure what I should be expecting to see, but I was expecting some kind of solar widget on the Home.
Attached are the logs 
[homebridge.log.txt](https://github.com/user-attachments/files/18301145/homebridge.log.txt)

grzegorz914, 2025-01-03:
To see all data please use EVE or Controller app because Home app doesn’t support custom characteristics. You can read here https://github.com/grzegorz914/homebridge-enphase-envoy?tab=readme-ov-file#about-the-plugin what is supported by Home app native.

grzegorz914, 2025-01-03:
@ipendle 
please update to beta.2 and get me the log again, the characteristics warning should be fixed

grzegorz914, 2025-01-03:
v9.6.12 is released, so I close this issue as solved.

closed 2025-01-03 by grzegorz914