grzegorz914/homebridge-enphase-envoy #184: Authentication

steve-lad, 2025-03-18:
Further to issue 183, the plugin is now installed, but I am not able to progress because of the self-signed certificate

> [3/18/2025, 6:18:39 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 Garage, Error: Start error: Error: Update home error: Error: self-signed certificate, trying again.

I think that envoy requires additional credentials in order to provide a correct token (user, serial number, password), but your plugin only allows to provide a password

steve-lad, 2025-03-18:
It may also be useful to know that I am using homebridge on a mac - prior attempts to use homebridge on a raspberry failed because of my network configuration - this error is typically mac!!!

grzegorz914, 2025-03-18:
Please enable Modern Firmware then in Credentiall will be user and Passwords required

steve-lad, 2025-03-18:
> Device: 192.168.30.150 Garage, Error: Start error: Error: Check JWT token error: Error: Get JWT token error: Error: Refresh token error: Error: Get token error: AxiosError: Request failed with status code 500, trying again.

steve-lad, 2025-03-18:
Perhaps it would be good to allow users to review (view) password and serial number - with mac autofill, it is impossible to know what has been entered and where, unfortunately!

grzegorz914, 2025-03-18:
You can see it in config.json

grzegorz914, 2025-03-18:
If You have some suggestions please let me know, just open new issue with proposed changes and I will look what can I do to improve this project.

closed 2025-03-19 by grzegorz914