grzegorz914/homebridge-enphase-envoy #185: Suggestions for improvements on Initial UI

steve-lad, 2025-03-18:
Two suggestions to facilitate interaction with your great plugin:

1. There should be a global credentials section, that is only presented prior to first authentication in the main popup; once authenticated, it can be moved to a separate tab under advanced (it does not need to be repeated in each "device" tab), but it must be visible in the initial setup as without this, nothing will happen

2. Default to modern for credentials, with a note (mouseover) that says that only password is required for versions prior to 7.x.x

grzegorz914, 2025-03-18:
1 - changed Envoy section to get better layout of credentials and other tabs in 9.3.4-beta

steve-lad, 2025-03-20:
I see the changes that you have made - good. Why do you not move credentials out of advanced - newbies will not be expecting to go into advanced to get the plug in to work??

grzegorz914, 2025-03-20:
Try 9.9.8-beta to see credentials changes

steve-lad, 2025-03-20:
IMHO, that is much better for a newbie to understand. Well done.

grzegorz914, 2025-04-01:
What do You Think about latest changes in UI?

steve-lad, 2025-04-01:
Once I have verified that all is working good, will give some detailed feedback, but first impressions are very positive.

steve-lad, 2025-04-02:
9.11.0-Beta.11

The configUI interface is much improved. The credentials section is easily seen, and easily set.

I wonder if it is necessary to have both an ACBatteries tab and  an Ensemble tab; I understand that this probably relates to how data is reported with different configurations, but I think that it would be clearer if there was just a Batteries tab, regardless of battery type as it is currently possible to configure in both, although one config is ignored.

Regarding the programmable Contacts, perhaps calling them sensors is misleading. For example, Backup Level Sensors could be more clearly named User (Defined) BackUp Level Switches (or Contacts) as they are not sensors, and this gives a clearer (if lengthier) idea that it is user definable. This applies to all contacts.

[image]

I am unclear as to the Individual accessory role. Perhaps i need to add the accessory again, but it appears to add nothing.

Finally, it looks like there are too many descriptions in the AC Batteries, Encharges and possibly other tabs - see below example screenshot.

[image]

These are minor details. The plugin works well. Good Job.

grzegorz914, 2025-04-02:
AC Batterie are available in both firmware < 7+ and > 7+, encharges exist only with firmware 7+. If you use as login mode first options then encharges section is hidden and cannot be configured. I think is ok as it is now.

Other things changed with latest beta, try and let me know.

grzegorz914, 2025-04-06:
9.12.x released with some changes in envoy section

steve-lad, 2025-04-15:
9.13 looks good; and is much simpler. Very good