grzegorz914/homebridge-enphase-envoy #186: Battery Charge State Sensor Please

steve-lad, 2025-03-19:
Feature Request.

Please could you add a battery charge state sensor. This would enable a when condition that could be used to execute an automation when the battery charge state exceeds a certain percentage (ie 80%).

Above a certain percentage, enphase dump production to the grid so that the battery is not charged to fast. This sensor would allow to activate loads to use this power rather than waste it!

grzegorz914, 2025-03-19:
This is actually there, Advanced Settings>>>> Ensemble>>Encharge>>State Sensor

steve-lad, 2025-03-19:
As you say, charge state is already there, but it only gives if it is charging or not. Equally the charge level is available, however it is difficult to use as a trigger in home, unless one passes via Eve, and even then it is hard to modify.

I am looking for a sensor as in production, where one can say it is on if charge level is > x%. Is that a clearer explanation?

grzegorz914, 2025-03-19:
Advanced Settings>>>> Ensemble>>Encharge>>Backup Level Sensors

steve-lad, 2025-03-19:
Thanks - that was what I was looking for. Unfortunately, it does not seem to work. It is visible in json, but is not visible in Eve or home? Is it related to the slider vs direct entry?

grzegorz914, 2025-03-19:
Can You enable debug mode in plugin config and post the log, I will look if this accessory is published correct. Check also You are set all mandatory properties like Accessory Type

steve-lad, 2025-03-19:
This is the json from the debug log - there is too much to sanitise in the rest :-)
```
"name": "Garage",
  "host": "192.168.30.150",
  "disableAccessory": false,
  "envoyFirmware7xx": true,
  "envoyFirmware7xxTokenGenerationMode": 0,
  "envoyPasswd": "removed",
  "envoyTokenInstaller": false,
  "envoySerialNumber": "removed",
  "enlightenUser": “removed”,
  "enlightenPasswd": "removed",
  "supportPlcLevel": false,
  "plcLevelControl": {
    "displayType": 0,
    "namePrefix": false
  },
  "supportPowerProductionState": true,
  "powerProductionStateControl": {
    "displayType": 0,
    "namePrefix": false
  },
  "powerProductionSummary": 1000,
  "powerProductionStateSensor": {
    "displayType": 0,
    "namePrefix": false
  },
  "powerProductionLevelSensors": [
    {
      "compareMode": 0,
      "displayType": 0,
      "namePrefix": false
    },
    {
      "name": "Production Level > 3Kw",
      "compareMode": 0,
      "powerLevel": 3000,
      "displayType": 3,
      "namePrefix": false
    }
  ],
  "energyProductionStateSensor": {
    "displayType": 0,
    "namePrefix": false
  },
  "energyProductionLevelSensors": [
    {
      "compareMode": 0,
      "energyLevel": 0,
      "displayType": 0,
      "namePrefix": false
    }
  ],
  "energyProductionLifetimeOffset": 0,
  "powerConsumptionTotalStateSensor": {
    "displayType": 0,
    "namePrefix": false
  },
  "powerConsumptionTotalLevelSensors": [
    {
      "compareMode": 0,
      "powerLevel": 0,
      "displayType": 0,
      "namePrefix": false
    }
  ],
  "energyConsumptionTotalLifetimeOffset": 0,
  "energyConsumptionTotalStateSensor": {
    "displayType": 0,
    "namePrefix": false
  },
  "energyConsumptionTotalLevelSensors": [
    {
      "compareMode": 0,
      "energyLevel": 0,
      "displayType": 0,
      "namePrefix": false
    }
  ],
  "powerConsumptionNetStateSensor": {
    "displayType": 0,
    "namePrefix": false
  },
  "powerConsumptionNetLevelSensors": [
    {
      "name": "Exporting >500",
      "compareMode": 3,
      "powerLevel": -500,
      "displayType": 3,
      "namePrefix": false
    }
  ],
  "energyConsumptionNetLifetimeOffset": 0,
  "energyConsumptionNetStateSensor": {
    "displayType": 0,
    "namePrefix": false
  },
  "energyConsumptionNetLevelSensors": [
    {
      "compareMode": 0,
      "energyLevel": 0,
      "displayType": 0,
      "namePrefix": false
    }
  ],
  "enchargeProfileControl": [
    {
      "profile": "backup",
      "displayType": 0,
      "namePrefix": false
    }
  ],
  "enchargeProfileSensors": [
    {
      "profile": "backup",
      "displayType": 0,
      "namePrefix": false
    }
  ],
  "enchargeStateSensor": {
    "displayType": 0,
    "namePrefix": false
  },
  "enchargeGridStateSensor": {
    "displayType": 0,
    "namePrefix": false
  },
  "enchargeGridModeSensors": [
    {
      "gridMode": "on-grid",
      "displayType": 0,
      "namePrefix": false
    }
  ],
  "enchargeBackupLevelSensors": [
    {
      "name": "Backup>85%",
      "compareMode": 0,
      "backupLevel": 85,
      "displayType": 3,
      "namePrefix": false
    }
  ],
  "enpowerGridStateControl": {
    "displayType": 0,
    "namePrefix": false
  },
  "enpowerGridStateSensor": {
    "displayType": 0,
    "namePrefix": false
  },
  "enpowerGridModeSensors": [
    {
      "gridMode": "on-grid",
      "displayType": 0,
      "namePrefix": false
    }
  ],
  "enpowerDryContactsControl": false,
  "enpowerDryContactsSensors": false,
  "solarGridStateSensor": {
    "displayType": 0,
    "namePrefix": false
  },
  "solarGridModeSensors": [
    {
      "gridMode": "on-grid",
      "displayType": 0,
      "namePrefix": false
    }
  ],
  "generatorStateControl": {
    "displayType": 0,
    "namePrefix": false
  },
  "generatorStateSensor": {
    "displayType": 0,
    "namePrefix": false
  },
  "generatorModeControl": [
    {
      "mode": "off",
      "displayType": 0,
      "namePrefix": false
    }
  ],
  "generatorModeSensors": [
    {
      "displayType": 0,
      "namePrefix": false
    }
  ],
  "dataRefreshControl": {
    "displayType": 0,
    "namePrefix": false
  },
  "dataRefreshSensor": {
    "displayType": 0,
    "namePrefix": false
  },
  "metersDataRefreshTime": 5,
  "productionDataRefreshTime": 10,
  "liveDataRefreshTime": 5,
  "ensembleDataRefreshTime": 15,
  "disableLogDeviceInfo": false,
  "disableLogInfo": false,
  "disableLogSuccess": false,
  "disableLogWarn": false,
  "disableLogError": false,
  "enableDebugMode": true,
  "restFul": {
    "enable": false,
    "debug": false
  },
  "mqtt": {
    "enable": false,
    "debug": false,
    "auth": false,
    "passwd": "removed"
  },
  "envoyToken": "removed"
}.
```

grzegorz914, 2025-03-19:
I have checked on my side and is published correct, try to remove the`%` from the name, in other case I need see the debug, You can post the debug file.

```
"enchargeBackupLevelSensors": [
    {
      "name": "Backup>85%",
      "compareMode": 0,
      "backupLevel": 85,
      "displayType": 3,
      "namePrefix": false
    }
  ],
```

steve-lad, 2025-03-19:
For your info, I have added another production sensor of the same type and it appeared immediately

steve-lad, 2025-03-19:
That fixed it - was it an issue of counting :-) (with two uncharge?)

steve-lad, 2025-03-19:
However the state is wrong - if it meets the condition, it should be open? (at least that's how the others work) so either it is not measuring what I expect, or there is an issue.

The SOC for the pair of batteries is 100%, the condition is >85%, so the contact should be open?

steve-lad, 2025-03-19:
It is closed

grzegorz914, 2025-03-19:
try beta.17 and I need the debug, the file can be downloaded from there:

[image]

steve-lad, 2025-03-19:
`Device: envoy.local Garage, Error: Start error: Error: Update ensemble status error: ReferenceError: ensemblesSupported is not defined, trying again.`

steve-lad, 2025-03-19:
Back on 16

grzegorz914, 2025-03-19:
try 18

steve-lad, 2025-03-19:
`Device: envoy.local Garage, Ensemble, encharges agg percent full: 100 %.`

steve-lad, 2025-03-19:
In the logs, I only see one of the 2 battery states, and the aggregate state, so if you are polling across the 2 and averaging, that would explain the difference (perhaps)

grzegorz914, 2025-03-19:
I really need the debug log in other case I can not fix it.

steve-lad, 2025-03-19:
Sanitised debug file

[homebridge.log](https://github.com/user-attachments/files/19347453/homebridge.log)

grzegorz914, 2025-03-19:
fixed in beta.21

steve-lad, 2025-03-19:
For the moment, looks good. Thanks for your efforts, and for this plugin. I will propose some part numbers, and my view of a howto to you tomorrow

grzegorz914, 2025-03-19:
9.9.4.released

closed 2025-03-19 by grzegorz914