grzegorz914/homebridge-enphase-envoy #188: Some accessories that show in Home cannot be modified in the plugin

steve-lad, 2025-03-20:
Currently have 5 accessories in the Home App. 

3 are contact sensors that I have configured, that show in json, and are correct.

The two others are the encharges backup level, and a "light bulb" that is always on and represents, I guess, the envoy. 

In the home bridge UI, accessories, I can rename these accessories or hide them. However, in Home, if I rename the accessory shown as encharges,  as soon as I reset the plugin or upgrade, it reverts to encharges.

Equally, I am unable to hide or change the "icon" associated with the envoy, which is I presume is the parent, as it does not show in config Ui or in json. I would ideally like to be able to hide this "accessory" as I can in home bridge.

This may be a HomeKit limitation, but I am surprised that I cannot at least change the icon in the plugin.

[image]

grzegorz914, 2025-03-20:
Direct from Home app is not possibly to hide accessory, just HomeKit doesn’t offer this possibility. We can add this options to config UI but this option will hide all accessory of same type. For example if we disable encharge then all encharges will be hidden.

steve-lad, 2025-03-20:
Understand and sorry to be demanding - the plugin works fine and we are really only talking eye candy. I can hide the Garage from the summary view anyway.

As you see from the above capture, the parent accessory of encharges is Garage. I would like to keep encharges, but hide the Garage or alternatively, give it an type/icon other than a light bulb, as HomeKit then treats it as a light and includes it in the lights summary - I know there are not many possibilities, but maybe as a switch, if it cannot be made configurable via the config UI?

grzegorz914, 2025-03-20:
Try 9.9.7-beta, here You can disable encharges lightbulbs, Advanced Settings>>Ensemble>>Encharge>>Monitoring Accessory

steve-lad, 2025-03-20:
That works - but is not what I was requesting unfortunately.

After using the option, this is the result:

[image]

Reenabling I get this:

[image]

The encharges is useful and meaningful; but the garage (9F93) is nothing, just a light that is 100% all of the time!!

grzegorz914, 2025-03-20:
What is Garage?

grzegorz914, 2025-03-20:
If garage is Envoy then You need to set Advanced Settings>>Production>>>Power Summary oaf all your Microinverters in Watt

grzegorz914, 2025-03-20:
Encharges can’t be called Battery because by en-haste in old system Battery already exist and we cannot duplicate it.

steve-lad, 2025-03-20:
OK - thanks for the power summary tip. Now it is meaningful. I will change the name from Garage as well so that it makes more sense.

Regarding encharges, understand the system issue, but does that have to be an issue user side?

steve-lad, 2025-03-20:
I have looked at the various HM types for accessories - If home bridge allows,does it not make sense to treat the envoy output (PV production as % of Max) as an outlet with a percentage?

If so, could this not also be applied as the symbol for encharges as well?

This way these two accessories would not show in either lights or security, if I have understood correctly

grzegorz914, 2025-03-20:
Every services need its unique name, so if we change this name also battery need to be changed. I don’t think is good idea, what happens if you change this name in Home app direct?

grzegorz914, 2025-03-20:
The outlet cannot display percentage.

grzegorz914, 2025-03-20:
I will add possibility to change the name from config UI.

steve-lad, 2025-03-20:
Great. 

Looking here, it seems that you can define primary characteristics for any accessory, but I do not have your experience

`https://developer.apple.com/documentation/homekit/configuring-a-home-automation-device`

grzegorz914, 2025-03-20:
Try beta 10 and let me know, the name can be set in encharge section

steve-lad, 2025-03-20:
Not good. See below

[image]

Encharges have disappeared, and the 9F53 has been appended to the envoy name assigned

grzegorz914, 2025-03-20:
Please enable the encharges accessory, I have changed the logic

steve-lad, 2025-03-20:
Perfect - bad tester

[image]

grzegorz914, 2025-03-20:
I close this as complete, 9.9.7 released.

closed 2025-03-20 by grzegorz914